openhab/org.openhab.binding.zwave #263: [Transactions+Security] Not yet identified devices always treated as asleep

pizerg, 2016-12-15:
In the method "sendNextMessage" from TransactionsManager the following piece of code:

```
           if (node.isAwake() == false) {
                        logger.debug("NODE {}: Node not awake!", nodeId);
                        continue;
           }
```

It always considers the node not awake if it's in the stage IDENTIFY_NODE (for example after adding a new device or even with the Controller (node 1)) as the fields used for validation of awake/asleep have not yet been set until IDENTIFY_NODE is completed.

cdjackson, 2016-12-15:
This is already fixed. The version on GH is very old unfortunately so you will have many problems - sorry.

pizerg, 2016-12-15:
OK, I'll close the issue and will wait until your next commit on GH to continue testing.

closed 2016-12-15 by pizerg