openhab/org.openhab.binding.zwave #273: FLiRS nodes do not initialize

jspuij, 2016-12-21:
I've noticed that with the master branch, my FLiRS siren does not start to initialize. The controller reports this node as failed, so I guess initialization is not started, but a FLiRS node never sends a wakeup message. So despite of the failed node state, you have to start sending messages to this node to start the initialization process. Have to dig a bit deeper to confirm this though.
OZCP initializes the node without any problems.

cdjackson, 2016-12-21:
I wouldn’t advise spending any time on this - there’s a major rewrite coming and initialisation is pretty much totally rewritten.

cdjackson, 2016-12-21:
I should add that in the new system, my FLiRS device (i.e. a lock) works fine...

jspuij, 2016-12-22:
@cdjackson That's your transaction-refactoring branch right? Want me to have a go at it and test? Still have a 30+ node network running here ;-)

closed 2016-12-22 by jspuij

cdjackson, 2016-12-22:
Sure - feel free. Any feedback would be welcome…