openhab/org.openhab.binding.zwave #281: Add Everspring AN163 power switch with metering to database

jwoytek, 2016-12-26:
Please add the Everspring AN163 power switch wth metering to the database. XML will be attached.

* OH2 Version: 2.0.0-SNAPSHOT Build #663
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): 2.0.0.201612261716

If the problem is associated with a device, please provide the following -:

* Device name: Everspring AN163
* Device version:
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list):
* If the device is not in the database, provide a link to the manual.
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.

`<node>
  <deviceClass>
    <basicDeviceClass>ROUTING_SLAVE</basicDeviceClass>
    <genericDeviceClass>BINARY_SWITCH</genericDeviceClass>
    <specificDeviceClass>POWER_SWITCH_BINARY</specificDeviceClass>
  </deviceClass>
  <homeId>0xca9b946a</homeId>
  <nodeId>5</nodeId>
  <version>4</version>
  <manufacturer>0x60</manufacturer>
  <deviceId>0x5</deviceId>
  <deviceType>0x4</deviceType>
  <listening>true</listening>
  <frequentlyListening>false</frequentlyListening>
  <routing>true</routing>
  <security>false</security>
  <beaming>true</beaming>
  <maxBaudRate>40000</maxBaudRate>
  <nodeInformationFrame>
    <commandClass>SWITCH_BINARY</commandClass>
    <commandClass>BASIC</commandClass>
    <commandClass>MANUFACTURER_SPECIFIC</commandClass>
    <commandClass>VERSION</commandClass>
    <commandClass>SWITCH_ALL</commandClass>
    <commandClass>ASSOCIATION</commandClass>
    <commandClass>METER</commandClass>
    <commandClass>CONFIGURATION</commandClass>
    <commandClass>ALARM</commandClass>
  </nodeInformationFrame>
  <supportedCommandClasses>
    <entry>
      <commandClass>NO_OPERATION</commandClass>
      <noOperationCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
      </noOperationCommandClass>
    </entry>
    <entry>
      <commandClass>ALARM</commandClass>
      <alarmCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <alarms/>
        <v1Supported>false</v1Supported>
        <isGetSupported>true</isGetSupported>
      </alarmCommandClass>
    </entry>
    <entry>
      <commandClass>SWITCH_BINARY</commandClass>
      <binarySwitchCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <isGetSupported>true</isGetSupported>
      </binarySwitchCommandClass>
    </entry>
    <entry>
      <commandClass>VERSION</commandClass>
      <versionCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <libraryType>LIB_SLAVE_ROUTING</libraryType>
        <protocolVersion>3.52</protocolVersion>
        <applicationVersion>2.0</applicationVersion>
      </versionCommandClass>
    </entry>
    <entry>
      <commandClass>CONFIGURATION</commandClass>
      <configurationCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <configParameters/>
      </configurationCommandClass>
    </entry>
    <entry>
      <commandClass>BASIC</commandClass>
      <basicCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <isGetSupported>true</isGetSupported>
      </basicCommandClass>
    </entry>
    <entry>
      <commandClass>ASSOCIATION</commandClass>
      <associationCommandClass>
        <version>2</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <maxGroups>2</maxGroups>
      </associationCommandClass>
    </entry>
    <entry>
      <commandClass>METER</commandClass>
      <meterCommandClass>
        <version>2</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <meterType>ELECTRIC</meterType>
        <meterScales>
          <meterScale>E_KWh</meterScale>
          <meterScale>E_W</meterScale>
        </meterScales>
        <canReset>true</canReset>
        <isGetSupported>true</isGetSupported>
        <isSupportRequestSupported>true</isSupportRequestSupported>
      </meterCommandClass>
    </entry>
    <entry>
      <commandClass>SWITCH_ALL</commandClass>
      <switchAllCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <isGetSupported>true</isGetSupported>
        <mode>SWITCH_ALL_INCLUDE_ON_OFF</mode>
      </switchAllCommandClass>
    </entry>
    <entry>
      <commandClass>MANUFACTURER_SPECIFIC</commandClass>
      <manufacturerSpecificCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <initSerialNumber>false</initSerialNumber>
        <deviceManufacturer>96</deviceManufacturer>
        <deviceType>4</deviceType>
        <deviceId>5</deviceId>
      </manufacturerSpecificCommandClass>
    </entry>
  </supportedCommandClasses>
  <securedCommandClasses/>
  <associationGroups>
    <entry>
      <int>1</int>
      <associationGroup>
        <index>1</index>
        <associations/>
      </associationGroup>
    </entry>
    <entry>
      <int>2</int>
      <associationGroup>
        <index>2</index>
        <associations/>
      </associationGroup>
    </entry>
  </associationGroups>
  <nodeNeighbors>
    <int>1</int>
    <int>2</int>
    <int>3</int>
    <int>4</int>
    <int>6</int>
    <int>7</int>
  </nodeNeighbors>
  <lastSent>2016-12-26 21:21:28.360 UTC</lastSent>
  <lastReceived>2016-12-26 21:21:28.403 UTC</lastReceived>
</node>`

jwoytek, 2016-12-26:
Sorry. Hopefully this XML is formatted a little better! 

```
<node>
  <deviceClass>
    <basicDeviceClass>ROUTING_SLAVE</basicDeviceClass>
    <genericDeviceClass>BINARY_SWITCH</genericDeviceClass>
    <specificDeviceClass>POWER_SWITCH_BINARY</specificDeviceClass>
  </deviceClass>
  <homeId>0xca9b946a</homeId>
  <nodeId>5</nodeId>
  <version>4</version>
  <manufacturer>0x60</manufacturer>
  <deviceId>0x5</deviceId>
  <deviceType>0x4</deviceType>
  <listening>true</listening>
  <frequentlyListening>false</frequentlyListening>
  <routing>true</routing>
  <security>false</security>
  <beaming>true</beaming>
  <maxBaudRate>40000</maxBaudRate>
  <nodeInformationFrame>
    <commandClass>SWITCH_BINARY</commandClass>
    <commandClass>BASIC</commandClass>
    <commandClass>MANUFACTURER_SPECIFIC</commandClass>
    <commandClass>VERSION</commandClass>
    <commandClass>SWITCH_ALL</commandClass>
    <commandClass>ASSOCIATION</commandClass>
    <commandClass>METER</commandClass>
    <commandClass>CONFIGURATION</commandClass>
    <commandClass>ALARM</commandClass>
  </nodeInformationFrame>
  <supportedCommandClasses>
    <entry>
      <commandClass>NO_OPERATION</commandClass>
      <noOperationCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
      </noOperationCommandClass>
    </entry>
    <entry>
      <commandClass>ALARM</commandClass>
      <alarmCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <alarms/>
        <v1Supported>false</v1Supported>
        <isGetSupported>true</isGetSupported>
      </alarmCommandClass>
    </entry>
    <entry>
      <commandClass>SWITCH_BINARY</commandClass>
      <binarySwitchCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <isGetSupported>true</isGetSupported>
      </binarySwitchCommandClass>
    </entry>
    <entry>
      <commandClass>VERSION</commandClass>
      <versionCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <libraryType>LIB_SLAVE_ROUTING</libraryType>
        <protocolVersion>3.52</protocolVersion>
        <applicationVersion>2.0</applicationVersion>
      </versionCommandClass>
    </entry>
    <entry>
      <commandClass>CONFIGURATION</commandClass>
      <configurationCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <configParameters/>
      </configurationCommandClass>
    </entry>
    <entry>
      <commandClass>BASIC</commandClass>
      <basicCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <isGetSupported>true</isGetSupported>
      </basicCommandClass>
    </entry>
    <entry>
      <commandClass>ASSOCIATION</commandClass>
      <associationCommandClass>
        <version>2</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <maxGroups>2</maxGroups>
      </associationCommandClass>
    </entry>
    <entry>
      <commandClass>METER</commandClass>
      <meterCommandClass>
        <version>2</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <meterType>ELECTRIC</meterType>
        <meterScales>
          <meterScale>E_KWh</meterScale>
          <meterScale>E_W</meterScale>
        </meterScales>
        <canReset>true</canReset>
        <isGetSupported>true</isGetSupported>
        <isSupportRequestSupported>true</isSupportRequestSupported>
      </meterCommandClass>
    </entry>
    <entry>
      <commandClass>SWITCH_ALL</commandClass>
      <switchAllCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <isGetSupported>true</isGetSupported>
        <mode>SWITCH_ALL_INCLUDE_ON_OFF</mode>
      </switchAllCommandClass>
    </entry>
    <entry>
      <commandClass>MANUFACTURER_SPECIFIC</commandClass>
      <manufacturerSpecificCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
        <initSerialNumber>false</initSerialNumber>
        <deviceManufacturer>96</deviceManufacturer>
        <deviceType>4</deviceType>
        <deviceId>5</deviceId>
      </manufacturerSpecificCommandClass>
    </entry>
  </supportedCommandClasses>
  <securedCommandClasses/>
  <associationGroups>
    <entry>
      <int>1</int>
      <associationGroup>
        <index>1</index>
        <associations/>
      </associationGroup>
    </entry>
    <entry>
      <int>2</int>
      <associationGroup>
        <index>2</index>
        <associations/>
      </associationGroup>
    </entry>
  </associationGroups>
  <nodeNeighbors>
    <int>1</int>
    <int>2</int>
    <int>3</int>
    <int>4</int>
    <int>6</int>
    <int>7</int>
  </nodeNeighbors>
  <lastSent>2016-12-26 21:21:28.360 UTC</lastSent>
  <lastReceived>2016-12-26 21:21:28.403 UTC</lastReceived>
</node>
```

cdjackson, 2016-12-26:
I’ve created a new device (http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/492 <http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/492>) it would be appreciated if you could add any configuration.

jwoytek, 2016-12-27:
Awesome. The module supports four configuration parameters, 1-4. I'm attaching the user manual here. Page 4, section 3 contains a chart detailing the four parameters, byte sizes, and valid values. 

Thanks! 
[FCCID.io-2248775.pdf](https://github.com/openhab/org.openhab.binding.zwave/files/673499/FCCID.io-2248775.pdf)

cdjackson, 2016-12-27:
Thanks. It would be great if you could add the parameters to the database itself. There's a guide on the site. 

Thanks

jwoytek, 2016-12-27:
When I log in and go to do that, the top of the page says that the site
owner needs to give me access before I can make changes. I will try again
shortly in case that was a temporary issue.

Jonathan

cdjackson, 2016-12-27:
I’ve now given you access to edit - you need to actually ask for it as if I automatically give access sooner or later someone will do something we don’t like...

jwoytek, 2016-12-27:
Sadly, all too true. I'll get this in shortly. Thank you!

jwoytek, 2016-12-28:
Tested in the latest snapshot build of OH2. Everything seems to work! Thanks!

closed 2016-12-28 by jwoytek