openhab/org.openhab.binding.zwave #287: [ZMNHZD]: multiple battery-level entries found in XML export for OH2

yellowmm, 2016-12-30:
* OH2 Version: openhab-offline-2.0.0-SNAPSHOT from 20161229
* Binding Version: org.openhab.binding.zwave/2.0.0-SNAPSHOT from 20161229
* Device name: GOAP ZMNHZD - Multifunctional Weather Station	 
* Device version: n/a
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/364?layout=openhab2)

It seems that the XML exporter for OH2 generates same channel-ids for **battery-level** for all endpoints (you must know the weather station contains 4 devices with batteries):
`<channel id="battery-level" typeId="system.battery-level">`

For other channel types its working:
`<channel id="sensor_temperature1" typeId="sensor_temperature">`

My expectation is e.g.:
`<channel id="battery-level1" typeId="system.battery-level">`

What do you think?

cdjackson, 2016-12-31:
I'll take a look at the database exporter - I didn't assume that there would be multiple batteries when I wrote it...

cdjackson, 2016-12-31:
This should now be fixed - I'll hopefully do a database update later.

yellowmm, 2017-01-02:
Thanks @cdjackson. This is really good news.

closed 2017-01-03 by yellowmm