openhab/org.openhab.binding.zwave #288: [bug] Wrong electric meter results in GreenWave PowerNode

kubawolanin, 2016-12-31:
* OH2 Version: openHAB 2.0 Build #684
* Binding Version: 2.0.0.201612291742

* Device name: GreenWave Multi-socket PowerNode
* Device version: GWPN6
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): [GWPN6](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/100)

Hello! I've updated my openHAB along with all bindings (I jumped from distro 623 IIRC).
Since then I've observed that my PowerNode electric meters provide wrong measurements. Looks like the value of `Electric meter (watts)` for `Switch 2` is swapped with the one attached ot `Switch 1`
[image]

Didn't find anything meaningful in logs whatsoever.
I happen to have a backup of my `node5.xml` file from before the update.
[node5-BAD.xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/679602/node5-BAD.xml.txt) - current, improper or missing values
[node5-GOOD.xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/679603/node5-GOOD.xml.txt) - older, everything worked properly

Thank you!

cdjackson, 2016-12-31:
Please provide debug logs showing the problem.

kubawolanin, 2016-12-31:
Hi Chris,
Here's all I could grep for `NODE 5`:
```
2016-12-12 09:48:22.278 [WARN ] [.commandclass.ZWaveMeterCommandClass] - NODE 5: Invalid meter type 5
2016-12-12 09:49:01.192 [WARN ] [.commandclass.ZWaveMeterCommandClass] - Unsupported Command 0x42 for command class METER (0x32).
2016-12-12 10:48:10.563 [ERROR] [.commandclass.ZWaveMeterCommandClass] - NODE 5: Meter Report: Buffer too short: length=10, required=11
2016-12-12 10:48:11.231 [ERROR] [.commandclass.ZWaveMeterCommandClass] - NODE 5: Meter Report: Buffer too short: length=10, required=11
2016-12-12 18:48:08.911 [ERROR] [.commandclass.ZWaveMeterCommandClass] - NODE 5: Meter Report: Buffer too short: length=10, required=11
2016-12-13 06:17:48.137 [ERROR] [.commandclass.ZWaveMeterCommandClass] - NODE 5: Meter Report: Buffer too short: length=6, required=7
2016-12-16 18:06:44.262 [ERROR] [.commandclass.ZWaveMeterCommandClass] - NODE 5: Meter Report: Buffer too short: length=10, required=11
```

```
2016-12-19 11:47:51.921 [ERROR] [ocol.ZWaveController$ZWaveSendThread] - NODE 5: Timeout while sending message. Requeueing - 2 attempts left!
2016-12-19 11:47:51.923 [ERROR] [l.serialmessage.SendDataMessageClass] - NODE 5: Got an error while sending data. Resending message.
2016-12-19 11:47:56.927 [ERROR] [ocol.ZWaveController$ZWaveSendThread] - NODE 5: Timeout while sending message. Requeueing - 1 attempts left!
2016-12-19 11:47:56.929 [ERROR] [l.serialmessage.SendDataMessageClass] - NODE 5: Got an error while sending data. Resending message.
2016-12-19 11:48:01.934 [ERROR] [ocol.ZWaveController$ZWaveSendThread] - NODE 5: Timeout while sending message. Requeueing - 0 attempts left!
2016-12-19 11:48:01.935 [ERROR] [l.serialmessage.SendDataMessageClass] - NODE 5: Got an error while sending data. Resending message.
2016-12-19 11:48:06.940 [WARN ] [ocol.ZWaveController$ZWaveSendThread] - NODE 5: Too many retries. Discarding message: Message: class=SendData[0x13], type=Request[0x00], priority=Get, dest=5, callback=149, payload=05 03 71 04 00
```

Thank you!

cdjackson, 2016-12-31:
I really need a debug log that shows the data being received.

> 2016-12-12 09:48:22.278 [WARN ] [.commandclass.ZWaveMeterCommandClass] - NODE 5: Invalid meter type 5
> 2016-12-12 09:49:01.192 [WARN ] [.commandclass.ZWaveMeterCommandClass] - Unsupported Command 0x42 for command class METER (0x32).

Errors like this indicate something bad is being sent from the device - I would try resetting it. I have a TKB device that periodically sends crap and it needs to be power cycled - maybe it’s the same issue.

kubawolanin, 2016-12-31:
Thanks, I did reset the device but it didn't help. Here's my log:
```
2016-12-31 13:44:55.519 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Command received zwave:device:e91a634b:node5:switch_binary1 --> ON
2016-12-31 13:44:55.522 [DEBUG] [dclass.ZWaveBinarySwitchCommandClass] - NODE 5: Creating new message for application command SWITCH_BINARY_SET
2016-12-31 13:44:55.525 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 5: Encapsulating message, instance / endpoint 1
2016-12-31 13:44:55.529 [DEBUG] [class.ZWaveMultiInstanceCommandClass] - NODE 5: Creating new message for command MULTI_CHANNEL_ENCAP endpoint 1
2016-12-31 13:44:55.543 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 5: Sending REQUEST Message = 01 0E 00 13 05 07 60 0D 01 01 25 01 FF 25 C0 B3 
2016-12-31 13:44:55.606 [DEBUG] [l.serialmessage.SendDataMessageClass] - NODE 5: Sent Data successfully placed on stack.
2016-12-31 13:44:55.628 [DEBUG] [l.serialmessage.SendDataMessageClass] - NODE 5: SendData Request. CallBack ID = 192, Status = Transmission complete and ACK received(0)
2016-12-31 13:44:55.631 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 5: Starting initialisation from DONE
2016-12-31 13:44:55.646 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got an event from Z-Wave network: ZWaveTransactionCompletedEvent
2016-12-31 13:44:55.648 [DEBUG] [ocol.ZWaveController$ZWaveSendThread] - NODE 5: Response processed after 71ms/3717ms.
2016-12-31 13:44:55.949 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 5: Application Command Request (ALIVE:DONE)
2016-12-31 13:44:55.951 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 5: Starting initialisation from DONE
2016-12-31 13:44:55.955 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 5: Incoming command class MULTI_INSTANCE
2016-12-31 13:44:55.956 [DEBUG] [class.ZWaveMultiInstanceCommandClass] - NODE 5: Received MULTI_INSTANCE command V2
2016-12-31 13:44:55.958 [DEBUG] [class.ZWaveMultiInstanceCommandClass] - NODE 5: Requested Command Class = METER (0x32)
2016-12-31 13:44:55.960 [DEBUG] [class.ZWaveMultiInstanceCommandClass] - NODE 5: Endpoint = 1, calling handleApplicationCommandRequest.
2016-12-31 13:44:55.961 [DEBUG] [.commandclass.ZWaveMeterCommandClass] - NODE 5: Received METER command V2
2016-12-31 13:44:55.964 [DEBUG] [.commandclass.ZWaveMeterCommandClass] - NODE 5: Meter: Type=Electric(1), Scale=W(2), Value=12.9
2016-12-31 13:44:55.967 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got an event from Z-Wave network: ZWaveMeterValueEvent
2016-12-31 13:44:55.969 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got a value event from Z-Wave network, endpoint = 1, command class = METER, value = 12.9
2016-12-31 13:44:55.972 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Updating channel state zwave:device:e91a634b:node5:meter_watts1 to 12.9 [DecimalType]
2016-12-31 13:45:04.956 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 5: Application Command Request (ALIVE:DONE)
2016-12-31 13:45:04.958 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 5: Starting initialisation from DONE
2016-12-31 13:45:04.962 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 5: Incoming command class MULTI_INSTANCE
2016-12-31 13:45:04.964 [DEBUG] [class.ZWaveMultiInstanceCommandClass] - NODE 5: Received MULTI_INSTANCE command V2
2016-12-31 13:45:04.966 [DEBUG] [class.ZWaveMultiInstanceCommandClass] - NODE 5: Requested Command Class = METER (0x32)
2016-12-31 13:45:04.968 [DEBUG] [class.ZWaveMultiInstanceCommandClass] - NODE 5: Endpoint = 1, calling handleApplicationCommandRequest.
2016-12-31 13:45:04.969 [DEBUG] [.commandclass.ZWaveMeterCommandClass] - NODE 5: Received METER command V2
2016-12-31 13:45:04.971 [DEBUG] [.commandclass.ZWaveMeterCommandClass] - NODE 5: Meter: Type=Electric(1), Scale=W(2), Value=7.4
2016-12-31 13:45:04.975 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got an event from Z-Wave network: ZWaveMeterValueEvent
2016-12-31 13:45:04.977 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got a value event from Z-Wave network, endpoint = 1, command class = METER, value = 7.4
2016-12-31 13:45:04.979 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Updating channel state zwave:device:e91a634b:node5:meter_watts1 to 7.4 [DecimalType]
2016-12-31 13:45:13.947 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 5: Application Command Request (ALIVE:DONE)
2016-12-31 13:45:13.947 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 5: Starting initialisation from DONE
2016-12-31 13:45:13.949 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 5: Incoming command class MULTI_INSTANCE
2016-12-31 13:45:13.950 [DEBUG] [class.ZWaveMultiInstanceCommandClass] - NODE 5: Received MULTI_INSTANCE command V2
2016-12-31 13:45:13.951 [DEBUG] [class.ZWaveMultiInstanceCommandClass] - NODE 5: Requested Command Class = METER (0x32)
2016-12-31 13:45:13.953 [DEBUG] [class.ZWaveMultiInstanceCommandClass] - NODE 5: Endpoint = 1, calling handleApplicationCommandRequest.
2016-12-31 13:45:13.954 [DEBUG] [.commandclass.ZWaveMeterCommandClass] - NODE 5: Received METER command V2
2016-12-31 13:45:13.955 [DEBUG] [.commandclass.ZWaveMeterCommandClass] - NODE 5: Meter: Type=Electric(1), Scale=W(2), Value=45.1
2016-12-31 13:45:13.957 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got an event from Z-Wave network: ZWaveMeterValueEvent
[... cut ...]
2016-12-31 13:48:59.580 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got a value event from Z-Wave network, endpoint = 1, command class = METER, value = 0.2
2016-12-31 13:48:59.582 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Updating channel state zwave:device:e91a634b:node5:meter_watts1 to 0.2 [DecimalType]
```

[image]

What's interesting is that the highlighted item does now change properly (electric meter for TV)
But the second item (HTPC meter) doesn't update at all. I tried plugging it into different sockets with no luck.

cdjackson, 2016-12-31:
Well the errors now seem to be gone (at least in this log) and as far as I can see, it looks fine. The binding is updating endpoint 1 (i.e. channel name with the 1 on the end) when it received endpoint 1.

This might be related to the endpoint bug in the Greenwave devices where it sometimes sends data to the wrong endpoint.

kubawolanin, 2016-12-31:
Thank you Chris for your insight. 
I'm closing this issue now since it might be a bug on PowerNode's side. 
If I'll manage to get it working somehow, I will post my results here.
I hope there would be a feature that would allow me to update my devices OTA some day.

Happy New Year! 🙂

cdjackson, 2016-12-31:
There’s a fix in the pipeline that might resolve this . see https://github.com/openhab/org.openhab.binding.zwave/pull/272 <https://github.com/openhab/org.openhab.binding.zwave/pull/272>

kubawolanin, 2017-06-02:
I'm still experiencing improper behaviour of this device. I receive updates for `Theater_TV_Energy` only (`channel="zwave:device:e91a634b:node5:meter_watts1"`) even though I'm expecting other sockets to report measurements as well.
What's more, I observed that `Theater_TV_Energy` receives update like it was `Theater_PowerNode_Energy`

items:
```
    Switch Theater_PowerNode  "PowerNode" <television> (Theater) { channel="zwave:device:e91a634b:node5:switch_binary" }
    Switch Theater_TV      "TV" <television> (Theater) { channel="zwave:device:e91a634b:node5:switch_binary1" }
    Switch Theater_HTPC    "HTPC"      <television> (Theater) { channel="zwave:device:e91a634b:node5:switch_binary2" }
    Switch Theater_Audio   "Audio"     <television> (Theater) { channel="zwave:device:e91a634b:node5:switch_binary3" }
    Switch Theater_Lamp    "Lamp"   <television> (Theater) { channel="zwave:device:e91a634b:node5:switch_binary4" }
    Switch Theater_Speaker "Głośniki"  <television> (Theater) { channel="zwave:device:e91a634b:node5:switch_binary5" }
    Switch Theater_NAS     "NAS"       <television> (Theater) { channel="zwave:device:e91a634b:node5:switch_binary6" }
    Number Theater_PowerNode_Energy "Listwa - prąd" <television> (Theater) { channel="zwave:device:e91a634b:node5:meter_watts" }
    Number Theater_TV_Energy      "Telewizor - prąd" <television> (Theater) { channel="zwave:device:e91a634b:node5:meter_watts1" }
    Number Theater_HTPC_Energy    "HTPC - prąd"      <television> (Theater) { channel="zwave:device:e91a634b:node5:meter_watts2" }
    Number Theater_Audio_Energy   "Audio - prąd"     <television> (Theater) { channel="zwave:device:e91a634b:node5:meter_watts3" }
    Number Theater_Lamp_Energy    "Światło - prąd"   <television> (Theater) { channel="zwave:device:e91a634b:node5:meter_watts4" }
    Number Theater_Speaker_Energy "Głośniki - prąd"  <television> (Theater) { channel="zwave:device:e91a634b:node5:meter_watts5" }
    Number Theater_NAS_Energy     "NAS - prąd"       <television> (Theater) { channel="zwave:device:e91a634b:node5:meter_watts6" }
```

logs:
```
2017-06-02 13:18:15.511 [INFO ] [class.ZWaveMultiInstanceCommandClass] - NODE 5: Received a multi instance encapsulation with a destination endpoint = 5. 
2017-06-02 13:18:15.513 [WARN ] [class.ZWaveMultiInstanceCommandClass] - NODE 5: The originating endpoint is 1. Please notify author.
```

```
2017-06-02 13:29:30.526 [ItemStateChangedEvent     ] - Theater_TV_Energy changed from 32.3 to 27.8
```

```
2017-06-02 13:20:12.497 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 18 00 04 00 05 12 60 0D 01 05 32 02 21 34 00 00 01 57 00 09 00 00 01 A2 44 
2017-06-02 13:20:12.500 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
2017-06-02 13:20:12.502 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 18 00 04 00 05 12 60 0D 01 05 32 02 21 34 00 00 01 57 00 09 00 00 01 A2 44 
2017-06-02 13:20:12.503 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 18 00 04 00 05 12 60 0D 01 05 32 02 21 34 00 00 01 57 00 09 00 00 01 A2 44 
2017-06-02 13:20:12.505 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 05 12 60 0D 01 05 32 02 21 34 00 00 01 57 00 09 00 00 01 A2 
2017-06-02 13:20:12.505 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 5: Application Command Request (ALIVE:DONE)
2017-06-02 13:20:12.507 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 5: Starting initialisation from DONE
2017-06-02 13:20:12.507 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@1223572 already registered
2017-06-02 13:20:12.508 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 5: Incoming command class MULTI_INSTANCE
2017-06-02 13:20:12.509 [DEBUG] [class.ZWaveMultiInstanceCommandClass] - NODE 5: Received MULTI_INSTANCE command V2
2017-06-02 13:20:12.509 [INFO ] [class.ZWaveMultiInstanceCommandClass] - NODE 5: Received a multi instance encapsulation with a destination endpoint = 5. 
2017-06-02 13:20:12.509 [WARN ] [class.ZWaveMultiInstanceCommandClass] - NODE 5: The originating endpoint is 1. Please notify author.
2017-06-02 13:20:12.510 [DEBUG] [class.ZWaveMultiInstanceCommandClass] - NODE 5: Requested Command Class = METER (0x32)
2017-06-02 13:20:12.511 [DEBUG] [class.ZWaveMultiInstanceCommandClass] - NODE 5: Endpoint = 1, calling handleApplicationCommandRequest.
2017-06-02 13:20:12.511 [DEBUG] [.commandclass.ZWaveMeterCommandClass] - NODE 5: Received METER command V2
2017-06-02 13:20:12.512 [DEBUG] [.commandclass.ZWaveMeterCommandClass] - NODE 5: Meter: Type=Electric(1), Scale=W(2), Value=34.3
2017-06-02 13:20:12.512 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveMeterValueEvent
2017-06-02 13:20:12.513 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got an event from Z-Wave network: ZWaveMeterValueEvent
2017-06-02 13:20:12.513 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Got a value event from Z-Wave network, endpoint = 1, command class = METER, value = 34.3
2017-06-02 13:20:12.515 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 5: Updating channel state zwave:device:e91a634b:node5:meter_watts1 to 34.3 [DecimalType]
[... cut ...]  01 FF 
2017-06-02 13:20:39.556 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: Recv Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 05 12 60 0D 01 05 32 02 21 34 00 00 01 4C 00 09 00 00 00 00 
2017-06-02 13:20:39.560 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: class=ApplicationCommandHandler, callback id=75, expected=SendData, cancelled=false      MISMATCH

```

kubawolanin, 2017-06-02:
[node5.xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/1047585/node5.xml.txt)

cdjackson, 2017-06-02:
Which version of OH are you using - this bug was raised against a very old version?  Given that this was fixed I don't think we should reopen a problem from a long time ago and it would be better to report the problem again as it is almost certainly different.

closed 2017-06-02 by cdjackson