openhab/org.openhab.binding.zwave #304: Exception during ZWave thread: Input 2. {}

big-col, 2017-01-09:
OH2 build #697
binding version 2.0.0.201701061129

Noticed this error in the log - anything to worry about?

2017-01-09 23:32:59.117 [ERROR] [ve.internal.protocol.ZWaveController] - Exception during ZWave thread: Input 2. {}
java.lang.NullPointerException
	at org.openhab.binding.zwave.handler.ZWaveThingHandler.updateNodeProperties(ZWaveThingHandler.java:1393)[210:org.openhab.binding.zwave:2.0.0.201701061129]
	at org.openhab.binding.zwave.handler.ZWaveThingHandler.ZWaveIncomingEvent(ZWaveThingHandler.java:1168)[210:org.openhab.binding.zwave:2.0.0.201701061129]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.notifyEventListeners(ZWaveController.java:567)[210:org.openhab.binding.zwave:2.0.0.201701061129]
	at org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer.advanceNodeStage(ZWaveNodeInitStageAdvancer.java:1152)[210:org.openhab.binding.zwave:2.0.0.201701061129]
	at org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer.handleNodeQueue(ZWaveNodeInitStageAdvancer.java:230)[210:org.openhab.binding.zwave:2.0.0.201701061129]
	at org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer.ZWaveIncomingEvent(ZWaveNodeInitStageAdvancer.java:1306)[210:org.openhab.binding.zwave:2.0.0.201701061129]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.notifyEventListeners(ZWaveController.java:567)[210:org.openhab.binding.zwave:2.0.0.201701061129]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingResponseMessage(ZWaveController.java:293)[210:org.openhab.binding.zwave:2.0.0.201701061129]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingMessage(ZWaveController.java:217)[210:org.openhab.binding.zwave:2.0.0.201701061129]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.access$7(ZWaveController.java:208)[210:org.openhab.binding.zwave:2.0.0.201701061129]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController$ZWaveInputThread.run(ZWaveController.java:1324)[210:org.openhab.binding.zwave:2.0.0.201701061129]

Full logfile attached in case needed.

Thanks
Colin

This issue list is to report bugs or enhancements for the openHAB Z-Wave binding. If you're not sure about something, please consider discussing it on the [forum](http://community.openhab.org) before raising an issue - more people follow the forum than monitor the issues list, so if you are having trouble with a device, or want to ask a question, you will have a much larger audience there.

In order to ensure we have the best information to deal with the issue, please fill in the following information (delete as necessary depending on the issue).

* OH2 Version:
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console):

If the problem is associated with a device, please provide the following -:

* Device name:
* Device version:
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list):
* If the device is not in the database, provide a link to the manual.
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.

Provide a full debug level log of the problem. Preferably, you should try use the [log viewer](http://www.cd-jackson.com/index.php/openhab/zwave-log-viewer) to see if you can identify the problem.

If a device will not initialise, ensure the log is provided from the time OH starts.

Thank you.

cdjackson, 2017-01-27:
I believe that this is caused by https://github.com/openhab/org.openhab.binding.zwave/pull/314 which is now fixed. If the problem persists with the released version of OH2, or the current snapshot, please reopen.

closed 2017-01-27 by cdjackson

big-col, 2017-01-28:
OK thanks

antares2001, 2017-03-25:
I'm having the same issue on the most recent unstable OH2:

2017-03-25 14:05:13.837 [ERROR] [ve.internal.protocol.ZWaveController] - Exception during ZWave thread: Input 2. {}
java.lang.IllegalStateException: Could not change thing type because callback is missing
	at org.eclipse.smarthome.core.thing.binding.BaseThingHandler.changeThingType(BaseThingHandler.java:601)[106:org.eclipse.smarthome.core.thing:0.9.0.201703201701]
	at org.openhab.binding.zwave.handler.ZWaveThingHandler.updateThingType(ZWaveThingHandler.java:330)[209:org.openhab.binding.zwave:2.1.0.201703240758]
	at org.openhab.binding.zwave.handler.ZWaveThingHandler.ZWaveIncomingEvent(ZWaveThingHandler.java:1175)[209:org.openhab.binding.zwave:2.1.0.201703240758]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.notifyEventListeners(ZWaveController.java:567)[209:org.openhab.binding.zwave:2.1.0.201703240758]
	at org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer.advanceNodeStage(ZWaveNodeInitStageAdvancer.java:1117)[209:org.openhab.binding.zwave:2.1.0.201703240758]
	at org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer.handleNodeQueue(ZWaveNodeInitStageAdvancer.java:230)[209:org.openhab.binding.zwave:2.1.0.201703240758]
	at org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer.ZWaveIncomingEvent(ZWaveNodeInitStageAdvancer.java:1306)[209:org.openhab.binding.zwave:2.1.0.201703240758]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.notifyEventListeners(ZWaveController.java:567)[209:org.openhab.binding.zwave:2.1.0.201703240758]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingResponseMessage(ZWaveController.java:293)[209:org.openhab.binding.zwave:2.1.0.201703240758]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingMessage(ZWaveController.java:217)[209:org.openhab.binding.zwave:2.1.0.201703240758]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.access$7(ZWaveController.java:208)[209:org.openhab.binding.zwave:2.1.0.201703240758]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController$ZWaveInputThread.run(ZWaveController.java:1324)[209:org.openhab.binding.zwave:2.1.0.201703240758]