openhab/org.openhab.binding.zwave #305: Product to be added to the database

Phil2015, 2017-01-10:
Hi there!

I am asking if there is any possibility of adding the Fibaro Single Switch 2 (FGS-213) to the zwave database?
I'm using openHAB 1.8.3 with latest snapshot of zwave and the switch is not listed in the database yet.

The product shows up as: Device ID 1000 - Device Type 403
There are FGS-211 and FGS-212 (previous versions of the Single Switch) in the database (FGS-212: http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/114)

Any help you be appreciated. 
And I do hope that I catched the right spot where to mention it....

Phil

espentj, 2017-01-21:
Just for your information. This swtich works fine in Openhab2. I run the FGS213 Single Switch there without any issues.

Phil2015, 2017-01-21:
Thank you for the information! 
I have a 1.8 setup and don't want to switch to OH2 until there is a final release...

cdjackson, 2017-01-21:
This issue list is only for OH2 - can you open an issue on OH1 issue and I’ll add it when I get a chance (next week).

closed 2017-01-21 by cdjackson