openhab/org.openhab.binding.zwave #308: Outdated and incomplete NorthQ Q-Gas 9121 configuration

rstets, 2017-01-14:
* OH2 Version: 2.0.0-SNAPSHOT
* Device name: NorthQ Q-Gas 9121
* Link: http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/179

Device configuration parameters in cd-jackson db are based on the manufacturer's doc v2 released in 2013.
There is a doc v3 released in 2016 — http://northq.com/wp-content/uploads/2016/01/Technical-Integration.pdf

There are 2 reasons I'm filing this:
- reason 1 is that I cannot set the initial pulse count for my gas meter due to incorrect "max value" set for config_9_4 param (it's 50, but has to be 2147483647)
- reason 2 is that the gas meter's thing xml has no channel definitions so I see no items from the thing in any of the openhab2 UIs.

rstets, 2017-01-15:
Somehow I've managed to build the binding with IntelliJ Idea and deploy it to my openhab2.
It now works with my NorthQ Q-Gas 9121 and has items. Yay!

`ZWaveNode3NQ9121NorthQGasMeter_GasMeterCubicMeters changed from 30.03 to 30.08`

cdjackson, 2017-01-23:
I guess we need to look at adding a new version of these devices into the database. We can't ignore the old versions as I'm sure people still have these. This means that we probably should create a new version of the device for the newer firmware.

This needs to be done in the database since the database is exported directly into the binding so if we update the binding, any changes will be lost the next time the database is updated.

cdjackson, 2017-01-27:
for reference -:

http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide

rstets, 2017-02-01:
Thank you @cdjackson I'll have a look at the guide and add a new version as soon as I have some free time.

macolu, 2017-03-09:
I have the same device, I had to do the same changes in XML file to make it work.
 @rtsets, thank you so much for the diff! It worked for me as well.

I had to do some dirty XML patching in the binding jar. This will be overwritten at the next update. So it would be great to have DB updated! Is there any way I can help on this point?

cdjackson, 2017-03-09:
> Is there any way I can help on this point?

Sure - if there's anything that needs doing on the database then it would be good to get this updated (see above). If you don't have an account, then please register and email me to update your access.

rstets, 2017-03-26:
@cdjackson Added the nq9121 with firmware v3 to the database — http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/561
requested a review.

cdjackson, 2017-03-26:
Thanks - I saw the updates and will take a look later this evening.

rstets, 2017-03-26:
btw. it's conflicting with the http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/179 — that one has min/max firmware versions set to "all"

cdjackson, 2017-03-26:
Ok - so I guess the other one needs the version set. I guess we can set it to V3 max (or 2.255 possibly - I forget if it’s == or >=).

The other thing is that the endpoints need adding to your new device - this should automatically be done if you upload the XML that OH creates for the device.  This is needed before we can go much further…

rstets, 2017-03-26:
@cdjackson where do I get the XML? I've tried the one I've used to build the binding locally (the one from #309) but I get "ERR2: XML file format is not known, or has incomplete information. You must use openHAB 1.8 (released) or newer" — so I guess it's not the one.

cdjackson, 2017-03-26:
It’s an auto generated XML - OH stores it in the /userdata/zwave folder (note that /userdata might be mapped somewhere else depending on your package installation). It’s a different format than the database file.

rstets, 2017-03-26:
found it. updated the db entry

closed 2017-04-10 by cdjackson

kovacstiborgodollo, 2018-10-13:
I am having similar problems with my NorthQ 9121. I am getting the message "Unknown Device". Could you please advise me, how I could fix it? The firmware in the device seems to be 3.35

rstets, 2018-10-15:
@kovacstiborgodollo you'll have to add a new config to the zwave device db (see the thread above) for 3.35 — my update was for "Firmware Versions | From 3.0 to 3.34"