openhab/org.openhab.binding.zwave #311: [005017 Popp Solar Powered Outdoor Siren] wrongly recognized as [009402 10 Year Smoke Detector with Siren Function]

beckjim, 2017-01-15:
Hello chris,
since commit "Database resynchronisation (#171)" we have in
popp_005107_0_0.xml:      <property name="manufacturerRef">0000:0000</property>
and in 
popp_009402_0_0.xml:      <property name="manufacturerRef">0004:0002</property>
which is wrong.

According to my device and to the

beckjim, 2017-01-15:
sorry accidently submit before finishing ...

According to my device and to http://products.z-wavealliance.org/products/2085 the 005107 should have 
Product Type ID: 0x0004
Product ID: 0x0002 
The 009402 should have accroding to http://products.z-wavealliance.org/products/2078
Product Type ID: 0x0004
Product ID: 0x0004 

thank you.
regards
Stefan

beckjim, 2017-01-15:
Screenshot from HABMIN
[image]

beckjim, 2017-01-22:
Hello Chris,
in commit #171 (https://github.com/openhab/org.openhab.binding.zwave/pull/171/files)
you've changed the device id for the POPP 005071. 
Has there been any special reason for doing this?

Before the commit, the device id was correct and the device got correctly and sucessfully discovered.
This is the crucial part of the commit changes the correct id to wrong ones:

     
     <properties>
       <property name="vendor">POPP</property>
       <property name="modelId">005107</property>
       <property name="manufacturerId">0154</property>
-      <property name="manufacturerRef">0004:0002</property>
+      <property name="manufacturerRef">0000:0000</property>
       <property name="dbReference">367</property>
       <property name="defaultAssociations">1</property>
     </properties>

According to the zwave db at http://products.z-wavealliance.org/Products/2078/XML
the "009402 10 Year Smoke Detector with Siren Function" should have 
  <ManufacturerId>0x0154</ManufacturerId>
  <ProductTypeId>0x0004</ProductTypeId>
  <ProductId>0x0004</ProductId>
rather than 
  <ManufacturerId>0x0154</ManufacturerId>
  <ProductTypeId>0x0004</ProductTypeId>
  <ProductId>0x0002</ProductId>
which belongs to the 005107 outdoor siren (see http://products.z-wavealliance.org/Products/2085/XML).

Thank you!

Regards
Stefan

beckjim, 2017-01-25:
Hello,

I've attached the XML-File generated by openhab2 ...
[node9.zip](https://github.com/openhab/org.openhab.binding.zwave/files/730577/node9.zip)

regards
Stefan

cdjackson, 2017-01-25:
I’m not sure why this was changed, but I have a vague memory that someone swapped it because their device was incorrectly discovered. Possibly there are some incorrect devices out there.

beckjim, 2017-01-25:
Is there anything I can do? Provide any data, logs, ...?

cdjackson, 2017-01-25:
http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/424 <http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/424>

I added the following comment -:

Super User 2016-09-24 08:47
This device has conflicts with the POPP 5107. 1.1 of this device is confirmed as the 9402.

Which version does your device say it is? Maybe we can resolve this with versions - if not, I’m not sure what to do as it looks like the manufacturer has used the same type/id for two devices, which clearly is not good...

beckjim, 2017-01-25:
I assume you're referring to firmware version ...
Firmware version is 1.3 (as in screenshot above). 

This is from the XML I've already uploaded:
     <commandClass>VERSION</commandClass>
      <versionCommandClass>
        <version>2</version>
        <instances>1</instances>
        <versionSupported>2</versionSupported>
        <libraryType>LIB_SLAVE_ENHANCED</libraryType>
        <protocolVersion>4.5</protocolVersion>
        <applicationVersion>1.3</applicationVersion>
        <hardwareVersion>1</hardwareVersion>
      </versionCommandClass>
    </entry>

Maybe the IDs have changed since version 1.1 from September.

cdjackson, 2017-01-25:
Yes - app version is what I was looking for.

What do you mean that the IDs might have changed since September? They can’t change can they?

beckjim, 2017-01-25:
I mean, if somebody had a device in September with firmware 1.1., that device provided different IDs than mine with firmware 1.3  ... just trying to find an explanation ...
(I assume that device IDs are coded in the firmware.)

My device has been correctly discovered back in August '16, when I made my very first steps with zwave/openhab2. (I haven't had much time since then, so I gave openhab another try in Januar with nightly builds ...) 

(Just sent you a 'virtual' beer via paypal, assuming you're still on holiday ...)

cdjackson, 2017-01-25:
But the basic problem is that the device IDs are the same - your device has the same ID as the other device. Back in August when this first came up we changed your device to 0000:0000 and added the comment since it has the same IDs.  I think what we’ll do is to use the version to differentiate the two devices - should work ok.

Thanks for the beer - I actually got back from holiday Monday night so trying to catch up properly now ;)

beckjim, 2017-01-25:
Yes, including the version would help, at least for now (if discovery can include the version for device lookup).
Thanks a lot ...

cdjackson, 2017-01-25:
Yes - discovery uses the version so this is simple (and I’ve already done it). I’ll update the binding tonight hopefully….

beckjim, 2017-01-25:
Perfect.

cdjackson, 2017-01-29:
Any update on this?

beckjim, 2017-01-30:
Hi Chris, yes. Super. Works as expected!
(sorry for the late reply ...)
Thanks a lot!

closed 2017-01-30 by beckjim

haltmann, 2017-03-07:
Hello,

my Solar Outdoor Siren is with Firmware-Version 1.1 and therfor found as Smokedetector with Siren. What can i do, that it is as the right Thing detected?