openhab/org.openhab.binding.zwave #312: CRC errors

Edgesystems, 2017-01-16:
I'm having problems with dimmer changes (turned on by switch) not updating the GUI. Checked the log and gets a lot of CRC errors, this works on normal switches (Fibaro wall plug). The dimmer is a "FGD212 Dimmer 2" but I have some Qubinos that doesn't work either.

21:06:08.977 [ERROR] [.ZWaveCRC16EncapsulationCommandClass] - NODE 4: CRC check failed message contains E0 BC  but should be E0 00
21:06:09.702 [ERROR] [.ZWaveCRC16EncapsulationCommandClass] - NODE 4: CRC check failed message contains 77 BE  but should be 77 00
21:06:14.700 [ERROR] [.ZWaveCRC16EncapsulationCommandClass] - NODE 4: CRC check failed message contains 17 C1  but should be 17 00

what could it be? Feels like some kind of bug to me.

cdjackson, 2017-01-17:
Please provide a drug log - it's impossible to debug without this.

Edgesystems, 2017-01-17:
Drug?

cdjackson, 2017-01-17:
Sorry - autocorrect. Should be debug...

Edgesystems, 2017-01-18:
This is the output when I turn the dimmer on via the switch.

```
2017-01-18 18:56:46.865 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0E 00 04 00 04 07 56 01 26 03 19 BD E0 CF 0F
2017-01-18 18:56:46.867 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
2017-01-18 18:56:46.869 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0E 00 04 00 04 07 56 01 26 03 19 BD E0 CF 0F
2017-01-18 18:56:46.870 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 0E 00 04 00 04 07 56 01 26 03 19 BD E0 CF 0F
2017-01-18 18:56:46.871 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 04 07 56 01 26 03 19 BD E0 CF
2017-01-18 18:56:46.872 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 4: Application Command Request (ALIVE:DYNAMIC_VALUES)
2017-01-18 18:56:46.873 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 4: Incoming command class CRC_16_ENCAP
2017-01-18 18:56:46.873 [DEBUG] [.ZWaveCRC16EncapsulationCommandClass] - NODE 4: Received CRC 16 Encapsulation Request
2017-01-18 18:56:46.874 [ERROR] [.ZWaveCRC16EncapsulationCommandClass] - NODE 4: CRC check failed message contains E0 CF  but should be E0 00
2017-01-18 18:56:46.875 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: Sent Message: class=SendData[0x13], type=Request[0x00], priority=Get, dest=4, callback=167, payload=04 03 71 04 08
2017-01-18 18:56:46.877 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: Recv Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 04 07 56 01 26 03 19 BD E0 CF
2017-01-18 18:56:46.880 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: class=ApplicationCommandHandler, callback id=167, expected=ApplicationCommandHandler, cancelled=false        transaction complete!
2017-01-18 18:56:46.881 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveTransactionCompletedEvent
2017-01-18 18:56:46.883 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 4: Node advancer - DYNAMIC_VALUES: Transaction complete (SendData:Request) success(true)
2017-01-18 18:56:46.885 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 4: Node advancer - checking initialisation queue. Queue size 16.
2017-01-18 18:56:46.887 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 4: Node advancer - message removed from queue. Queue size 15.
2017-01-18 18:56:46.889 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 4: Node advancer - DYNAMIC_VALUES: queue length(15), free to send(true)
2017-01-18 18:56:46.889 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 4: Initialisation retry timer started 80000
2017-01-18 18:56:46.890 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 4: Node advancer - queued packet. Queue length is 15
2017-01-18 18:56:46.891 [DEBUG] [ocol.ZWaveController$ZWaveSendThread] - Took message from queue for sending. Queue length = 0
2017-01-18 18:56:46.892 [DEBUG] [ve.internal.protocol.ZWaveController] - Message queued. Queue length = 0. Queue={}
2017-01-18 18:56:46.892 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0A 00 13 04 03 71 04 09 25 A8 10
2017-01-18 18:56:46.893 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 4: Sending REQUEST Message = 01 0A 00 13 04 03 71 04 09 25 A8 10
2017-01-18 18:56:46.894 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Got an event from Z-Wave network: ZWaveTransactionCompletedEvent
2017-01-18 18:56:46.900 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 04 01 13 01 E8
2017-01-18 18:56:46.906 [DEBUG] [ocol.ZWaveController$ZWaveSendThread] - NODE 4: Response processed after 0ms/4677ms.
2017-01-18 18:56:46.907 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
2017-01-18 18:56:46.909 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 04 01 13 01 E8
2017-01-18 18:56:46.910 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 04 01 13 01 E8
2017-01-18 18:56:46.911 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=SendData[0x13], type=Response[0x01], priority=High, dest=255, callback=0, payload=01
2017-01-18 18:56:46.913 [DEBUG] [l.serialmessage.SendDataMessageClass] - NODE 4: Sent Data successfully placed on stack.
2017-01-18 18:56:47.311 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 11 00 04 00 04 0A 56 01 31 05 04 22 00 08 F8 D3 CF 4D
2017-01-18 18:56:47.314 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
2017-01-18 18:56:47.317 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 11 00 04 00 04 0A 56 01 31 05 04 22 00 08 F8 D3 CF 4D
2017-01-18 18:56:47.318 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 11 00 04 00 04 0A 56 01 31 05 04 22 00 08 F8 D3 CF 4D
2017-01-18 18:56:47.320 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 04 0A 56 01 31 05 04 22 00 08 F8 D3 CF
2017-01-18 18:56:47.321 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 4: Application Command Request (ALIVE:DYNAMIC_VALUES)
[... cut ...]
2017-01-18 18:56:48.998 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: Process NOTIFICATION_REPORT V5
2017-01-18 18:56:49.000 [ERROR] [ssage.ApplicationCommandMessageClass] - Error processing frame: Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 04 04 71 05 00 00 CF  >> Attempt to read message payload out of bounds: Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 04 04 71 05 00 00 CF  (8)

```

cdjackson, 2017-01-21:
Do you know if this is decoded ok with other software? I ran some tests here and it seems the decoder is working ok so I don't understand what these packets are comprised of at the moment... I've tried a number of things to get the CRC to match (short packet, long packet) and they don't work.

There are some other very strange messages in your log - what controller are you using?

Edgesystems, 2017-01-21:
Pine64 with their controller. One thing I noticed is that the expected crc always ends with 00. Seems strange.

cdjackson, 2017-01-22:
Yes - I agree that this is strange and it made me think it's a bug in the binding but I can't see any problem when I run it through the debugger - it seems to be correctly processing the data but just getting the wrong result. 

The controller is returning some frames I've not seen before so it seems to be running some custom firmware or maybe pine are using the zip version of firmware. 

Have you tried these devices with another controller to see if they are decided there?

I will have another look when I get a chance - maybe tonight.

cdjackson, 2017-01-22:
Ok - I know what’s wrong. Your controller is adding extra bytes to the frame and the implementation in the binding doesn’t check the length. I’ll try and fix this tonight (indian time!)

closed 2017-01-22 by cdjackson

Edgesystems, 2017-01-22:
You're awesome!

maschenborn, 2017-02-26:
I think I may have the same problem with Pine64 and Fibaro Button:

`08:32:23.693 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 18: Application Command Request (ALIVE:SET_ASSOCIATION)
08:32:23.694 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 18: Incoming command class CRC_16_ENCAP
08:32:23.695 [DEBUG] [.ZWaveCRC16EncapsulationCommandClass] - NODE 18: Received CRC 16 Encapsulation Request
08:32:23.696 [ERROR] [.ZWaveCRC16EncapsulationCommandClass] - NODE 18: CRC check failed message contains FA BC  but should be FA 00
08:32:23.697 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 255: Transaction not completed: node address inconsistent.  lastSent=255, incoming=255
08:32:26.238 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 10 00 04 00 12 09 56 01 5B 03 1E 80 01 27 AA BC 51
08:32:26.244 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
08:32:26.247 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 10 00 04 00 12 09 56 01 5B 03 1E 80 01 27 AA BC 51
08:32:26.249 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 10 00 04 00 12 09 56 01 5B 03 1E 80 01 27 AA BC 51
08:32:26.253 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 12 09 56 01 5B 03 1E 80 01 27 AA BC
08:32:26.255 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 18: Application Command Request (ALIVE:SET_ASSOCIATION)
08:32:26.257 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 18: Incoming command class CRC_16_ENCAP
08:32:26.259 [DEBUG] [.ZWaveCRC16EncapsulationCommandClass] - NODE 18: Received CRC 16 Encapsulation Request
08:32:26.262 [ERROR] [.ZWaveCRC16EncapsulationCommandClass] - NODE 18: CRC check failed message contains AA BC  but should be AA 00
08:32:26.264 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 255: Transaction not completed: node address inconsistent.  lastSent=255, incoming=255
08:32:27.748 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 10 00 04 00 12 09 56 01 5B 03 1F 80 01 10 9A BC 57
08:32:27.753 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
08:32:27.757 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 10 00 04 00 12 09 56 01 5B 03 1F 80 01 10 9A BC 57
08:32:27.760 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 10 00 04 00 12 09 56 01 5B 03 1F 80 01 10 9A BC 57
08:32:27.764 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 12 09 56 01 5B 03 1F 80 01 10 9A BC
08:32:27.766 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 18: Application Command Request (ALIVE:SET_ASSOCIATION)
08:32:27.768 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 18: Incoming command class CRC_16_ENCAP
08:32:27.770 [DEBUG] [.ZWaveCRC16EncapsulationCommandClass] - NODE 18: Received CRC 16 Encapsulation Request
08:32:27.772 [ERROR] [.ZWaveCRC16EncapsulationCommandClass] - NODE 18: CRC check failed message contains 9A BC  but should be 9A 00
08:32:27.774 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 255: Transaction not completed: node address inconsistent.  lastSent=255, incoming=255
`

cdjackson, 2017-02-26:
This issue was fixed - what version of the binding are you using?

Edgesystems, 2017-02-26:
Manually installed 2.1

cdjackson, 2017-02-26:
@Edgesystems is it working for you or not?

Edgesystems, 2017-02-26:
Yes it works. Everything runs more smoothly too, better response time etc.

maschenborn, 2017-02-27:
I use binding-zwave - 2.0.0 fresh install with openhabian.
I thought, it always fetches the newest version?!?

cdjackson, 2017-02-27:
If you are using the stable version, then you will be using the binding from January and I don't think this fix was added as this was fixed 1 day before it was released.

maschenborn, 2017-03-08:
Hi Chris,

I have these two identical Wall Plugs.
One is working - and the other does not work.

Could you add the ID of the unknown device in the db?
(I wrote this before - but without screenshot - maybe you just did that?)

Thank you

Michael

08:53 Uhr:

Edgesystems, 2017-03-08:
You can do it your self. Request access to the db and upload the xml file.

maschenborn, 2017-03-08:
I'm already loggedin in cd-jackson - but I have no access. Where can I request the access?

cdjackson, 2017-03-08:
I’ve updated your access.

PR #319 (2017-01-22): Add length check to CRC command class
Fixes #312