openhab/org.openhab.binding.zwave #313: [zwave-binding] Add new device: Popp Z-Weather

moonwhaler, 2017-01-19:
Hello,

please add support for the Device Popp Z-Weather to the database, It's already there, but it seems (even while using the latest (apt) nightly of openhab 2 + Zwave binding), mine isn't recognised:

0154:0100:0400:1.0 (Unknown device)

http://products.z-wavealliance.org/products/2102

Manufacturer: 0154 (Popp & Co.)
Type: 0100
ID: 0400

This was tested again with a brand new setup including the new apt-repository, but although other devices are being added sucessfully, this one is not.

If you need something else, please tell me.
Thank you!

beckjim, 2017-01-19:
Hello,

fyi. 
it seams that some IDs for the POPP devices are wrong.
(e.g. my issue [005017 Popp Solar Powered Outdoor Siren] wrongly recognized as [009402 10 Year Smoke Detector with Siren Function] )

Your device is currently known in the zwave binding db with the following ids. So it is obvious that it can't be identified correctly.

    
    <properties>
      <property name="vendor">Bönig und Kallenbach oHG</property>
      <property name="modelId">POPE005206</property>
      <property name="manufacturerId">0169</property>
      <property name="manufacturerRef">0001:0001</property>
      <property name="dbReference">234</property>
      <property name="defaultAssociations">1</property>
    </properties>

best regards
Stefan

cdjackson, 2017-01-20:
If you can help to provide better information I'm sure it would be appreciated by those with these devices.

moonwhaler, 2017-01-20:
I suppose I am not able to extract more information than HABmin / PaperUI gives me. Maybe you could check whether the data stored in the DB is the same as mentioned by me?

cdjackson, 2017-01-20:
Sorry - my response was in response to the comment from @beckjim who said "it seams that some IDs for the POPP devices are wrong”.

I will check the weather ids but it might be next week as I'm on holiday in India at the moment. Of course you can also check at www.cd-jackson.com :)

moonwhaler, 2017-01-20:
The manufacturer ID seems to be alright (@ your site), the URL is "http://www.popp.eu/".
As far as I can see the device (my first post) is missing.

cdjackson, 2017-01-20:
I don't think the device is missing ...

http://cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/234

Can you confirm this is the same as you have?  The type /I'd is different but that's not uncommon for manufacturers to use multiple sets of values.

moonwhaler, 2017-01-20:
It sounds very much like the Popp device. Although this one seems to be 
from another manufacturer. Could be they're reselling it under a new name.

cdjackson, 2017-01-21:
Quite likely but what manufacturer do you mean - the link is to the popp device isn't it?

moonwhaler, 2017-01-21:
No the manufacturer is called 'Bönig und Kallenbach oHG' following your 
URL. Which is different than Popp. The device seems similar, though.

cdjackson, 2017-01-21:
Why do you say it is different. The manufacturer names and Ids come from the official Wave list and this is how most or all of the popp devices are listed  

When I get a chance I'll check the info once provided.

moonwhaler, 2017-01-21:
My device is being sold by a company called Popp - this is what's printed 
on the packaging. That's why I wrote that Popp may be a reseller of said 
device you linked.

cdjackson, 2017-01-21:
It’s really not relevant why is printed on the packaging. The packaging doesn’t mean anything and we often see devices being repackaged. What matters is the IDs in the device. The ID that we used to call POPP is now called Bönig und Kallenbach oHG as that’s what the ZWave spec calls it. The weather device I linked earlier also is sold with POPP on the packaging! (http://www.zwave-shopping.com/en/sensors/popp/4416-popp-z-wave-weatherstation-z-weather-pope005206.html <http://www.zwave-shopping.com/en/sensors/popp/4416-popp-z-wave-weatherstation-z-weather-pope005206.html>) - the POPP 5206 weather station which I think is what you have?.

I suspect that these are the same device, but POPP might now have their own ID. Can you please compare the configuration of your device with the parameters in the device I linked earlier and confirm if they are the same. If so we can use the same configuration. Please also provide the XML for this node (in the userdata/zwve folder).

moonwhaler, 2017-01-21:
Yes. That is exactly the device I bought.
I've attached the XML file you requested.

[POPP_ZWEATHER_XML.zip](https://github.com/openhab/org.openhab.binding.zwave/files/721234/POPP_ZWEATHER_XML.zip)

cdjackson, 2017-01-21:
Thanks - do you also confirm that the configuration in the database is the same?

moonwhaler, 2017-01-21:
Based on the XML file?

cdjackson, 2017-01-21:
No - just comparing what’s in the database on the link I attached earlier, and what’s in the manual that you have for your device. I’m just interested in the associations and parameters.

moonwhaler, 2017-01-21:
Actually, there was no manual except maybe how to switch the device into 
pairing mode, but I will try to look if it's at least the same at some 
basic level.

sennerholm, 2017-01-21:
I also have the same popp device (xml), and it's called Z-Weather (005206) from the reseller, so it should be the same. Is it an easy way to try?

Sincerely
Mikael

cdjackson, 2017-01-22:
I prefer to be sure if possible. I think it's probably the same but i don't want to wast time editing the database if it's wrong.

sennerholm, 2017-01-22:
Is it possible to build the binding locally with an add on to the database?
Then I can try it out.

Sincerely Mikael

cdjackson, 2017-01-22:
Sure, you can build he binding yourself but then you'd also need to build the XML manually as well unless we add this to the database....

Is there no manual online with any parameters/ configuration?  If not maybe we just have to assume it's the same as the other device.

sennerholm, 2017-01-22:
I think it's the same as the other. So I copy
https://github.com/openhab/org.openhab.binding.zwave/blob/master/ESH-INF/thing/zwaveme_zweather_0_0.xml
And change the ID.

Then I build the mvn project. Copy the artifacts to the server and restart.

What do I need to trigger the new code? Is heal device enought?

Sincerely Mikael

sennerholm, 2017-01-23:
Hi!

When i place the snapshot jar in the addons directory openhab2 doesn't
found it (at least is writing the rc1 candidate in the ui), how should I
version it so it understand it's newer?

I tried to change the manufactor in the local xml fil and restarted
Openhab2 and then it recognize the device and most of the values seems ok.

2017-01-22 22:15:41.719 [ThingUpdatedEvent         ] - Thing
'zwave:device:8c5ef8b6:node40' has been updated.
2017-01-22 22:15:41.802 [ItemStateChangedEvent     ] -
zwave_device_8c5ef8b6_node40_sensor_dewpoint changed from NULL to 1
2017-01-22 22:15:41.875 [ItemStateChangedEvent     ] -
zwave_device_8c5ef8b6_node40_sensor_relhumidity changed from NULL to 91
2017-01-22 22:15:41.962 [ItemStateChangedEvent     ] -
zwave_device_8c5ef8b6_node40_sensor_barpressure changed from NULL to 101.8
2017-01-22 22:15:42.063 [ItemStateChangedEvent     ] -
zwave_device_8c5ef8b6_node40_sensor_temperature changed from NULL to 2.3
2017-01-22 22:15:42.141 [ItemStateChangedEvent     ] -
zwave_device_8c5ef8b6_node40_sensor_velocity changed from NULL to 0
2017-01-22 22:15:42.225 [ItemStateChangedEvent     ] -
zwave_device_8c5ef8b6_node40_sensor_luminance changed from NULL to 0
2017-01-22 22:15:42.296 [ItemStateChangedEvent     ] -
zwave_device_8c5ef8b6_node40_battery_level changed from NULL to 81
2017-01-22 22:15:42.375 [ItemStateChangedEvent     ] -
zwave_device_8c5ef8b6_node40_meter_kwh changed from NULL to 1.06251
2017-01-22 22:15:42.473 [ItemStateChangedEvent     ] -
zwave_device_8c5ef8b6_node40_meter_pulse changed from NULL to 21846

The barpressure need to move the decimal ., either to 1018 or 1.018
depending on whats most common. I will try to mount it this evening and see
if I can get some velocity from it on tuesday.

Sincerely
Mikael

2017-01-22 13:24 GMT+01:00 Mikael Sennerholm <mikael@sennerholm.net>:

> I think it's the same as the other. So I copy
> https://github.com/openhab/org.openhab.binding.zwave/
> blob/master/ESH-INF/thing/zwaveme_zweather_0_0.xml
> And change the ID.
>
> Then I build the mvn project. Copy the artifacts to the server and restart.
>
> What do I need to trigger the new code? Is heal device enought?
>
> Sincerely Mikael
>
>
>> Sure, you can build he binding yourself but then you'd also need to build
>> the XML manually as well unless we add this to the database....
>>
>> Is there no manual online with any parameters/ configuration? If not
>> maybe we just have to assume it's the same as the other device.
>>
>> Sent from my iPhone
>>
>> >
>> > Is it possible to build the binding locally with an add on to the
>> database?
>> > Then I can try it out.
>> >
>> > Sincerely Mikael
>> >
>> > Den 22 jan. 2017 02:33 skrev "Chris Jackson" <notifications@github.com
>> >:
>> >
>> > > I prefer to be sure if possible. I think it's probably the same but i
>> > > don't want to wast time editing the database if it's wrong.
>> > >
>> > > Sent from my iPhone
>> > >
>> > > > On 22 Jan 2017, at 04:04, Mikael Sennerholm <
>> notifications@github.com>
>> > > >
>> > > > I also have the same popp device (xml), and it's called Z-Weather
>> > > (005206) from the reseller, so it should be the same. Is it an easy
>> way to
>> > > try?
>> > > >
>> > > > Sincerely
>> > > > Mikael
>> > > >
>> > > >
>> > >
>> > > <https://github.com/openhab/org.openhab.binding.zwave/issues
>> /313#issuecomment-274301018>,
>> ADUvbbD9TidXM1H8feiBcBZ_f7WKZFkGks5rUrHZgaJpZM4LoH6H>

cdjackson, 2017-01-23:
You will need to uninstall the old binding using Karaf before you can add a new binding into addons folder.

> The barpressure need to move the decimal ., either to 1018 or 1.018

The device specifies the precision in the zwave message, so if this is wrong, then it probably can’t be fixed in the binding - I’d need to see a debug log its the actual received message to be sure though.

moonwhaler, 2017-01-23:
So after checking everything this seems very much to be the exact same device, but to be 100% sure, we should wait for Mikaels proof of that assumption. :+1:

sennerholm, 2017-01-23:
Hi

I have now mounted the device and get some wind values from it. I'm not
sure if the unit is m/s or knot/s but probably m/s.

So I vote for adding the device so more people can try it out.

Sincerely Mikael

cdjackson, 2017-01-29:
I've created a new device that's the same as the old one. Please let me know if it's ok.

sennerholm, 2017-01-30:
Hi

To tired to try today, will give it a try tomorrow. I will try to follow
the last comment in:
https://community.openhab.org/t/how-to-install-a-snapshot-version-of-a-binding-in-a-oh2b4-installation/14254/7

(but change from snapshot-2.0 to 2.1)

Sincerely
Mikael

moonwhaler, 2017-02-01:
Updated to the latest nightly (2.1). The device was found immediately, added - but no values get written to it "NaN". Since the device may be underpowered (battery, since it's currently placed in my home), I will have to check this again.

sennerholm, 2017-02-01:
I needed to have mine outside and after some hours in sun I pressed the
awake button.

Sincerely Mikael

moonwhaler, 2017-02-14:
It seems to be working now as intended. The internal "power saving mechanism" drives me bonkers sometimes (connection lost at night etc.), but at least some values are reported.

Here's the manual in case Chris wants to validate this device once more: 
http://www.benext.eu/static/manual/popp/z-weather-005206.pdf

cdjackson, 2017-02-14:
> Would be great to manually change some value offsets

If the device doesn't provide this function, then it won't be done in the binding. 

If you can, can you upload the manual to the database (there's a documents section at the bottom of the page).

I'll close this now - thanks for the testing.

closed 2017-02-14 by cdjackson

moonwhaler, 2017-02-14:
Hi Chris,

not able to find the doc upload on your page (http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/512). Do you have a direct url?

Thanks.

cdjackson, 2017-02-14:
It's toward the bottom - where it says "Reference Documents" - you need to be logged in to see the green + button to upload though.

Thanks.

moonwhaler, 2017-02-14:
I've registered an account, but still don't see any "+" button where it says "Reference Documents".

cdjackson, 2017-02-14:
I need to edit access - I'll do this tonight....

cdjackson, 2017-02-14:
I’ve updated your access, so you should be all set - thanks.