openhab/org.openhab.binding.zwave #316: [zway] Z-Wave ZHC5010 not recognized in OH2 RC1 in zway binding

rhjensen79, 2017-01-20:
Z-Wave ZHC5010 is not recognized in OH2 RC1.
The version used is : ZHC5010-WH-4L
It should already be present in the z-wave database, but it's not recognized, by some strange reason.
Links : https://youcontrol.dk/media/wysiwyg/ZHC5010_datablad_1.pdf
Information from OH2.
￼
[image]

cdjackson, 2017-01-20:
If this is for the zway binding (rather than the zwave binding) then it probably needs to be in the main OH2 addons issues since zway doesn’t use this database.

closed 2017-01-20 by cdjackson

rhjensen79, 2017-01-20:
Why did you close it ?
It's not for the way binding. I'm using the zwave binding, so i don't understand, why this is not the right place to place the issue ?

cdjackson, 2017-01-21:
Ok - I closed it because you said "zway binding". The zway binding is a different binding. I've just closed your latest one for the same reason - if you are using the zwave binding then please just reopen it (but please change the title so it's clear).