openhab/org.openhab.binding.zwave #317: [zwave] Z-Wave ZHC5010 not recognized in OH2 RC1 in zway binding

rhjensen79, 2017-01-21:
Z-Wave ZHC5010 is not recognized in OH2 RC1.
The version used is : ZHC5010-WH-4L
It should already be present in the z-wave database, but it's not recognized, by some strange reason.
Links : https://youcontrol.dk/media/wysiwyg/ZHC5010_datablad_1.pdf
Information from OH2.

[image]

cdjackson, 2017-01-21:
Please open this in the openhab2 issues list. This issue list is only for the zwave binding.

cdjackson, 2017-01-21:
Title updated to remove reference to zway binding.

rhjensen79, 2017-01-21:
Thanks. It was autocorrect, that changed the name, and i did not see it until i was to late.

cdjackson, 2017-01-21:
I've updated the database to add your device references. I won't update the binding until next week though.

I would also suggest to reinitialise the device (in HABmin, select the thing and in the tools menu select the reinitialise option). There's clearly something wrong as half the information displayed in your image is invalid.

rhjensen79, 2017-01-21:
Thanks.

I’m away traveling, so i might not be able to test, it for the next 14 days. But i will test it as soon as i’m back, and the binding is updated.

Thanks for your help.

/Robert

cdjackson, 2017-01-21:
That’s fine - I’m also away on holiday right now :)

cdjackson, 2017-01-27:
This should now be resolved so closing. If it's still not working then please reopen.

rhjensen79, 2017-02-04:
Sorry it's still not being detected. 

Do i need to do anything special, for the binding to be updated with the ned DB information, other than uninstall and reinstall binding ?

[image]

[image]

Thanks.

cdjackson, 2017-02-04:
> Do i need to do anything special, for the binding to be updated

Yes - you need to be running the snapshot version, and you need to install the latest version. It won't update by itself (ie the bindng doesn't read direct from the database - you need to update).

Please discuss on the forum if you're not sure about how to update your system.

rhjensen79, 2017-02-06:
It seams to be fixed, both by my test, and by others.
link : https://community.openhab.org/t/logicsoft-zhc5010-z-wave-switch-not-recognized-in-openhab/12160/61

Closing issue.

closed 2017-02-06 by rhjensen79