openhab/org.openhab.binding.zwave #318: POPP battery indicator is not shown correctly

espentj, 2017-01-21:
I have a POPP smoke sensor:
Device discovery resolved to thingType zwave:popp_004001_00_000
that is not able to correctly display it´s battery status. Note I have 10 of this device and all are exhibiting the same behaviour.

I am wondering if the definition for the battery sensors values are being erronously translated somehow, but I don´t quite know how to investigate further.
One of the devices is NODE 8, and I have included the debug log here. The complete log is also included.

[POPP-Battery-error-logfiles.zip](https://github.com/openhab/org.openhab.binding.zwave/files/721210/POPP-Battery-error-logfiles.zip)

The problem has been there for as long as I have run Openhab2 (august 2016), and I am currently running the latest SNAPSHOT as of January 21 - 2017.

The XML file for node 8 is also included:  [node8.xml.zip](https://github.com/openhab/org.openhab.binding.zwave/files/721211/node8.xml.zip)

cdjackson, 2017-01-27:
You don't seem to say what is not correct - they all exhibit the "same behavior" - but WHAT behavior?

From you log, it looks fine -:

```
2017-01-21 10:57:02.659 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 0A 03 80 03 64 
2017-01-21 10:57:02.664 [DEBUG] [ommandclass.ZWaveBatteryCommandClass] - NODE 10: Battery report value = 100
2017-01-21 10:57:02.665 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveCommandClassValueEvent
2017-01-21 10:57:02.666 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 10: Got an event from Z-Wave network: ZWaveCommandClassValueEvent
2017-01-21 10:57:02.667 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 10: Got a value event from Z-Wave network, endpoint = 0, command class = BATTERY, value = 100
2017-01-21 10:57:02.668 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 10: Updating channel state zwave:device:0216d7a2:node10:battery-level to 100 [DecimalType]
```

The above log shows that the received data from the device is a value of 64 (ie 100%). This is decoded and the channel is updated. I don't see any problem here. 

Maybe it's worth discussing with others on the forum.

closed 2017-01-27 by cdjackson

espentj, 2017-01-27:
Sorry, the error is that the smoke sensor always displays 100% as the value for remaining battery. Even when it is close to depleted. It could however be due to an error in the device itself (but then the error is exihibited on all devices).

cdjackson, 2017-01-27:
Ok - the log shows the device is reporting 100%, and the binding is showing this, so from my perspective the binding is correctly interpretting the data and presenting it to the system. If it's never changing then I would say it's a device problem.