openhab/org.openhab.binding.zwave #320: Channel config restore previous dimmer lost

marcelrv, 2017-01-22:
Recently the restore previous dimmer values were introduced.
https://github.com/openhab/org.openhab.binding.zwave/pull/279

currently However in paperUI & habmin this channel confguration option seems to be no longer available

* OH2 Version: RC1

cdjackson, 2017-01-22:
It’s still included in the channel configuration -:

            <parameter name="config_restoreLastValue" type="boolean">
                <label>Restore Last Value</label>
                <description>Restore Last Dimming level on ON.</description>
                <default>true</default>
                <options>
                    <option value="true">Restore Last Value</option>
                    <option value="false">Restore Full Brightness</option>
                </options>
            </parameter>

Note that the name might have changed.

marcelrv, 2017-01-22:
yes, I've checked that before submitting...

but in the UI option is not there (also not in habmin)
[image]

marcelrv, 2017-01-24:
Was digging bit deeper..
http://192.168.3.50:8080/rest/channel-types/zwave%3Aswitch_dimmer
returns no parameters, hence it is not due to the UI's that the option for config_restoreLastValue is not available

```
{
  "parameters": [],
  "parameterGroups": [],
  "description": "The brightness channel allows to control the brightness of a light. It is also possible to switch the light on and off.",
  "label": "Brightness",
  "category": "DimmableLight",
  "itemType": "Dimmer",
  "kind": "STATE",
  "tags": [],
  "UID": "zwave:switch_dimmer"
}
```

cdjackson, 2017-01-28:
I'm not sure where to go with this - I suspect that it must be an ESH issue as the binding doesn't do anything with these channel configs - they are (I think) served up directly from the XML configuration system.

Do you have any channel configs working in other bindings?

marcelrv, 2017-01-28:
I wasn't sure where to report this.
So indeed I checked if it is still working for other bindings and this is the case.

So it is something specific.
One of the suspects I have is that the combination of the dynamic config and channel config is behind this

marcelrv, 2017-01-28:
From the recent PRs this is one that could be relevant
https://github.com/eclipse/smarthome/pull/2759

cdjackson, 2017-01-28:
Thanks. I’m not really sure why the channel config should be linked to the thing config - however, given it seems to be zwave specific, it’s as good a starting point as any ;)

I’ll see what I can find.

cdjackson, 2017-01-28:
It’s working ok in my development environment which is probably a couple of weeks out of date due to me being on holiday. I’ll get everything up to date and see where we end up…..

cdjackson, 2017-01-28:
It still works ok in my dev environment after upgrading all repos, target environment etc… I also did a complete reinstall of OH2 release version and it also works fine - both in PaperUI and HABmin.

I don’t know what else to do - it seems to work fine to me.

marcelrv, 2017-01-28:
thanks very much for checking.
I'm still on the last beta. Gonna update to the release and test again.

marcelrv, 2017-01-28:
Upgraded to latest snapshot and its working again. So probaly something specific to the rc release.
Thanks for checking.

closed 2017-01-28 by marcelrv