openhab/org.openhab.binding.zwave #321: Aeon ZW096 timeouts on param 33

mhilbush, 2017-01-22:
On Jan 15 in this commit, you added parameter 33 to the Aeon ZW096.  I just installed the most current zwave build and I'm now seeing timeouts during initialization when trying to get param 33 on all 4 of my ZW096 devices.

https://github.com/openhab/org.openhab.binding.zwave/commit/10ef8dd4e444e01c15192e3e29350424a2eebce2#diff-2e1986c9d77a03c7bb9029d5d75dac79

I do see this parameter referenced in the Aeon ZW096 spec, although, oddly, it says there is no default value.  If there's no default value, is it possible that you're unable to get the value of this parameter?
[image]

My ZW096's are running version 1.3.

mhilbush, 2017-01-22:
Hmm.  I see there's a write-only checkbox in the "edit parameter" section of the database editor.  Would this be an appropriate use of that checkbox for parameter 33?

cdjackson, 2017-01-23:
If you send one of these values to it, does it work?  If so, then let’s use the write only option. Otherwise let’s delete it… (I didn’t add this so I’d need to check who did).

mhilbush, 2017-01-23:
I'm not entirely sure what's supposed to happen when sending it a value.  It's defined as a 4 byte word, which looks like it should contain RGB values (0-255) in bytes 2-4?  Nevertheless, no matter what integer value I put in the field, it generates this single log statement.
```
2017-01-23 07:50:58.705 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 30: Configuration update received
```

I'd be ok with either deleting it, or making it write only.

mhilbush, 2017-01-26:
I marked param 33 write-only in the db for the time being.

cdjackson, 2017-01-26:
I've updated the binding with this...

cdjackson, 2017-01-29:
Can we close this one?

closed 2017-01-29 by mhilbush