openhab/org.openhab.binding.zwave #323: Trying to initialize a new node after cancelling inclusion process

pizerg, 2017-01-25:
AddNodeMessageClass wrongly initializes a new node when the Inclusion is cancelled via  "doRequestStop" as (at least with razberry controller) the serial message when successfully including a new node and when cancelling is the same -> bytes [0x01, 0x06, 0xNEW_NODE_ID, 0x00].

Haven't tested with RemoveNodeMessageClass but might have the same issue.

cdjackson, 2017-01-25:
Which branch is this in? The new branch?

pizerg, 2017-01-26:
Both

cdjackson, 2017-01-26:
Ok, thanks - is it trying to initialise a node 0? I thought I'd added a check for this, but maybe not in this instance.

pizerg, 2017-01-27:
It tries to initialize whichever node would be added next, as [0xNEW_NODE_ID] returns the node id that would have been assigned. The protection for node 0 is already on the code

cdjackson, 2017-01-28:
Do you have a log file showing the frame flow? I’m not exactly sure what the  [0x01, 0x06, 0xNEW_NODE_ID, 0x00] relates to - is this the content of the AddNode message? So the 0x06 is the DONE message?

pizerg, 2017-01-30:
Sorry I was not clear enough, I do not have the logs right now but if it's neccessary I can generate an example again. The bytes I mentioned come from 
`public boolean handleRequest(ZWaveController zController, SerialMessage lastSentMessage, SerialMessage incomingMessage)` 

and correspond to
```
incomingMessage.getMessagePayloadByte(0)
incomingMessage.getMessagePayloadByte(1)
incomingMessage.getMessagePayloadByte(2)
incomingMessage.getMessagePayloadByte(3)
```
So yes, the 0x06 is the DONE message and the byte 0xNEW_NODE_ID carries the next "available" node id even if the process was cancelled.

cdjackson, 2018-09-08:
Closing as this is very old and the inclusion code was completely rewritten.

PR #787 (2018-01-02): Refactor inclusion code into separate class for readability and testing
This is mainly a refactoring of the inclusion/exclusion code to remove it from the ```ZWaveController``` class into its own class for testability and readability! It also addresses #323 by not using the node number in the final call - only in the "adding node" response.

A controller priority queue has been added to the transaction manager to ensure that high priority controller messages can be sent immediately. This prevents inclusion commands that might be time critical from getting lost behind other traffic.

Closes #32 [... cut]
closed 2018-09-08 by cdjackson