openhab/org.openhab.binding.zwave #331: Fibaro Smoke Alarm triggers alarm_general, not alarm_smoke!

euphi, 2017-01-28:
If my Fibaro Smoke detector triggers a Smoke Alarm, this results in the channel alarm_general to be set to ON. However, it shall be the channel alarm_smoke.

* OH2 Version: 2.0.0 stable

* Device name: Fibaro Smoke Detector
* Device version:
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/129):

Logfile (Smoke Alarm):
```
12:45:15.826 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0F 00 04 00 04 09 71 05 00 00 00 FF 01 03 00 70 
12:45:15.836 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
12:45:15.840 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0F 00 04 00 04 09 71 05 00 00 00 FF 01 03 00 70 
12:45:15.844 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 0F 00 04 00 04 09 71 05 00 00 00 FF 01 03 00 70 
12:45:15.849 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 04 09 71 05 00 00 00 FF 01 03 00 
12:45:15.854 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 4: Application Command Request (ALIVE:DONE)
12:45:15.860 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 4: Starting initialisation from DONE
12:45:15.865 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@14fb17 already registered
12:45:15.870 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 4: Incoming command class ALARM
12:45:15.875 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: Received ALARM command V5
12:45:15.881 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: Process NOTIFICATION_REPORT V5
12:45:15.886 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: NOTIFICATION report - 0 = 0, event=3, status=255
12:45:15.891 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: Alarm Type = SMOKE (0)
12:45:15.892 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0C 00 04 00 04 06 31 05 01 22 00 E0 02 
12:45:15.901 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveAlarmValueEvent
12:45:15.906 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Got an event from Z-Wave network: ZWaveAlarmValueEvent
12:45:15.912 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Got a value event from Z-Wave network, endpoint = 0, command class = ALARM, value = 255
12:45:15.920 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
12:45:15.925 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter NOTIFICATION event is 3, type OnOffType
12:45:15.930 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Updating channel state zwave:device:ffabd0a2:node4:alarm_general to ON [OnOffType]

```

Logfile (Tampered alarm):
```
12:45:15.826 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0F 00 04 00 04 09 71 05 00 00 00 FF 01 03 00 70 
12:45:15.836 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
12:45:15.840 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0F 00 04 00 04 09 71 05 00 00 00 FF 01 03 00 70 
12:45:15.844 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 0F 00 04 00 04 09 71 05 00 00 00 FF 01 03 00 70 
12:45:15.849 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 04 09 71 05 00 00 00 FF 01 03 00 
12:45:15.854 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 4: Application Command Request (ALIVE:DONE)
12:45:15.860 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 4: Starting initialisation from DONE
12:45:15.865 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@14fb17 already registered
12:45:15.870 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 4: Incoming command class ALARM
12:45:15.875 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: Received ALARM command V5
12:45:15.881 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: Process NOTIFICATION_REPORT V5
12:45:15.886 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: NOTIFICATION report - 0 = 0, event=3, status=255
12:45:15.891 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: Alarm Type = SMOKE (0)
12:45:15.892 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0C 00 04 00 04 06 31 05 01 22 00 E0 02 
12:45:15.901 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveAlarmValueEvent
12:45:15.906 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Got an event from Z-Wave network: ZWaveAlarmValueEvent
12:45:15.912 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Got a value event from Z-Wave network, endpoint = 0, command class = ALARM, value = 255
12:45:15.920 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
12:45:15.925 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter NOTIFICATION event is 3, type OnOffType
12:45:15.930 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Updating channel state zwave:device:ffabd0a2:node4:alarm_general to ON [OnOffType]
```

Notes:

  * There are different items linked to alarm_general and to alarm_smoke, so it seems not to be caused by the Openhab2 configuration
  * The alarm was triggered as test alarm (pressing the button for ~ 6 seconds), but this shall not make any difference to a real alarm.
  * Same issue with second smoke detector of same type

euphi, 2017-01-28:
[node4.zip](https://github.com/openhab/org.openhab.binding.zwave/files/736868/node4.zip)

euphi, 2017-01-28:
Sorry, it seems that the other smoke detector actually DOES trigger an alarm_smoke:

```
13:06:46.607 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0F 00 04 00 07 09 71 05 00 00 00 FF 01 03 00 73 
13:06:46.623 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
13:06:46.630 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0F 00 04 00 07 09 71 05 00 00 00 FF 01 03 00 73 
13:06:46.639 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 0F 00 04 00 07 09 71 05 00 00 00 FF 01 03 00 73 
13:06:46.648 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 07 09 71 05 00 00 00 FF 01 03 00 
13:06:46.653 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 7: Application Command Request (ALIVE:DONE)
13:06:46.658 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 7: Starting initialisation from DONE
13:06:46.664 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@6ed02a already registered
13:06:46.667 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0C 00 04 00 07 06 31 05 01 22 00 CB 2A 
13:06:46.669 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 7: Incoming command class ALARM
13:06:46.684 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 7: Received ALARM command V5
13:06:46.690 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 7: Process NOTIFICATION_REPORT V5
13:06:46.695 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 7: NOTIFICATION report - 0 = 0, event=3, status=255
13:06:46.701 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 7: Alarm Type = SMOKE (0)
13:06:46.706 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveAlarmValueEvent
13:06:46.711 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Got an event from Z-Wave network: ZWaveAlarmValueEvent
13:06:46.717 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Got a value event from Z-Wave network, endpoint = 0, command class = ALARM, value = 255
13:06:46.723 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
13:06:46.728 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter NOTIFICATION event is 3, type OnOffType
13:06:46.734 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Updating channel state zwave:device:ffabd0a2:node7:alarm_smoke to ON [OnOffType]
13:06:46.742 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
13:06:46.746 [DEBUG] [pdb.internal.MapDBPersistenceService] - store called for ZWave_Fire_Treppe_Smoke
13:06:46.748 [DEBUG] [ap.internal.MapTransformationService] - Transformation resulted in '!!ALARM!!'
13:06:46.756 [DEBUG] [pdb.internal.MapDBPersistenceService] - Stored 'ZWave_Fire_Treppe_Smoke' with state 'ON' in mapdb database

```

I still don't see any difference in configuration....?!

euphi, 2017-01-28:
Ok.. It seems that this issue is already solved, but only for newly configured devices. What happened was:

* I configured the smoke detectors with a beta / snapshot build of Openhab2 (I updated quite often, so I don't know the version)
* I remember there was some confusion for alarm_general (two channels with same name).
* I upgraded to OpenHab2 stable
* I tested the alarm function for both smoke detectors. It triggered the alarm_general, but not the alarm_smoke. I had no time for further investigation.
* Due to some problems with the second detector, I deleted it and re-added it.
* Today I tested the alarm again (with the "old" smoke detector) to get information for this bug report.
* After I created this issue I tried again with the second ("new") smoke detector and this time it was fine.
* I noted in the PaperUI that there is no alarm_general anymore, but now an alarm_tamper (beside others).

**Conclusion:** Newly installed Smoke detector work fine. To make old configurations work, you need to deleted them and re-add the Smoke detector.

I did so for my first Smoke detector and it is fine now, too.

I will keep this issue open, because maybe it is possible to perform an upgrade for old configurations at least for the alarm_smoke channel?

cdjackson, 2017-01-28:
It's not possible to upgrade - you need to delete the thing, add it back and add the links to the items again (sorry).

closed 2017-01-28 by cdjackson