openhab/org.openhab.binding.zwave #332: HS-WD100+ Size error 1<>2 from config_10_1

abedwardsw, 2017-01-29:
Hi there, I am trying to update parameter # 10 on my homeseer HS-WD100+ lightswitch.    The current (and default) value is 3.   When I try to put in a new value (such as 2), I get the following error in the openhab logs.    Do you know if this might be a bug (should we check if size is <= expected size instead of just equal to?)  or something wrong with the database setup?

Please let me know if there is anything else I can do to help.

```
[ERROR] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Size error 1<>2 from config_10_1
```

* OH2 Version: 2.0.0
* Binding Version : 2.0.0
* Device name: HS-WD100+
* Device version: 5.16
* db link: http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/346
* node xml file attached [node9.txt](https://github.com/openhab/org.openhab.binding.zwave/files/737702/node9.txt)

Debug output:
```
2017-01-29 09:55:00.875 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Configuration update received
2017-01-29 09:55:00.881 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Configuration update config_10_1 to 2
2017-01-29 09:55:00.882 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 9: Sending REQUEST Message = 01 0D 00 13 09 06 70 04 0A 02 00 02 25 AF 1A
2017-01-29 09:55:00.892 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 04 01 13 01 E8
2017-01-29 09:55:00.908 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 07 00 13 AF 00 00 02 46
2017-01-29 09:55:00.909 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Got an event from Z-Wave network: ZWaveTransactionCompletedEvent
2017-01-29 09:55:00.909 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 9: Sending REQUEST Message = 01 0A 00 13 09 03 70 05 0A 25 B0 06
2017-01-29 09:55:00.916 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 04 01 13 01 E8
2017-01-29 09:55:00.933 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 07 00 13 B0 00 00 02 59
2017-01-29 09:55:00.944 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0C 00 04 00 09 06 70 06 0A 02 00 02 84
2017-01-29 09:55:00.945 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Got an event from Z-Wave network: ZWaveConfigurationParameterEvent
2017-01-29 09:55:00.945 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Got a value event from Z-Wave network, endpoint = 0, command class = CONFIGURATION, value = org.openhab.binding.zwave.internal.protocol.ZWaveConfigurationParameter@188ed506
2017-01-29 09:55:00.945 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Update CONFIGURATION 10/2 to 2
2017-01-29 09:55:00.946 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Config about to update 7 parameters...
2017-01-29 09:55:00.946 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing switchall_mode
2017-01-29 09:55:00.946 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing config_10_1
2017-01-29 09:55:00.946 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing config_10_1 len=3
2017-01-29 09:55:00.946 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing config_10_1 - id = '10'
2017-01-29 09:55:00.946 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing config_10_1 - size = '1'
2017-01-29 09:55:00.946 [ERROR] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Size error 1<>2 from config_10_1
2017-01-29 09:55:00.946 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing config_7_1
2017-01-29 09:55:00.946 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing config_7_1 len=3
2017-01-29 09:55:00.946 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing config_7_1 - id = '7'
2017-01-29 09:55:00.946 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing config_8_1
2017-01-29 09:55:00.946 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing config_8_1 len=3
2017-01-29 09:55:00.946 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing config_8_1 - id = '8'
2017-01-29 09:55:00.947 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing group_org.openhab.binding.zwave.internal.protocol.ZWaveAssociationGroup@21a39982
2017-01-29 09:55:00.947 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing config_9_1
2017-01-29 09:55:00.947 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing config_9_1 len=3
2017-01-29 09:55:00.947 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing config_9_1 - id = '9'
2017-01-29 09:55:00.947 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing config_4_1
2017-01-29 09:55:00.947 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing config_4_1 len=3
2017-01-29 09:55:00.947 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Processing config_4_1 - id = '4'
2017-01-29 09:55:00.947 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Got an event from Z-Wave network: ZWaveTransactionCompletedEvent
```

Thank you.

cdjackson, 2017-01-29:
The database should be updated to say that it's 2 bytes long, but other than this it looks like it's working -:

[image]

I've updated the database -:

http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/346

closed 2017-01-29 by cdjackson

abedwardsw, 2017-01-29:
thanks @cdjackson , will test once the next release comes out.

I guess the reason I though it wasn't working is that it stays as "pending" in the HABMin console.  Whereas if I update any of the other configuration items, they update properly and the pending goes away.

cdjackson, 2017-01-29:
Yes - I didn’t check well, but I think the UI might not have updated, but it looks like the command was set on the device, and read back ok. I’ll try and update the binding tonight - if there’s further problems, then feel free to reopen...