openhab/org.openhab.binding.zwave #334: AEON Multisensor 6 Size error 4<>1 from config_255_4_wo

abedwardsw, 2017-01-29:
I get the following message in my logs everytime I restart openhab or update the configuration of the device.  I guess the database might need updating?  I have seen this same error in a few others that have posted their logs relating to the MS6 as well.

```
[ERROR] [ding.zwave.handler.ZWaveThingHandler] - NODE 8: Size error 4<>1 from config_255_4_wo
```

* OH2 Version: 2.0.0
* Binding Version : 2.0.0
* Device name: AEON Labs Multisensor 6
* Device version: 1.8
* Link to [database-387](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/387):
* xml file [node8.txt](https://github.com/openhab/org.openhab.binding.zwave/files/737786/node8.txt)

I see in the zwave-device-database the word size is 4, however in my node8.xml file it shows the size as 1.   I looked into the [documentation](https://aeotec.freshdesk.com/support/solutions/articles/6000057073-multisensor-6-user-guide-) it looks like 255 seems to have 2 different sizes listed depending on the action (either 1 or 4).   

On a separate note, I was going to add the documentation link to the database, I created an account (abe), if you want you can give me update rights, otherwise if you have a moment, you could add the documentation to your database if you want.

```
2017-01-29 12:11:12.714 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 8: Processing config_255_4_wo
2017-01-29 12:11:12.714 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 8: Processing config_255_4_wo len=4
2017-01-29 12:11:12.714 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 8: Processing config_255_4_wo - id = '255'
2017-01-29 12:11:12.714 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 8: Processing config_255_4_wo - size = '4'
2017-01-29 12:11:12.714 [ERROR] [ding.zwave.handler.ZWaveThingHandler] - NODE 8: Size error 4<>1 from config_255_4_wo
```

Thank you.

cdjackson, 2017-01-29:
This parameter is both 1 byte and 4 bytes - we had to choose one option, so we choose 4 to allow the device to be reset.

Sorry - not much we can do at the moment.

closed 2017-01-29 by cdjackson

abedwardsw, 2017-01-30:
@cdjackson  just curious, would it be worth trying to consider supporting a comma separated list for the input size.  ie.. if you put 4,1 then that would mean either size is fine?  If you think it makes sense I could try to update the code and send you a pull request in my spare time.

cdjackson, 2017-01-30:
I don't think it's that simple - there are different sizes, and different values and different functions (they should have been two different config parameters in my opinion).

We might be able to add two parameters with different sizes. This is not a common configuration - I've never seen the use of two sizes for the same parameter before and I'm not really sure it matters too much (does it?).  I think one option resets the device to factory reset, and the other resets it and removes it from the network. It's not something I'm too inclined to spend time on at the moment.

abedwardsw, 2017-01-30:
np, thanks.  Agree.  If it was a simple fix then yes.  But otherwise agree not worth the time.  thanks for the response.