openhab/org.openhab.binding.zwave #335: Timout with Fibaro Plug FGWP-102 (due to wrong DB entires?)

philomatic, 2017-01-30:
Hi Chris,

we're facing Timeout problems with the FGWP-102 Plug from Fibaro.

* Binding Version: Current Master (5df0ced)
* Device version: FGWPE/F-102 ZW5 (Firmware 3.2)
* DB Link: http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/464
* Manual: http://manuals.fibaro.com/content/manuals/en/FGWPEF-102/FGWPEF-102-EN-A-v2.0.pdf
* XML File from OH2: [node2.xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/738715/node2.xml.txt)
* Log File (Full): [openhab_full.log.txt](https://github.com/openhab/org.openhab.binding.zwave/files/738725/openhab_full.log.txt)
* Log File (Recurring Pattern): [openhab_pattern.log.txt (https://github.com/openhab/org.openhab.binding.zwave/files/738743/openhab_pattern.log.txt)

See the attached log(s). It's Node 2 in the logs, I'm sorry for the inconvenience, I can't set up a test with this node only atm.

I guess the problem are malformed DB entries. 
The configuration params aren't correct, they're aren't the same parameter IDs as the FGWP-101 has (as stated in the DB). Could this lead to these problems?

I just registered for the DB:
pg@vs-sh.de

If my assumption is correct, please activate my account as I would like to update the entries.

Best regards
Patrick from VSS

philomatic, 2017-01-30:
Digged a little deeper:

Seems like Fibaro simply updated the firmware and changed a bunch of param IDs.
So, the DB entries aren't actually wrong, they're fitting for 102 v1.0 Plugs with zwave firmware 2.1-2.5.
A new XML is needed for v2.0 102 Plugs with zwave firmware 3.2.

Can you guide me through the process? Is there an automated tool for the correct param id detection?

cdjackson, 2017-01-30:
I've updated and activated your account. 

It seems like you need to create a new device. First you probably should edit the old one to set the max version of firmware then I think you should be able to upload your XML as a new Device. 

Then you can copy the configuration from the old device to the new one - this is in the toolbox menu in the top right of the device summary window. 

With your current access you can't delete anything - you can only edit. So if the changes need parameters to be deleted then please mark them as PLEASE DELETE or something.

Any problems just yell.

philomatic, 2017-01-30:
Ok, unfortunately we've uploaded the node.xml as new device before updating the version bounds for the old one. This resulted in nonsense for

http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/464

(See endpoint 0)

I'm sorry for that, but this needs to be reverted I guess.
Honestly I thought there would be some kind of preview after uploading XMls :/

Furthermore, I've updated the version bounds (according to fibaros homepage) and added a new device to the DB:

http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/513

We'll update association groups and parameters soon, however copying from 464 won't help much as Fibaro reworked nearly all parameters (including ID, partly value semantic and documentation texts)...

But I'm not sure with the following point:

> Endpoint 0 has no command class linked to the basic class.

I guess for the plug this should be SWITCH_BINARY and set it for 513, like in 464.
Am I correct?

Best regards
Patrick

cdjackson, 2017-01-30:
If you upload the XML when the endpoints are already included, then the software should ignore the new file, so nothing should have changed. I suspect these errors/warnings were already there.

If the specific class is binary switch, then yes, ticking the SWITCH_BINARY should be fine.

philomatic, 2017-01-30:
Ok, thanks for your advice.

See the current diff for further investigation:
http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/464?layout=diff

Best regards

cdjackson, 2017-01-30:
Hmmm - strange. Maybe I've disabled that check :(

cdjackson, 2017-04-16:
As far as I know the FGWP has been updated in the database so closing this. Any further issues, please update the database directly.

closed 2017-04-16 by cdjackson