openhab/org.openhab.binding.zwave #336: False Alarms with Fibaro FGSD-002

euphi, 2017-01-30:
* OH2 Version: 2.0.0 stable
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): 2.0.0 stable

If the problem is associated with a device, please provide the following -:

* Device name: FGSD-002
* Device version:
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/129):

The smoke detector triggers various false alarm during wakeup.
A real alarm triggers the alarm three times.

During wakeup, the smoke detector sends notification messages for an alarm, e.g.:

`01 0F 00 04 00 04 09 71 05 00 00 00 FF 07 03 00 76`

In case of a real alarm the detector sends three messages:
```
01 0F 00 04 00 04 09 71 05 00 00 00 FF 07 03 00 76
01 0F 00 04 08 04 09 71 05 00 00 00 FF 07 03 00 7E
01 0F 00 04 00 04 09 71 05 00 00 00 FF 07 03 00 76
```

All three messages triggers an update "ON" to the linked channel.

Log of "false" alarm:

```
2017-01-29 21:51:10.673 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0F 00 04 00 04 09 71 05 00 00 00 FF 07 03 00 76 
2017-01-29 21:51:10.673 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 09 00 13 06 00 00 02 00 00 E1 
2017-01-29 21:51:10.678 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=SendData[0x13], type=Request[0x00], priority=High, dest=255, callback=0, payload=06 00 00 02 
2017-01-29 21:51:10.681 [DEBUG] [l.serialmessage.SendDataMessageClass] - NODE 4: SendData Request. CallBack ID = 6, Status = Transmission complete and ACK received(0)
2017-01-29 21:51:10.685 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: Sent Message: class=SendData[0x13], type=Request[0x00], priority=Get, dest=4, callback=6, payload=04 05 71 04 00 07 03 
2017-01-29 21:51:10.691 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: Recv Message: class=SendData[0x13], type=Request[0x00], priority=High, dest=255, callback=0, payload=06 00 00 02 
2017-01-29 21:51:10.694 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: class=SendData, callback id=6, expected=ApplicationCommandHandler, cancelled=false      MISMATCH
2017-01-29 21:51:10.698 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
2017-01-29 21:51:10.707 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0F 00 04 00 04 09 71 05 00 00 00 FF 07 03 00 76 
2017-01-29 21:51:10.712 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 0F 00 04 00 04 09 71 05 00 00 00 FF 07 03 00 76 
2017-01-29 21:51:10.716 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 04 09 71 05 00 00 00 FF 07 03 00 
2017-01-29 21:51:10.722 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 4: Application Command Request (ALIVE:DYNAMIC_VALUES)
2017-01-29 21:51:10.725 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 4: Incoming command class ALARM
2017-01-29 21:51:10.728 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: Received ALARM command V5
2017-01-29 21:51:10.732 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: Process NOTIFICATION_REPORT V5
2017-01-29 21:51:10.736 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: NOTIFICATION report - 0 = 0, event=3, status=255
2017-01-29 21:51:10.739 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: Alarm Type = BURGLAR (0)
2017-01-29 21:51:10.743 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveAlarmValueEvent
2017-01-29 21:51:10.746 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Got an event from Z-Wave network: ZWaveAlarmValueEvent
2017-01-29 21:51:10.750 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Got a value event from Z-Wave network, endpoint = 0, command class = ALARM, value = 255
2017-01-29 21:51:10.754 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
2017-01-29 21:51:10.758 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
2017-01-29 21:51:10.761 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter NOTIFICATION event is 3, type OnOffType
2017-01-29 21:51:10.765 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Updating channel state zwave:device:ffabd0a2:node4:alarm_tamper to ON [OnOffType]
2017-01-29 21:51:10.775 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
2017-01-29 21:51:10.777 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
2017-01-29 21:51:10.778 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
2017-01-29 21:51:10.779 [DEBUG] [ome.core.thing.internal.ThingManager] - Delegating update 'ON' for item 'ZWave_Fire_Ian_Tamper' to handler for channel 'zwave:device:ffabd0a2:node4:alarm_general'
2017-01-29 21:51:10.781 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: Sent Message: class=SendData[0x13], type=Request[0x00], priority=Get, dest=4, callback=6, payload=04 05 71 04 00 07 03 
2017-01-29 21:51:10.784 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: Recv Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 04 09 71 05 00 00 00 FF 07 03 00 
2017-01-29 21:51:10.785 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: class=ApplicationCommandHandler, callback id=6, expected=ApplicationCommandHandler, cancelled=false        transaction complete!
2017-01-29 21:51:10.798 [DEBUG] [ntime.internal.engine.RuleEngineImpl] - Executing rule 'Warnings'
2017-01-29 21:51:10.788 [DEBUG] [pdb.internal.MapDBPersistenceService] - store called for ZWave_Fire_Ian_Tamper
2017-01-29 21:51:10.804 [DEBUG] [pdb.internal.MapDBPersistenceService] - Stored 'ZWave_Fire_Ian_Tamper' with state 'ON' in mapdb database
```

Log of real alarm:

```

2017-01-30 21:34:27.570 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0F 00 04 00 04 09 71 05 00 00 00 FF 07 03 00 76 
2017-01-30 21:34:27.580 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
[... 105 lines cut ...]
2017-01-30 21:34:28.287 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=08 04 07 9C 02 04 00 FF 00 00 
2017-01-30 21:34:28.307 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 4: Application Command Request (ALIVE:DONE)
2017-01-30 21:34:28.308 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 4: Starting initialisation from DONE
2017-01-30 21:34:28.309 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@d85c69 already registered
2017-01-30 21:34:28.311 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 4: Incoming command class SENSOR_ALARM
2017-01-30 21:34:28.312 [DEBUG] [ndclass.ZWaveAlarmSensorCommandClass] - NODE 4: Received SENSOR_ALARM command V1
2017-01-30 21:34:28.313 [DEBUG] [ndclass.ZWaveAlarmSensorCommandClass] - NODE 4: Alarm Report: Source=4, Type=General(0), Value=255
2017-01-30 21:34:28.314 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveAlarmSensorValueEvent
2017-01-30 21:34:28.316 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Got an event from Z-Wave network: ZWaveAlarmSensorValueEvent
2017-01-30 21:34:28.317 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Got a value event from Z-Wave network, endpoint = 0, command class = SENSOR_ALARM, value = 255
2017-01-30 21:34:28.336 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 7: Transaction not completed: node address inconsistent.  lastSent=7, incoming=255
2017-01-30 21:34:28.340 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
2017-01-30 21:34:28.351 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0D 00 04 00 04 07 9C 02 04 00 FF 00 00 90 
2017-01-30 21:34:28.356 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 0D 00 04 00 04 07 9C 02 04 00 FF 00 00 90 
[... cut ...]
2017-01-30 21:34:28.426 [DEBUG] [ndclass.ZWaveAlarmSensorCommandClass] - NODE 4: Alarm Report: Source=4, Type=General(0), Value=255
2017-01-30 21:34:28.430 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveAlarmSensorValueEvent
2017-01-30 21:34:28.435 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Got an event from Z-Wave network: ZWaveAlarmSensorValueEvent
2017-01-30 21:34:28.439 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Got a value event from Z-Wave network, endpoint = 0, command class = SENSOR_ALARM, value = 255
```

cdjackson, 2017-01-30:
Please refer to discussions on the forum.

helmar74, 2017-02-10:
I think bug is fixed in newest Snapshot version:
https://community.openhab.org/t/zwave-fibaro-smoke-sensor-state-change-during-regular-wakeup/20013/99

closed 2017-02-10 by cdjackson

helmar74, 2017-02-10:
It's a pleasure ;-)

euphi, 2017-02-11:
Is there a simple way to install it with 2.0 stable? E.g. just copying a jar file to some directory?

helmar74, 2017-02-11:
Hi,

I don't know. I've updated to "unstable" 2.1.0 without any problem. Maybe you can ask chris for support to get it running under 2.0 stable. I followed jaydee's description. You should also be able to go back to stable 2.1 as soon it arrives...

cdjackson, 2017-02-11:
It’s not so easy, but not too difficult - depends on what you think…

First uninstall the binding using PaperUI or HABmin, then drop the snapshot binding into the addons folder. Log on to the Karaf console and type the following command. Hopefully it will work...

feature:install openhab-transport-serial