openhab/org.openhab.binding.zwave #340: [development] Timeout issues

pizerg, 2017-02-03:
Lately I've been testing development branch with an installation with a decent amount of nodes (more than 15) and I've noticed some transactions take some time to start (making actuation and data flow less responsive compared to "master" version which is really smooth) and quite often I see  "Node Init response (X) TIMEOUT_WAITING_FOR_DATA" and "XXXXXXX Timeout at state WAIT_DATA. 3 retries remaining.". 

Digging into the code I've seen that MAX_OUTSTANDING_TRANSACTIONS is not being used at the moment, only allowing 1 outstanding transaction at a time, is there a reason for that ?

Also I found that the retries remaining are always 3 as the code that deals with attempts remaining is commented out in ZWaveTransactionManager, but I think the messages are being retried anyway so this might be handled somewhere else.

Finally I also see several of "Start transaction timer to xxxx - yyyy ms" where yyyy is around 2000ms, maybe this is supposed to be this way but I thought this should always be around 5000ms (the actual default transaction timeout), so maybe this is the issue causing my timeout problems.

Adrián

cdjackson, 2017-02-03:
> Digging into the code I've seen that MAX_OUTSTANDING_TRANSACTIONS is not being used at the moment, only allowing 1 outstanding transaction at a time, is there a reason for that ?

I found that more than 1 transaction outstanding started to cause problems elsewhere. This is the same as the master branch which also only has the ability to handle a single outstanding transaction. For the first parts of the transaction (up to receiving the ACK from the device) the stick can only handle a single transaction. After this, we could allow more, but we need to be careful not to overload a device. My original thought was to allow multiple transactions to different nodes, but it started to become complex and I’m not sure it’s worth the effort.

I will tidy the code to remove these remnants if we leave it this way.

> Also I found that the retries remaining are always 3 as the code that deals with attempts remaining is commented out in ZWaveTransactionManager, but I think the messages are being retried anyway so this might be handled somewhere else.
> 
Yes - I decided to remove retries from the transaction at transaction level. I think this is stated in the thread on the forum. There seems to me to be limited point (in my opinion ;) ) - the stick already does 3 retries at its level, so at this level I don’t think its needed (but I don’t close this decision if we find that it is needed). Retries are handled in some cases higher up the stack - so for the initialisation code, it has a retry mechanism. The downside is a timeout takes (lets say) 5 seconds at this level, so if we repeat this 3 times we have a blocking period of 15 seconds.

> Finally I also see several of "Start transaction timer to xxxx - yyyy ms" where yyyy is around 2000ms, maybe this is supposed to be this way but I thought this should always be around 5000ms (the actual default transaction timeout), so maybe this is the issue causing my timeout problems.
> 
There are different timers used - if I remember correctly this is described in the transaction manager source file doc? Depending on which phase of the transaction we’re waiting for, different timeouts are used.  This was changed from  what’s in master where we only have a single 5 second timer for the whole transaction - in this code, I have some shorter timer for the initial parts where we’re just waiting for the response from the stick, or the initial ack, and longer for the final response. This actually allows a longer overall timeout than we had before - so long as the transaction keeps progressing.

I certainly welcome feedback and thoughts on the way this works - I’ve tuned this from many hours of testing and am currently using a network with 20 devices and it’s working well. That said, I have an open mind on how it can be improved - docs on this lower layer are limited (i.e. not included in the public standard) so there’s an element of trial and error here.

It would be great if you can do some analyses on what is happening in your system to understand the reason if possible, and we can discuss if there’s improvements we can make to now this works.

pizerg, 2017-03-28:
I had some time to revisit this issue and found out that most of my timeout issues were due to sensors that are not fully compliant (for example AEON Door Sensor 6, when linked securely it reports Wake Up without security).

Having said that, I still think it would be a good addition to allow multiple transactions at the same time (but only to different nodes), that way if for some reason one of the sensors is not working properly (slow response time or no response at all) we would not delay other messages that may have high priority (if you're executing several actions as a result of a rule trigger or some similar use cases).

What do you think? It would be for sure more complex at implementation level, but I believe we can have one queue per node and handle it in a quite clean way.

cdjackson, 2017-03-28:
I originally tried multiple transactions to different nodes - so this was always planned. However, it probably doesn’t really provide what you want as the controller only allows a single transaction at any time - at least from my experimentation, that’s what I found. So, this is the part up till when we get the ACK from the device. This means it’s not possible to have multiple SET commands outstanding at all. Currently I think the binding allows multiple outstanding transactions after this point already.

Maybe you have found that this works ok already? I however spent a lot of time on this, but found it didn’t work and caused lots of problems wit the controller.

pizerg, 2017-03-29:
I assumed that the controller would be able to correlate ACKs for several outstanding SET commands. If that's not the case I'll be closing this issue, as you previously stated we will not be improving too much just by getting more GET commands in parallel and it would certainly make the process much more complicated than it is now (and quite harder to debug as well). 

Thanks again.

Adrián

closed 2017-03-29 by pizerg

cdjackson, 2017-03-29:
I also assumed that it was possible to have multiple outstanding commands after the first ACK from the controller. I assumed that this is why there is a transaction ID on the end of the frame, and this ID is added when responses come back (it’s not sent over the air though). This was my assumption on which I based the original transaction rewrite, but it didn’t work. I forget the issues now - I think the controller started loosing frames and not responding (using Aeon G5) and the only way I could get this reliable was to go back to a single outstanding transaction up until the point at which the ACK is received from the device.

Of course, I’m basing this only on very old docs that I have found on the web that documents this interface and none of this is explicitly stated within the doc.

Currently I think the code will release a second transaction as soon as the remote device ACKs - this in itself is causing problems as it can send multiple commands to a single device so I need to resolve this as well.  ZWave doesn’t seem to be very good in this respect.

I will try and ask Sigma about some of this - I know they don’t intend to release the spec for the serial API but maybe I can get this answered at least (I have a call with them coming up).

laohixdxm, 2017-09-21:
I try to do summary from above discussion as below as I might experience delay to turn on/off zwave switch through OH2 & zwave binding but I only have 1 zwave sticker & 1 zwave switch.

*Before 1st ACK from device, only 1 outstanding request can be on the sending queue.
*After 1st ACK from device, multiple outstanding request can be on the sending queue.
*There are at most 3 retries for each request if failure. But during retry, sending will be blocked. 
I am NOT saying there must be timeout/delay issue for zwave binding. But above items might NOT be related to timeout/delay issue if it does exist.

NOT sure I have some understanding here. Thank you.

cdjackson, 2017-09-21:
Sorry - I don’t understand what you mean.

If you have timeouts, then I suggest to look at the log.

laohixdxm, 2017-09-21:
Sorry for typo. Don't worry about my timeout & I do NOT have timeout log available at this moment.

I just try to have some summary from long discussion above.

*Before 1st ACK from device, only 1 outstanding request can be on the sending queue.
*After 1st ACK from device, multiple outstanding request can be on the sending queue.
*There are at most 3 retries for each request if failure. But during retry, sending will be blocked. 

NOT sure I have some misunderstanding here. Thank you.

cdjackson, 2017-09-21:
Sorry - I’m not really in a position to comment further on this.

laohixdxm, 2017-09-21:
Sorry I may NOT describe myself clearly.Let me try again.

Am I correct to have below statements?
*Before 1st ACK from device, only 1 outstanding request can be on the sending queue.
*After 1st ACK from device, multiple outstanding request can be on the sending queue.
*There are at most 3 retries for each request if failure. But during retry, sending will be blocked.

An yes/no answer is all I expect. Thank you.