openhab/org.openhab.binding.zwave #342: Channel removed from 507601?

mtryfoss, 2017-02-03:
Hello!

I wonder why this was removed from the unit?

ESH-INF/thing/merten_507601_0_0.xml

-      <channel id="switch_binary" typeId="switch_binary">
-        <label>Switch</label>
-        <properties>
-          <property name="binding:*:OnOffType">SWITCH_BINARY,BASIC</property>
-        </properties>
-      </channel>

Now the device is non-functional :)

cdjackson, 2017-02-04:
I'm not sure what you refer to - it looks like this device hasn't changed for many months.  If it's missing configuration please feel free to update the database.

mtryfoss, 2017-02-04:
I also noticed that the device has not been updated since august, but commit 00ae5ba089c43dd32d69e4f7b4fddbed3bc24f91 includes this change. Just took a long time before it was exported?

However, I'm not quite sure what to add to rollback this.

cdjackson, 2017-02-04:
> I also noticed that the device has not been updated since august, but commit 00ae5ba <https://github.com/openhab/org.openhab.binding.zwave/commit/00ae5ba089c43dd32d69e4f7b4fddbed3bc24f91> includes this change. Just took a long time before it was exported?

You need to edit the database to add the channel that was deleted back into the command class - as per the file you referenced…

mtryfoss, 2017-02-05:
I think I got it right now :)

cdjackson, 2017-02-05:
Thanks.

You didn’t need to add the ‘config’ part - this is generated automatically (I’ve removed it).

I’ll update the binding today - hopefully it will solve your issues.

cdjackson, 2017-02-05:
Hmm - on second looks this might need a bit more work as you'll only have 1 switch. Do you have the XML generated by OH for this device? If so, can you post if please?

mtryfoss, 2017-02-06:
Here you go :)

I reverted the channel change and built the jar locally. The device is working again now.

[node15.xml.zip](https://github.com/openhab/org.openhab.binding.zwave/files/753863/node15.xml.zip)

cdjackson, 2017-12-30:
Closing as it seems the device is working.

closed 2017-12-30 by cdjackson