openhab/org.openhab.binding.zwave #343: Not existing channel in TF016 Heat-It Thermostat

MatthiasSpriegel, 2017-02-05:
Hi,

there is a not existing channel `(<commandClass>METER</commandClass>)` in the device:

* Device name: HEATIT Z-WAVE TF016
* Device version: 1.8
* Link to database: http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/412

[node23.txt](https://github.com/openhab/org.openhab.binding.zwave/files/752866/node23.txt)

The commandClass should be deleted to not confuse users or? Hope this is a correct description. Thanks in advance,
Matthias

cdjackson, 2017-02-05:
Please consider updating the database to add this channel.

http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide <http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide>

MatthiasSpriegel, 2017-02-05:
Did it, i have not clicked on "request review" yet, maybe it's better someone takes a look on the data before something is messed up "... the checks are not foolproof ..." ;-)

cdjackson, 2017-02-05:
I've removed the channel since it doesn't seem to be mentioned as a supported command class in the manual, even though the device reports it is supporting this.

closed 2017-02-05 by cdjackson