openhab/org.openhab.binding.zwave #349: New version of Fibaro motion sensor not recognized?

mtryfoss, 2017-02-06:
From the log:
2017-02-05 14:16:01.468 [WARN ] [wave.discovery.ZWaveDiscoveryService] - NODE 70: Device discovery could not resolve to a thingType! 010F:0801:1001::3.2

I got the new device (fibaro_fgms001_3_2.xml) in the binding (there are two entries for this in the database, so I didn't want to just upload the xml again).

[node70.xml.zip](https://github.com/openhab/org.openhab.binding.zwave/files/753886/node70.xml.zip)

mtryfoss, 2017-02-06:
The logfile was rotated. After checking some older ones, I found this:

2017-02-05 12:18:41.822 [DEBUG] [curityCommandClassWithInitialization] - NODE 70: SECURITY_INITIALIZE Initialising=false, Inclusion=true, Paired=false, lastRxMsg=1486293521822ms, lastTxMsg=12943ms
2017-02-05 12:18:41.822 [ERROR] [curityCommandClassWithInitialization] - NODE 70: SECURITY_INCLUSION_FAILED Failed at step SECURITY_SCHEME_GET: 10000ms passed since last request was sent, secure inclusion failed.

Will secure inclusion work in openhab now?

cdjackson, 2017-02-06:
If you can update the database manually for your device it would be appreciated.

Security is not included in OH2 at the moment - it's under test only.

mtryfoss, 2017-02-06:
Of course I can, but I'm a little bit unsure what to fix?

Tried uploading the xml.

cdjackson, 2017-02-06:
I'm assuming that you need to add the 0801:1001 into the database - if this is already there, then this isn't the problem.

I note in the log that your binding is trying to use security - this doesn't work in OH2 at the moment so you should disable it or devices might not work properly.

mtryfoss, 2017-02-06:
0801:1001 seems to be there.

I was a little bit unsure about the security option. I'm going to try re-inclusion without. The installation is in my cabin - far away from where I'm now, so I won't be able to test until the weekend.

Is it necessary to restart openhab/zwave binding after disabling security?

cdjackson, 2017-02-06:
> Is it necessary to restart openhab/zwave binding after disabling security?

I'm not 100% sure, but it's probably wise. In the first implementation which came from OH1 there were some static methods used, so these are probably only set on startup.

mtryfoss, 2017-02-06:
Speaking of restart. I've got one battery operated node (HORSTMANN HRT4-ZW) which will never initialize completely by itself without sending a couple of NIF's after an openhab restart.

Habmin reports something like "node not found in zwave network" and openhab rejects any reports from the device. After NIF, everything is working ok. The network has 5 of these units and the other 4 are working as they should.

I've not yet been able to investigate more.
Any ideas/hints?

mtryfoss, 2017-02-13:
How can I 100% for sure know what security is disabled in the binding? I tried disabling it and then do a full openhab restart followed by exclude/include. The SECURITY class is still reported and the device is not recognised.

cdjackson, 2017-02-13:
Security class will still be reported - that comes from the device. Security is disabled in the controller configuration parameters.

cdjackson, 2017-04-13:
It's been a couple of months so closing - if there's more news with the latest binding, then feeel free to reopen.

closed 2017-04-13 by cdjackson