openhab/org.openhab.binding.zwave #355: Sending commands to Thing Nodes which are currently marked as OFFLINE blocks Z-Wave

philomatic, 2017-02-10:
When I send a command to an item linked to an OFFLINE device, Z-Wave tries to reach this Node (3 times I think in about 30 seconds) which blocks the Z-Wave job queue. 

As the controller can't operate multi-tasking, other Z-Wave commands will be blocked.
They will reach their node with high delay or even be discarded.

This is especially a problem with battery operated actors, like alarm sirens, when working with sets of commands for scenarios.

Imho Z-Wave should not send commands to Nodes which are currently OFFLINE;
instead wait for a life signal or check them repeatedly every few minutes.

At least a config option for this should be available.

Best regards

cdjackson, 2017-02-10:
I didn't think this was possible - I thought that the framework only sent commands to bindings when the device was ONLINE. @kaikreuzer can you confirm?

Even if it does, timeout is 5 seconds - not 30 seconds, and we do not (ever!) send messages to battery devices unless they first send a wakeup to the binding otherwise they will time out even if they are online.

I will close this since it doesn't make sense. I suggest that you discuss what problem you are seeing on the forum.

closed 2017-02-10 by cdjackson

philomatic, 2017-02-16:
Hi Chris,

yes, you're absolutely right, I'm sorry I've slipped in line:

Indeed, the nodes are (wrongly) marked as ONLINE, however since they're not powered (one fibaro plug with AC (node 16), one alarm siren with battery (node 13)), they definetely shouldn't.
Logs attached.
[openhab.txt](https://github.com/openhab/org.openhab.binding.zwave/files/779502/openhab.txt)
[events.txt](https://github.com/openhab/org.openhab.binding.zwave/files/779501/events.txt)

ZWave can't reach them and timeouts, but still does not mark them as OFFLINE.

Any ideas?

Best regards

cdjackson, 2017-02-16:
In the current code it keeps the dead flag but reduces the number of retries to 1, but it still tries to communicate with the device. The problem is that if we never send anything to the node then you don't know if it comes back online or not - it would be permanently unusable after the first set if timeouts. In your case this might be what you need but most timeouts don't require such drastic action - we need to keep sending to a device so we know if it comes back online. 

If you have devices that are not powered I would delete the thing. Maybe ESH needs a state where a user can manually take a device offline and the binding will not try to communicate with it? @kaikreuzer what do you think. 

I will however take a look over your logs as the thing should be marked offline if the state goes to dead so if this isn't happening there might be a bug. However even once offline the binding will continue to send messages periodically as outlined above.

philomatic, 2017-02-16:
> I will however take a look over your logs as the thing should be marked offline if the state goes to dead so if this isn't happening there might be a bug. However even once offline the binding will continue to send messages periodically as outlined above.

I'm totally with you that this behavior is correct, trying to reach dead marked nodes in order to bring them back ONLINE. So, the periodic checks aren't a problem for me after all, maybe the period time could be a parameter in settings?

> If you have devices that are not powered I would delete the thing. Maybe ESH needs a state where a user can manually take a device offline and the binding will not try to communicate with it? @kaikreuzer what do you think.

Nice idea, unfortunately this won't solve my problems, as the nodes could be back powered on at any time. So a manual deletion / flag won't help me, but maybe others?

Imo, the real problem for me is that they're not going OFFLINE and then this happens (if I got this right):

> When I send a command to an item linked to an OFFLINE device, Z-Wave tries to reach this Node (3 times I think in about 30 seconds) which blocks the Z-Wave job queue.
>
> As the controller can't [...]

Except that the correspondending Thing is indeed marked wrongly as ONLINE.

Best regards and thanks for your effort!

cdjackson, 2017-02-16:
The poll time is currently configurable so you can adjust this now. 

It sounds to me like the device isn't being set to dead - if it is, then the number of timeouts gets set to 1 which means 5 second timeout rather than 15 seconds (I don't understand why you would see 30 seconds though unless here are two messages being sent and the first one isn't causing the dead condition).

If you are powering mains powered devices off periodically this is likely to cause you problems since I think Zwave isn't really designed for this. Non mains devices, or devices where power is not continuous should be flagged at protocol level - I'm just not sure how the controller will handle this but your routing is very likely to get messed up since routes are only updated during a nightly heal.