openhab/org.openhab.binding.zwave #357: zwave association to a multi endpoint node

tom-art, 2017-02-11:
Hi,

as already written in the openhab community forum I would like to make an association from a aeotec multisensor 6 (node 11) to a fibaro double relay switch (node 10). 
In habmin2 I can set the association to "Node 10" under Configuration->Things->my_motion_sensor->Association Groups. This seems to work. 
When I look at the association config in the paper UI I see a association to "node_10_0".

But I need an association to "node_10_1", which is the endpoint for the first (of two) relay switches of the fibaro double relay switch.

In habmin2 I can only choose "Node 10" (which results in node_10_0).
In the paper UI I can choose "node_10_1", but I can't save it. The save pencil icon is inactive.

@cdjackson told me that this isn't supported at the moment and I should raise an issue at the issue tracker.
(https://community.openhab.org/t/zwave-association-to-a-multi-endpoint-node/22648)

* OH2 Version: openHAB 2.0.0-1 (Release Build) with binding-zwave - 2.0.0

thx,
Tom

cdjackson, 2017-05-13:
This has been implemented in the development branch.

OCB73, 2018-01-22:
Any news about implementing the multi endpoint support or a workaround?

Regards
Oliver

cdjackson, 2018-01-22:
Status is as above and also as in my reply to you on the forum yesterday.

OCB73, 2018-01-22:
Thanks for the information, last forum visit was before asking here, sorry!

cdjackson, 2018-01-22:
I’d suggest to try the dev version - if you still have problems, then we can discuss further.