openhab/org.openhab.binding.zwave #359: IllegalArgumentException during initialization

mhilbush, 2017-02-11:
I'm seeing the exception below in my log after upgrading to the latest version.
```
openhab> bundle:list -s | grep zwave
190 | Active   |  80 | 2.1.0.201702102300    | ZWave Binding                                          | org.openhab.binding.zwave
```

I've tried deleting and readding the node6 thing, as well as deleting the node6.xml.  Not sure what else to try.

```
2017-02-11 16:17:27.035 [WARN ] [.core.thing.binding.BaseThingHandler] - Error while applying configuration changes: 'IllegalArgumentException: Invalid type '{org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveSwitchAllCommandClass.SwitchAllMode}' of configuration value!' - reverting configuration changes on thing 'zwave:device:zstick:node6'.
2017-02-11 16:17:27.039 [ERROR] [ome.core.thing.internal.ThingManager] - Exception occured while initializing handler of thing 'zwave:device:zstick:node6': java.lang.IllegalArgumentException: Invalid type '{org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveSwitchAllCommandClass.SwitchAllMode}' of configuration value!
java.util.concurrent.ExecutionException: java.lang.IllegalArgumentException: Invalid type '{org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveSwitchAllCommandClass.SwitchAllMode}' of configuration value!
	at java.util.concurrent.FutureTask.report(FutureTask.java:122)[:1.8.0_121]
	at java.util.concurrent.FutureTask.get(FutureTask.java:206)[:1.8.0_121]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.callAsynchronous(SafeMethodCaller.java:194)[98:org.eclipse.smarthome.core:0.9.0.201702100936]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:83)[98:org.eclipse.smarthome.core:0.9.0.201702100936]
	at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:67)[98:org.eclipse.smarthome.core:0.9.0.201702100936]
	at org.eclipse.smarthome.core.thing.internal.ThingManager$9.run(ThingManager.java:698)[105:org.eclipse.smarthome.core.thing:0.9.0.201702100936]
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)[:1.8.0_121]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)[:1.8.0_121]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)[:1.8.0_121]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)[:1.8.0_121]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)[:1.8.0_121]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)[:1.8.0_121]
	at java.lang.Thread.run(Thread.java:745)[:1.8.0_121]
Caused by: java.lang.IllegalArgumentException: Invalid type '{org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveSwitchAllCommandClass.SwitchAllMode}' of configuration value!
	at org.eclipse.smarthome.config.core.ConfigUtil.normalizeType(ConfigUtil.java:78)[94:org.eclipse.smarthome.config.core:0.9.0.201702100936]
	at org.eclipse.smarthome.config.core.ConfigUtil.normalizeType(ConfigUtil.java:56)[94:org.eclipse.smarthome.config.core:0.9.0.201702100936]
	at org.eclipse.smarthome.config.core.Configuration.setProperties(Configuration.java:184)[94:org.eclipse.smarthome.config.core:0.9.0.201702100936]
	at org.eclipse.smarthome.core.thing.binding.BaseThingHandler.updateConfiguration(BaseThingHandler.java:475)[105:org.eclipse.smarthome.core.thing:0.9.0.201702100936]
	at org.eclipse.smarthome.core.thing.binding.ConfigStatusThingHandler.updateConfiguration(ConfigStatusThingHandler.java:65)[105:org.eclipse.smarthome.core.thing:0.9.0.201702100936]
	at org.openhab.binding.zwave.handler.ZWaveThingHandler.updateNodeProperties(ZWaveThingHandler.java:1408)[190:org.openhab.binding.zwave:2.1.0.201702102300]
	at org.openhab.binding.zwave.handler.ZWaveThingHandler.bridgeStatusChanged(ZWaveThingHandler.java:457)[190:org.openhab.binding.zwave:2.1.0.201702102300]
	at org.openhab.binding.zwave.handler.ZWaveThingHandler.initialize(ZWaveThingHandler.java:150)[190:org.openhab.binding.zwave:2.1.0.201702102300]
	at org.eclipse.smarthome.core.thing.internal.ThingManager$9$1.call(ThingManager.java:701)[105:org.eclipse.smarthome.core.thing:0.9.0.201702100936]
	at org.eclipse.smarthome.core.thing.internal.ThingManager$9$1.call(ThingManager.java:1)[105:org.eclipse.smarthome.core.thing:0.9.0.201702100936]
	at org.eclipse.smarthome.core.common.SafeMethodCaller$CallableWrapper.call(SafeMethodCaller.java:181)[98:org.eclipse.smarthome.core:0.9.0.201702100936]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)[:1.8.0_121]
	... 3 more
```

I also saw this commit, which involves SwitchAll
https://github.com/openhab/org.openhab.binding.zwave/commit/f87036783ee731de5b1a2ba08dc41137bbdcad68

This is happening on a VRP03, which is one of the devices that uses FORCEVERSION in the config of several command classes, including SWITCH_ALL.  Not sure if that's related.

cdjackson, 2017-02-11:
The commit was meant to fix this but I’m trying to get all my system up to date. IT was introduced in the latest ESH as the type of the config is wrong for SWITCH_ALL. Can you try changing the SWITCH_ALL configuration for this device, and saving the thing config and seeing if that fixes things.

mhilbush, 2017-02-11:
I changed the config and saved.  It didn't throw the exception when the config was saved.  But, it threw it again when I restarted OH.

cdjackson, 2017-02-11:
Hopefully now fixed in #360 - I found another place where it was set wrong.

cdjackson, 2017-02-11:
I've triggered a rebuild - try again in 10 minutes or so.

mhilbush, 2017-02-11:
Ok.  I'll grab the new version as soon as it finishes building.

cdjackson, 2017-02-11:
Build is done... :)

mhilbush, 2017-02-11:
That resolved it.  Thanks!!

closed 2017-02-11 by mhilbush

cdjackson, 2017-02-11:
Thanks Mark.