openhab/org.openhab.binding.zwave #361: zwave 2.0, 2.1: Philio PAT02-1C not detected

rom169, 2017-02-12:
This issue list is to report bugs or enhancements for the openHAB Z-Wave binding. If you're not sure about something, please consider discussing it on the [forum](http://community.openhab.org) before raising an issue - more people follow the forum than monitor the issues list, so if you are having trouble with a device, or want to ask a question, you will have a much larger audience there.

In order to ensure we have the best information to deal with the issue, please fill in the following information (delete as necessary depending on the issue).

* OH2 Version:
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console):

If the problem is associated with a device, please provide the following -:

* Device name:
* Device version:
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list):
* If the device is not in the database, provide a link to the manual.
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.

Provide a full debug level log of the problem. Preferably, you should try use the [log viewer](http://www.cd-jackson.com/index.php/openhab/zwave-log-viewer) to see if you can identify the problem.

If a device will not initialise, ensure the log is provided from the time OH starts.

Thank you.

cdjackson, 2017-02-12:
Please provide information requested.

rom169, 2017-02-12:
Added in the first post.

cdjackson, 2017-02-12:
Thanks.

The reason it doesn't work is because of the version. Whoever added the data to the database set a version constraint which I suspect is not needed.

Can you double check that the parameters in the database look the same as what's in your manual and if so I'll remove this constraint and it should work.

rom169, 2017-02-12:
OK, will do it the next couple of days. There are some comments in the database that parameter descriptions are too long. Do you want me to send you a suggestion for schorter descriptions, so you can copy the actual descriptions to the overview field?

cdjackson, 2017-02-12:
Thanks.

It would be great if you wanted to update the database to clean these things up. If you register for an account I’ll give you access to edit directly if you like - it’s easier for us both I think :)

rom169, 2017-02-13:
Yes, sounds good. Is this via the normal Github account or an additional account?
I will be on a business trip this week and try to find some time next weekend.

cdjackson, 2017-02-13:
It's a separate account at www.cd-Jackson.com. Just let me know once you register and I'll update your access rights.

rom169, 2017-02-13:
Account for user rom169 created and activated.

cdjackson, 2017-02-13:
And updated with write access - thanks.

rom169, 2017-02-17:
Hi Chris,

I had a look on the database today and I think it's pretty straight 
forward, just a view questions occurred:

Endpoint-Command Classes:
If I do not have NIF information of a device, which ones to tick in the 
NIF Column? All beside default?
If a switch device supports BASIC and SWITCH_BINARY command class, which 
one to tick in the basic column?

Will my changes automatically be in the next build, or do you release my 
changes. I have about 10 different devices and I'm thinking about to 
check all of them for data base error messages.

Should I upload also the devices manuals? Copyrights?

If I have a sensor like the attached one, which has three different 
sensor variants, would it be better to have one database entry with 3 
types/ids or three database entries?

Regards
Andreas

cdjackson, 2017-02-17:
> On 17 Feb 2017, at 21:27, rom169 <notifications@github.com> wrote:
> 
> Hi Chris,
> 
> I had a look on the database today and I think it's pretty straight 
> forward, just a view questions occurred:
> 
> Endpoint-Command Classes:
> If I do not have NIF information of a device, which ones to tick in the 
> NIF Column? All beside default?

To be honest, this isn’t used anywhere - it’s really for information (as a developer I like to know sometimes for debugging, but the system doesn’t use this). Normally this is filled in from the XML file, but if you don’t have the info, don’t worry too much about ticking these...

> If a switch device supports BASIC and SWITCH_BINARY command class, which 
> one to tick in the basic column?

SWITCH_BINARY gets the basic tick - it means when we receive a BASIC, this is treated a SWITCH_BINARY.

> 
> Will my changes automatically be in the next build, or do you release my 
> changes. I have about 10 different devices and I'm thinking about to 
> check all of them for data base error messages.

I have to load them “manually” - it’s really semi auto - the files are generated automatically, but I need to manually do the PR. If you want to export yourself and create a PR you can (it can be quicker for me to process ;) ).

> Should I upload also the devices manuals? Copyrights?

I normally do - I’ve not worried too much about copywrite - my assumption is that since manufacturers are trying to sell stuff, and we’re just distributing data already available in the public domain, it shouldn’t be an issue. It’s certainly really good to have the manuals.

> If I have a sensor like the attached one, which has three different 
> sensor variants, would it be better to have one database entry with 3 
> types/ids or three database entries?

Ummm… If the device is sold as 3 variants (A/B/C?) and each variant has its own set of device type/id, then it’s probably best as 3 different devices. I’m not 100% sure though. From a quick look, if it’s really something like flood sensor, thermometer and motion sensor, then probably 3 devices.

rom169, 2017-02-18:
So for today I updated the following devices in the DB and requested 
approval:
Philio PAT02 all three A/B/C
Philio PAN11
Fibaro FGD211 (Firmware Versions: All after 2.1)
Everspring an179 (here I uploaded just a better manual)

If I understand you correctly, you want me to export the changed devices 
as OH2 xml and post this in a PR in github?
Can this happen before the change in the database is approved?

Just want to make sure I understand the process before updating the 
other 7 devices I own.

cdjackson, 2017-02-18:
Thanks - I noticed you have been busy.

Don’t worry about the export - I’ll do this shortly (i.e. now) - it will be easier for you :) To answer your question though - yes - the export will reflect the current state of the database even if not approved.

cdjackson, 2017-02-18:
I’ve merged the changes and kicked off a build - update should be done in 30 minutes. If you’re using apt-get or another package, then you’ll need to wait till tomorrow, but otherwise you can install once the build is done.

Thanks for your work.

rom169, 2017-02-18:
I'm on OH2 stable and install the newest zwave binding via bundle:update 
in karaf.

I have a view strange behaviours, e.g. some switches do not update the 
item, when triggered at the device. But I can trigger the device from 
the item. Are there dependencies between OH2 core stable and OH2 binding 
snapshot?

cdjackson, 2017-02-18:
I wasn’t aware this new feature to load snapshots was running yet, so nice.

However, yes, it’s possible that there might have been some runtime changes - I’m not 100% sure though.

The other thing to note is that due to some runtime changes I had to change the switch_all configuration, so this might mean you need to update some config - see here -:

https://community.openhab.org/t/solved-2-10-2017-zwave-binding-broke-all-switch-things/22649/24 <https://community.openhab.org/t/solved-2-10-2017-zwave-binding-broke-all-switch-things/22649/24>

rom169, 2017-02-26:
I changed to the snapshot repo and installed latest snapshot to avoid 
any unwanted effects by using core stable and zwave snapshot.

Philio PAT02-1C is detected now  (after I deleted the firmware reference 
in the db).

The strange effect with switches still exist. I looks like that all of 
my Philio devices (PAN06, PAN11, PST02C, PAT02C) do not update the 
items, when the switch is triggered at the device (e.g. door opens or 
wall switch triggered to turn on light). On the other hand, battery 
status or other measurements are reported correctly and I can trigger 
the switch via the item (PAN06, PAN11).

I get a similar effect with Fibaro FGS222. Here the switch_binary2 
updates the item correctly, switch_binary1 does not.

Should I open new issues for this topics?

During the next weeks I will go on to rework the database error messages 
of all the devices I use. Just one question here, as I have seen in the 
Habmin zwave network viewer, that battery device are associating with 
each other, which seems strange for me, as they are not always on. Is it 
correct to set battery devices in the db to "Supports Routing = yes"?

cdjackson, 2017-02-27:
>  I have seen in the Habmin zwave network viewer, that battery device are associating with
each other,

I'm not sure what you mean? Do you mean that battery devices are seen as neighbours of other battery devices? This is fine - it doesn't mean they are used for routing - routing is configured by the controller itself.

> Is it correct to set battery devices in the db to "Supports Routing = yes"?

No - you shouldn't change these - this information comes from the XML file, which in turn comes from the device. It's not actually used by the database since the binding uses the information from the device, but the database should be correct or it could be confusing.

Generally battery devices don't participate in routing since they are normally not listening.

cdjackson, 2017-03-24:
As stated above, the device is now detected so closing

closed 2017-03-24 by cdjackson