openhab/org.openhab.binding.zwave #363: Kwikset Lock Alarms

vthawkeye, 2017-02-12:
* OH2 Version: 2.0.0 RC1
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): 2.1.0.201702101707 (compiled locally with some modifications latest github checkout)
* Device name: Kwikset 910 deadbolt
* Device version: 3.37
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/283

I've done some digging into this lock into how to be notified of various events.  The follow are the 6 known alarms I receive from the device.  They are dropped because they are of unknown type:

>16:05:39.872 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 20: ALARM report - 17 = 1
>16:05:39.872 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 20: Unknown Alarm Type = 17, ignoring report.

Failed to operate

> 2017-02-08 21:03:58.563 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 20: ALARM report - 18 = 0
> 2017-02-08 21:03:58.563 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 20: Unknown Alarm Type = 18, ignoring report.

 Lock Button Pushed
 
> 16:48:41.187 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 20: ALARM report - 19 = 1
> 16:48:41.188 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 20: Unknown Alarm Type = 19, ignoring report.

Code Unlock

> 20:05:13.546 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 20: ALARM report - 21 = 1
> 20:05:13.550 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 20: Unknown Alarm Type = 21, ignoring report.

Manual Lock
 
> 19:58:38.939 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 20: ALARM report - 22 = 1
> 19:58:38.939 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 20: Unknown Alarm Type = 22, ignoring report.

 Manual Unlock
 
> 10:51:42.554 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 20: ALARM report - 24 = 1
> 10:51:42.554 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 20: Unknown Alarm Type = 24, ignoring report.

RF locked

> 10:59:01.054 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 20: ALARM report - 25 = 1
> 10:59:01.055 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 20: Unknown Alarm Type = 25, ignoring report.

Remote unlocked.

cdjackson, 2017-02-12:
I think this is probably consistent with the Abloy documentation -:

http://www.cd-jackson.com/zwave_device_uploads/321/Yale-Z-Wave-Developers-Guide.pdf

It's easy enough to handle these once the security binding is working a bit better...

vthawkeye, 2017-02-13:
Yes, appears to be mostly the same.  I can notice at least one difference.  I'll go through them when I have some time and make note of what is the same and what is different.

digitaldan, 2017-04-05:
@cdjackson curious how your thinking about handling this in the binding, I just did a quick look and it doesn't seem that easy to support.  Right now the alarm_general channel is a switch type, are you planning on breaking that out to multiple channel types to support the different lock states?  (RF, manual, etc..)

cdjackson, 2017-04-05:
Yeah - I know. I don't really understand why they did this - they should have just sent an update to the lock state...

Anyway... I've not thought about how to handle this yet. My first reaction is to provide access to the different channels, and then to require a rule to pull it together. It's not nice, but I don't really want to have to hardcode device functions into the binding. Ultimately, if we can find a way to describe the links between command classes, then we can add this to the database so we can avoid having to use rules.

The current database definition does allow multiple command classes to be linked to a single channel, but it's not flexible enough to describe the situation with these locks! I also plan to change the current definition so that we can support dynamic systems as well, so it needs some thought.

Suggestions welcome if you have any ideas ;)

digitaldan, 2017-04-05:
First off, I realize you have been living in this world far more then I have, so please excuse my naive opinions, I have just started looking at this code again :-)

>Anyway... I've not thought about how to handle this yet. My first reaction is to provide access to the different channels, and then to require a rule to pull it together. It's not nice, but I don't really want to have to hardcode device functions into the binding. 
 
What would be an example of a different channel in this case? 

Have you also thought about having an alarm_general_value (or something) channel which reports the raw number as an additional option? I guess continuing on this thought, the thing.xml file for the particular zwave device could map these values using channel options to help with GUIs like 
``` 
<options>
      <option value="24">Lock</option>
      <option value="25">Unlock</option>
  </options>
```
Are those codes actually discoverable when initializing the device btw?

>ultimately, if we can find a way to describe the links between command classes

Is the issue how do we describe the relationship between a report message from a device that should trigger a GET message back to the device?  In this case, when we get an Alarm, we want to query the lock for its state?  And also to not make a mess of the binding? :-)   Since its device by device, one thought is to store this in the thing.xml file using some notation, like having a command trigger configuration that stores an array of strings that contain a mapping like EVENT:ACTION?  For example, ALARM:DOOR_LOCK_GET or event ALARM=25:DOOR_LOCK_GET ?   I think there are some lights that would benefit from this as well (GE/Jasco).  But now that I'm writing this, I'm not sure this sound like such a great idea :-)

cdjackson, 2017-04-05:
> On 5 Apr 2017, at 22:57, Dan Cunningham <notifications@github.com> wrote:
> 
> First off, I realize you have been living in this world far more then I have, so please excuse my naive opinions, I have just started looking at this code again :-)
> 
> Anyway... I've not thought about how to handle this yet. My first reaction is to provide access to the different channels, and then to require a rule to pull it together. It's not nice, but I don't really want to have to hardcode device functions into the binding.
> 
> What would be an example of a different channel in this case?
> 
Let’s say one channel per command class - so we need one for the LOCK, one for BASIC (maybe) one for ALARM...

> Have you also thought about having an alarm_general_value (or something) channel which reports the raw number as an additional option?
> 
Yes - that’s what I mean. Report the alarm, and then leave it to the user to tie it together with a rule

> Are those codes actually discoverable when initializing the device btw?
> 
Yes - this is what I’m getting at about dynamic channels - effectively self defining the channels directly in the binding based on the information coming from the device. The slight challenge is that for ALARM classes, there isn’t really a standard for these enumerations. In OH, we device an enum, but in the ZWave docs, the V1 alarms aren’t standardised even if most devices are actually the same……. The reason I mention this is because these locks don’t follow the “standard”.  For the later classes, it was standardised in the spec.

> Is the issue how do we describe the relationship between a report message from a device that should trigger a GET message back to the device?
> 
Yes - we’d need to find a way to describe these relationships in a flexible way so it’s usable for more than just this class. There are a few other cases where such a thing is useful (there’s a remote control that sends central_scene to activate a scene, but then sends basic when the user cancels the scene.

> In this case, when we get an Alarm, we want to query the lock for its state?
> 
Maybe, but in this case we probably don’t need to if we can link the alarm to the state which I think is possible here (but I’d need to check).

> And also to not make a mess of the binding? :-) Since its device by device, one thought is to store this in the thing.xml file using some notation, like having a command trigger configuration that [...]
No - this is the idea I also have. I think I said earlier to define this in the database - that’s my way of saying “in the XML” since I stick everything in the database and then export it from there….

It needs some thoughts, but what you’ve written here is broadly inline with what I’m thinking - it’s just a question of how ;) (well, and when!)

cdjackson, 2017-04-16:
I think for now I will close this with the implementation of the alarm_number channel which allows rules to get the different command codes.

Any objections, please yell and we can look at opening if there's a proposal for alternatives.

closed 2017-04-16 by cdjackson