openhab/org.openhab.binding.zwave #367: [dev] Error handling multi-channel transactinos

pizerg, 2017-02-21:
Transaction manager cannot verify properly Multi Channel reports sometimes preventing successful device initialization (as could be part of static stages), see the following log:

```
2017-02-21 16:44:12,906 ZWaveNode [Thread-4] DEBUG: NODE 3: Decapsulating COMMAND_CLASS_MULTI_CHANNEL
2017-02-21 16:44:12,906 ZWaveNode [Thread-4] DEBUG: NODE 3: Incoming command class COMMAND_CLASS_MULTI_CHANNEL
2017-02-21 16:44:12,906 ZWaveNode [Thread-4] DEBUG: NODE 3: SECURITY not supported
2017-02-21 16:44:12,906 ZWaveCommandClass [Thread-4] DEBUG: NODE 3: Received COMMAND_CLASS_MULTI_CHANNEL V2 MULTI_CHANNEL_ENCAP
2017-02-21 16:44:12,906 ZWaveMultiInstanceCommandClass [Thread-4] DEBUG: NODE 3: Requested Command Class = COMMAND_CLASS_METER (0x32)
2017-02-21 16:44:12,906 ZWaveMultiInstanceCommandClass [Thread-4] DEBUG: NODE 3: Endpoint = 1, calling handleApplicationCommandRequest.
2017-02-21 16:44:12,906 ZWaveCommandClass [Thread-4] DEBUG: NODE 3: Received COMMAND_CLASS_METER V3 METER_SUPPORTED_REPORT
2017-02-21 16:44:12,906 ZWaveMeterCommandClass [Thread-4] DEBUG: NODE 3: Identified meter type Electric(1)
2017-02-21 16:44:12,907 ZWaveMeterCommandClass [Thread-4] DEBUG: NODE 3: Meter Scale = kWh(0)
2017-02-21 16:44:12,907 ZWaveMeterCommandClass [Thread-4] DEBUG: NODE 3: Meter Scale = W(2)
2017-02-21 16:44:12,907 ZWaveMeterCommandClass [Thread-4] DEBUG: NODE 3: Meter Scale = V(4)
2017-02-21 16:44:12,907 ZWaveMeterCommandClass [Thread-4] DEBUG: NODE 3: Meter Scale = A(5)
2017-02-21 16:44:12,907 ZWaveTransactionManager [Thread-4] DEBUG: NODE 3: Commands processed 1.
2017-02-21 16:44:12,907 ZWaveTransactionManager [Thread-4] DEBUG: NODE 3: Checking command COMMAND_CLASS_MULTI_CHANNEL.
2017-02-21 16:44:12,907 ZWaveTransactionManager [Thread-4] DEBUG: NODE 3: Checking transaction 26  ApplicationCommandHandler.
2017-02-21 16:44:12,907 ZWaveTransactionManager [Thread-4] DEBUG: NODE 3: Checking transaction : state >> WAIT_DATA
2017-02-21 16:44:12,907 ZWaveTransactionManager [Thread-4] DEBUG: NODE 3: Checking transaction : node  >> 3
2017-02-21 16:44:12,907 ZWaveTransactionManager [Thread-4] DEBUG: NODE 3: Checking transaction : class >> 96 == 50.
2017-02-21 16:44:12,907 ZWaveTransactionManager [Thread-4] DEBUG: NODE 3: Checking transaction : commd >> 13 == 4.
2017-02-21 16:44:12,907 ZWaveTransactionManager [Thread-4] WARN : NODE 3: Command NOT verified org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@4fe140c3.
2017-02-21 16:44:12,907 ZWaveTransactionManager [Thread-4] WARN : Transaction completed - outstandingTransactions 1
2017-02-21 16:44:12,907 ZWaveTransactionManager [Thread-4] WARN : Transaction completed - outstandingTransactions 1
2017-02-21 16:44:12,907 ZWaveTransactionManager [Thread-4] DEBUG: ZWaveReceiveThread queue empty
```

We can see it tries to match command class 96 (60 hex -> COMMAND_CLASS_MULTI_CHANNEL) with the decapsulated expected response command class 50 (32 hex -> COMMAND_CLASS_METER).
I think the problem is in "processCommand(ZWaveCommandClassPayload payload)" from ZWaveNode class, maybe the implementation is still pending as there's only a log line for some of the encapsulated commands.

cdjackson, 2017-02-25:
Thanks @pizerg I've pushed an update that I hope will handle encapsulated classes properly. It needs a tidy, but hopefully should resolve this.

pizerg, 2017-02-27:
Just added your update, will be testing it during the day and post the feedback asap. Thanks!

pizerg, 2017-03-03:
I've been performing some testing, multichannel now seems to be working OK but I think there's a fix that was missed somewhere with CRC16 (#319) and it's not handling them properly. Also related to #375 when the CRC16 encap handling fails in this piece of code (ZWaveCRC16EncapsulationCommandClass.java):

```
if (!Arrays.equals(messageCrc, byteBuffer.array())) {
    logger.info("NODE {}: CRC check failed message contains {} but should be {}", getNode().getNodeId(),
    SerialMessage.bb2hex(messageCrc), SerialMessage.bb2hex(byteBuffer.array()));

    return null;
}
```

The return null statement makes  ZWaveNode.java :: processCommand(ZWaveCommandClassPayload payload) crash "silently" due to a non controlled NPE.

pizerg, 2017-03-07:
To prevent this from crashing I've added an NPE protection in ZWaveNode :: processCommand(ZWaveCommandClassPayload payload) (but I cannot be sure if the issue was related to #319 or to an actual bad CRC check due to a network or sensor issue):

```
payload = crcCommandClass.handleCrcEncap(payload);

if(payload == null){
    return null;
}
```

Regards,

Adrián

cdjackson, 2017-04-13:
I've implemented the same NPE check - this is the correct solution in the event that there really is a CRC error.
This will be closed when development merges into master

closed 2017-06-24 by cdjackson