openhab/org.openhab.binding.zwave #369: [dev] dynamic states wrongly requested for security devices

pizerg, 2017-02-22:
doDynamicStages() in ZWaveNodeInitStageAdvancer does not encapsulate transactions when "instances == 1" so securely linked devices will just ignore the non-encrypted messages.

Adrián

cdjackson, 2017-02-27:
By "instances == 1" do you mean the multi-instance class V1, or multi-instance and multi-channel?

pizerg, 2017-02-27:
I actually meant a pice of code inside doDynamicStages():

```
if (instances == 1) {
     processTransactions(zdds.getDynamicValues(true));
} else {
     for (int i = 1; i <= instances; i++) {
            processTransactions(zdds.getDynamicValues(true), zwaveDynamicClass, i);
            if (initRunning == false) {
                   return;
            }
     }
}
```

I've also seen this code in doStaticStages() and I suppose it might present the same error with security devices.

cdjackson, 2017-04-14:
I don't really like the way the older multi-instance class is handled... Anyway, that's a question for another day...

I hope I've fixed this now - if instances = 1, then I use the endpointId. Instances should always be 1 unless this is a multi-instance class, and then endpointId should only ever be 0. If it's multi-channel, then endpointId can be any number.

I've not tested this with security so comments welcome.

pizerg, 2017-04-24:
I've run a couple of tests and now it's working as expected.

closed 2017-04-24 by pizerg