openhab/org.openhab.binding.zwave #375: [dev] "silent" NPE makes ZWaveReceiveThread crash

pizerg, 2017-02-27:
When a transaction is not correlated (so "currentTransaction" variable is null) then when  

`controller.handleIncomingMessage(currentTransaction, incomingMessage);`  

is called, then calls 

`handleIncomingResponseMessage(currentTransaction, incomingMessage);` 

or 

`handleIncomingRequestMessage(currentTransaction, incomingMessage);` 

which finally calls a handler in the specific message class (in my case I saw this behaviour with DeleteSUCReturnRoute),  some of the handlers use the transaction to get the node id like this 

`int nodeId = transaction.getSerialMessage().getMessagePayloadByte(0);` 

which makes the thread crash due to an uncontrolled NPE preventing any new incoming data to be processed.

cdjackson, 2017-02-28:
Thanks - I'll take a look at this tonight. I know some messages are protected against this, but I guess not all...

cdjackson, 2017-02-28:
I've added protection to hopefully all the classes this can happen in.

pizerg, 2017-03-07:
Protection added to RemoveNodeMessageClass is preventing the unlinking process to complete as expected, for some reason the last step finds no transaction to correlate and due to NPE protection does not notify the status to the controller who then does not proceed to remove the device and its associated xml file. 

This can be reproduced with any device, just link and then unlink to see that it's not linked anymore but the system does not notice it and "NODE {}: Node not found - has this node been removed?!?" logging message appears.

cdjackson, 2018-09-08:
Closing as this is very old. If it still exists in the latest version then please reopen.

closed 2018-09-08 by cdjackson