openhab/org.openhab.binding.zwave #377: ZWaveReceiveThread crashes

vthawkeye, 2017-02-28:
There are some NPE exceptions that cause the ZWaveReceiveThread in ZWaveTransactionManager to crash.

I added a try/catch block around the code in ZWaveTransactionManager.ZWaveReceiveThread.run() and found at least three NPEs:

- As noted in https://github.com/openhab/org.openhab.binding.zwave/issues/375 there is the same issue in DeleteReturnRouteMessageClass.  I used incomingMessage.getMessageNode() to get the node id instead of transaction.getSerialMessage().getMessagePayloadByte(0).  Not sure if this is correct, but seems to work.
- In ZWaveNode.processCommand() 1224 there was an NPE on using the endpoint variable.  I added a null check like the commandClass check right above that.
- In the ZWaveTransactionManager.ZWaveReceiveThread.run() method itself,  in the "Check for NAK/CAN" switch statement lastTransaction was null in the CAN case on line 384.  I added a null check on lastTransaction in this block.

Once I added these changes, I haven't seen anymore Exceptions.

If wanted I can provide the stacktraces and/or patches.

cdjackson, 2017-02-28:
#1 should be fixed

#2 I can't correlate - I wonder if you're using an older version? Anyway, I think I found the line you mean, and probably some others with the same issue.

#3 should be fixed

Thanks :)

I'll create an update shortly.

vthawkeye, 2017-03-01:
#2 I think it's one of my devices in particular that causes this.  I have a Aeon Labs Smart Strip.  I get errors on endpoint 5 and 6.

xsnrg, 2017-03-01:
I have noticed on my smart strips that 5 and 6 are weird.  The power usage is offset from the switch.  That is, you can find the switch at 1 and the power usage at 3, or was it the other way around...  been a while.  Anyway, by the time you get to 5 and 6, they don't actually exist, or at least are not addressable even though the endpoints appear to be there.

cdjackson, 2017-03-24:
Believed to be fixed so closing.

closed 2017-03-24 by cdjackson