openhab/org.openhab.binding.zwave #381: [dev] Security message handling for "marked as failed by controller" devices

pizerg, 2017-03-03:
I found an issue while testing Aeon's Door/Window Sensor Gen 6 linked securely. It was working for a few days OK, then it ran out of battery and the controller marked is as failed, no problem here. 

I charged the device and restarted the binding (I just had to reboot my system), after initializing the binding I noticed that the device was still stuck in FAILED_CHECK init stage (even if it was reporting proper values, requesting NONCE etc), for some reason the controller did not mark it as not-failed even if it was receiving values from it, nothing we can do about it as the controller has its own algorithms to handle which nodes are considered failed.

If it was non-securely linked the values reported by the device would be process normally and no values would be lost even if being stuck in FAILED_CHECK, but it's a security device and the decryption key is only initialized during the following init stages, so every time we get a report from the sensor this log message appears and the value is lost as it cannot be deciphered: 

`Node {}: SECURITY_ERR invalid key: {}`

cdjackson, 2017-03-03:
I would have expected that in the secure condition, when the devices wakes up, it sends a nonce_request. I guess your point is that the binding doesn't process this because the key is set later in the initialisation stage (??) however I would have expected that the controller now sees this device as alive again? Is this the case or not?

If yes, then while this request will fail, the next fail_check should pass, and the initialisation should complete (and the key should get set). So, we loose one packet of data, but given you presumably had this device offline for a while to recharge it, I don't see that this is a problem?

Or, does the controller never change this device to alive? I guess that this is really the problem you're raising and we need a way to get past this - correct?

pizerg, 2017-03-06:
It's the second option, one would expect that after a NONCE_REQUEST or any other kind of message the controller would set the device as alive, but apparently it's not what it's doing, despite receiving messages from the previously failed device the controller does not mark it as alive again (messages re-checked with zniffer)

pizerg, 2017-03-07:
Maybe we alse need to consider the case where the node is NOT marked as failed but is a sleeping device so it's possible it waits in the next stage (REQUEST_NIF) and loses the next report when the device wakes up as the network key is not set until security stages.

For most devices this might not be a big issue, but in case of a door/window opened/closed or similar it would be good if we can avoid that message from being lost.

cdjackson, 2018-09-08:
Closing as this is very old and should be fixed in the latest development version as the whole dead node concept was changed not to use the failed check.

closed 2018-09-08 by cdjackson