openhab/org.openhab.binding.zwave #390: Everspring AN157 issues with OH2.0.0

4integration, 2017-03-14:
HI,

Upgraded to OH2 final (previously Beta 4) and I have problems with AN157 plugs not responding.
Any tip on how to solve it?

* OH2 Version: OH 2.0.0
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): 2.0.0

If the problem is associated with a device, please provide the following -:

* Device name: Everspring AN157-2
* Device version: 
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/26

**Log when seaching + adding AN157 thing:**

> 19:55:51.449 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - Initializing ZWave thing handler.
19:55:51.450 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:1b7d2486:node6' changed from UNINITIALIZED (HANDLER_MISSING_ERROR) to INITIALIZING
19:55:51.453 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Controller status changed to ONLINE.
19:55:51.453 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Controller is ONLINE. Starting device initialisation.
19:55:51.455 [INFO ] [smarthome.event.ThingUpdatedEvent   ] - Thing 'zwave:device:1b7d2486:node6' has been updated.
19:55:51.457 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Command received zwave:device:1b7d2486:node6:switch_binary --> REFRESH
19:55:51.458 [WARN ] [ding.zwave.handler.ZWaveThingHandler] - Controller handler not found. Cannot handle command without ZWave controller.
19:55:51.459 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Updating node properties.
19:55:51.459 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Updating node properties. MAN=96
19:55:51.459 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:1b7d2486:node6' changed from INITIALIZING to OFFLINE (BRIDGE_OFFLINE): Controller is offline
19:55:51.460 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Updating node properties. MAN=96. SET. Was 96
19:55:51.466 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:1b7d2486:node6' changed from OFFLINE (BRIDGE_OFFLINE): Controller is offline to ONLINE
19:55:51.468 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:1b7d2486:node6' changed from ONLINE to UNINITIALIZED (HANDLER_INITIALIZING_ERROR)
19:55:51.466 [ERROR] [ome.core.thing.internal.ThingManager] - Exception occured while initializing handler of thing 'zwave:device:1b7d2486:node6': java.lang.NullPointerException
java.util.concurrent.ExecutionException: java.lang.NullPointerException
        at java.util.concurrent.FutureTask.report(FutureTask.java:122)[:1.8.0_65]
        at java.util.concurrent.FutureTask.get(FutureTask.java:206)[:1.8.0_65]
        at org.eclipse.smarthome.core.common.SafeMethodCaller.callAsynchronous(SafeMethodCaller.java:188)[101:org.eclipse.smarthome.core:0.9.0.b4]
        at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:81)[101:org.eclipse.smarthome.core:0.9.0.b4]
        at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:65)[101:org.eclipse.smarthome.core:0.9.0.b4]
        at org.eclipse.smarthome.core.thing.internal.ThingManager$9.run(ThingManager.java:710)[108:org.eclipse.smarthome.core.thing:0.9.0.b4]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)[:1.8.0_65]
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)[:1.8.0_65]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)[:1.8.0_65]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)[:1.8.0_65]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)[:1.8.0_65]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)[:1.8.0_65]
        at java.lang.Thread.run(Thread.java:745)[:1.8.0_65]
Caused by: java.lang.NullPointerException
        at org.openhab.binding.zwave.handler.ZWaveThingHandler.updateNodeProperties(ZWaveThingHandler.java:1353)[196:org.openhab.binding.zwave:2.0.0]
        at org.openhab.binding.zwave.handler.ZWaveThingHandler.bridgeStatusChanged(ZWaveThingHandler.java:457)[196:org.openhab.binding.zwave:2.0.0]
        at org.openhab.binding.zwave.handler.ZWaveThingHandler.initialize(ZWaveThingHandler.java:150)[196:org.openhab.binding.zwave:2.0.0]
        at org.eclipse.smarthome.core.thing.internal.ThingManager$9$1.call(ThingManager.java:713)[108:org.eclipse.smarthome.core.thing:0.9.0.b4]
        at org.eclipse.smarthome.core.thing.internal.ThingManager$9$1.call(ThingManager.java:1)[108:org.eclipse.smarthome.core.thing:0.9.0.b4]
        at org.eclipse.smarthome.core.common.SafeMethodCaller$CallableWrapper.call(SafeMethodCaller.java:179)[101:org.eclipse.smarthome.core:0.9.0.b4]
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)[:1.8.0_65]
        ... 3 more
19:55:56.137 [ERROR] [ocol.ZWaveController$ZWaveSendThread] - NODE 10: Timeout while sending message. Requeueing - 2 attempts left!
19:55:56.147 [ERROR] [l.serialmessage.SendDataMessageClass] - NODE 10: Got an error while sending data. Resending message.
19:55:56.148 [DEBUG] [ve.internal.protocol.ZWaveController] - Message queued. Queue length = 65. Queue={}
19:55:56.149 [DEBUG] [ocol.ZWaveController$ZWaveSendThread] - Took message from queue for sending. Queue length = 64

cdjackson, 2017-03-14:
Please use the snapshot version - the “final” version is quite old and I think this has been fixed.

4integration, 2017-03-14:
I installed the latest #335  but got

> 00:13:15.723 [DEBUG] [org.openhab.binding.zwave           ] - ServiceEvent REGISTERED - {org.eclipse.smarthome.config.core.status.ConfigStatusProvider}={service.id=355, service.bundleid=197, service.scope=singleton} - org.openhab.binding.zwave
00:13:15.723 [INFO ] [smarthome.event.ThingAddedEvent     ] - Thing 'zwave:device:1b7d2486:node6' has been added.
00:13:15.742 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - Initializing ZWave thing handler.
00:13:15.747 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:1b7d2486:node6' changed from UNINITIALIZED to INITIALIZING
00:13:15.758 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:1b7d2486:node6' changed from INITIALIZING to OFFLINE (BRIDGE_OFFLINE): Controller is offline
00:13:15.776 [DEBUG] [org.openhab.binding.zwave           ] - ServiceEvent UNREGISTERING - {org.eclipse.smarthome.config.core.status.ConfigStatusProvider}={service.id=355, service.bundleid=197, service.scope=singleton} - org.openhab.binding.zwave
00:13:15.782 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - Handler disposed. Unregistering listener.
00:13:15.785 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:1b7d2486:node6' changed from OFFLINE (BRIDGE_OFFLINE): Controller is offline to UNINITIALIZED (HANDLER_MISSING_ERROR)
00:13:15.846 [DEBUG] [org.openhab.binding.zwave           ] - ServiceEvent REGISTERED - {org.eclipse.smarthome.config.core.status.ConfigStatusProvider}={service.id=356, service.bundleid=197, service.scope=singleton} - org.openhab.binding.zwave
00:13:15.865 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - Initializing ZWave thing handler.
00:13:15.867 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:1b7d2486:node6' changed from UNINITIALIZED (HANDLER_MISSING_ERROR) to INITIALIZING
00:13:15.868 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Controller status changed to ONLINE.
00:13:15.868 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Controller is ONLINE. Starting device initialisation.
00:13:15.869 [INFO ] [smarthome.event.ThingUpdatedEvent   ] - Thing 'zwave:device:1b7d2486:node6' has been updated.
00:13:15.870 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Updating node properties.
00:13:15.871 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Updating node properties. MAN=96
00:13:15.871 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Updating node properties. MAN=96. SET. Was 96
00:13:15.874 [ERROR] [ome.core.thing.internal.ThingManager] - Exception occured while initializing handler of thing 'zwave:device:1b7d2486:node6': java.lang.NullPointerException
java.util.concurrent.ExecutionException: java.lang.NullPointerException
        at java.util.concurrent.FutureTask.report(FutureTask.java:122)[:1.8.0_65]
        at java.util.concurrent.FutureTask.get(FutureTask.java:206)[:1.8.0_65]
        at org.eclipse.smarthome.core.common.SafeMethodCaller.callAsynchronous(SafeMethodCaller.java:188)[101:org.eclipse.smarthome.core:0.9.0.b4]
        at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:81)[101:org.eclipse.smarthome.core:0.9.0.b4]
        at org.eclipse.smarthome.core.common.SafeMethodCaller.call(SafeMethodCaller.java:65)[101:org.eclipse.smarthome.core:0.9.0.b4]
        at org.eclipse.smarthome.core.thing.internal.ThingManager$9.run(ThingManager.java:710)[108:org.eclipse.smarthome.core.thing:0.9.0.b4]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)[:1.8.0_65]
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)[:1.8.0_65]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180)[:1.8.0_65]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293)[:1.8.0_65]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)[:1.8.0_65]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)[:1.8.0_65]
        at java.lang.Thread.run(Thread.java:745)[:1.8.0_65]
Caused by: java.lang.NullPointerException
        at org.openhab.binding.zwave.handler.ZWaveThingHandler.updateNodeProperties(ZWaveThingHandler.java:1353)[197:org.openhab.binding.zwave:2.1.0.201703131855]
        at org.openhab.binding.zwave.handler.ZWaveThingHandler.bridgeStatusChanged(ZWaveThingHandler.java:457)[197:org.openhab.binding.zwave:2.1.0.201703131855]
        at org.openhab.binding.zwave.handler.ZWaveThingHandler.initialize(ZWaveThingHandler.java:150)[197:org.openhab.binding.zwave:2.1.0.201703131855]
        at org.eclipse.smarthome.core.thing.internal.ThingManager$9$1.call(ThingManager.java:713)[108:org.eclipse.smarthome.core.thing:0.9.0.b4]
        at org.eclipse.smarthome.core.thing.internal.ThingManager$9$1.call(ThingManager.java:1)[108:org.eclipse.smarthome.core.thing:0.9.0.b4]
        at org.eclipse.smarthome.core.common.SafeMethodCaller$CallableWrapper.call(SafeMethodCaller.java:179)[101:org.eclipse.smarthome.core:0.9.0.b4]
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)[:1.8.0_65]
        ... 3 more
00:13:15.887 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:1b7d2486:node6' changed from ONLINE to UNINITIALIZED (HANDLER_INITIALIZING_ERROR)
00:13:15.891 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:1b7d2486:node6' changed from INITIALIZING to OFFLINE (BRIDGE_OFFLINE): Controller is offline
00:13:15.900 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:1b7d2486:node6' changed from OFFLINE (BRIDGE_OFFLINE): Controller is offline to ONLINE
00:13:17.438 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 10: Stage PING. Initialisation retry timer triggered. Increased to 80000

cdjackson, 2017-03-15:
What device is this happening on?

4integration, 2017-03-15:
The device is Everspring AN157 controlled by openHAB 2.0.0 on Raspberry PI with Aeotec Z-wave USB Stick.
Did that answer your question?

4integration, 2017-05-13:
Any news on this? 
Tested with latest snapshot and still getting NullPointerException when adding device:
`14:13:27.103 [DEBUG] [ve.internal.protocol.ZWaveController] - ZWave controller start inclusion - mode 2
14:13:27.108 [DEBUG] [ol.serialmessage.AddNodeMessageClass] - Setting controller into INCLUSION mode, highPower:true networkWide:true.
14:13:27.110 [DEBUG] [ve.internal.protocol.ZWaveController] - Message queued. Queue length = 1. Queue={}
14:13:27.110 [DEBUG] [ocol.ZWaveController$ZWaveSendThread] - Took message from queue for sending. Queue length = 0
14:13:27.111 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 05 00 4A C1 01 70
14:13:27.112 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 255: Sending REQUEST Message = 01 05 00 4A C1 01 70
14:13:27.119 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 07 00 4A 01 01 00 00 B2
14:13:27.125 [DEBUG] [ve.internal.protocol.ZWaveController] - Receive queue TAKE: Length=0
14:13:27.127 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 07 00 4A 01 01 00 00 B2
14:13:27.130 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 07 00 4A 01 01 00 00 B2
14:13:27.132 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=AddNodeToNetwork[0x4A], type=Request[0x00], priority=High, dest=255, callback=0, payload=01 01 00 00
14:13:27.134 [DEBUG] [ol.serialmessage.AddNodeMessageClass] - Add Node: Learn ready.
14:13:27.136 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveInclusionEvent
14:13:27.153 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: Sent Message: class=AddNodeToNetwork[0x4A], type=Request[0x00], priority=High, dest=255, callback=0, payload=C1 01
14:13:27.151 [INFO ] [smarthome.event.BindingEvent        ] - org.openhab.binding.zwave.event.BindingEvent@6cd0e6
14:13:27.155 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: Recv Message: class=AddNodeToNetwork[0x4A], type=Request[0x00], priority=High, dest=255, callback=0, payload=01 01 00 00
14:13:27.156 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: class=AddNodeToNetwork, callback id=0, expected=AddNodeToNetwork, cancelled=false        transaction complete!
14:13:27.157 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveTransactionCompletedEvent
14:13:27.159 [DEBUG] [ocol.ZWaveController$ZWaveSendThread] - NODE 255: Response processed after 43ms/4938ms.
14:13:29.711 [DEBUG] [al.protocol.ip.MySensorsIpConnection] - Bridge is connected, connection skipped
14:13:39.712 [DEBUG] [al.protocol.ip.MySensorsIpConnection] - Bridge is connected, connection skipped
14:13:44.843 [INFO ] [smarthome.event.InboxRemovedEvent   ] - Discovery Result with UID 'zwave:device:1b7d2486:node6' has been removed.
14:13:44.851 [INFO ] [smarthome.event.ThingAddedEvent     ] - Thing 'zwave:device:1b7d2486:node6' has been added.
14:13:44.868 [DEBUG] [org.openhab.binding.zwave           ] - ServiceEvent REGISTERED - {org.eclipse.smarthome.config.core.status.ConfigStatusProvider}={service.id=318, service.bundleid=201, service.scope=singleton} - org.openhab.binding.zwave
14:13:44.908 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - Initializing ZWave thing handler.
14:13:44.913 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:1b7d2486:node6' changed from UNINITIALIZED to INITIALIZING
14:13:44.921 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:1b7d2486:node6' changed from INITIALIZING to OFFLINE (BRIDGE_OFFLINE): Controller is offline
14:13:44.987 [DEBUG] [org.openhab.binding.zwave           ] - ServiceEvent UNREGISTERING - {org.eclipse.smarthome.config.core.status.ConfigStatusProvider}={service.id=318, service.bundleid=201, service.scope=singleton} - org.openhab.binding.zwave
14:13:44.999 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:1b7d2486:node6' changed from OFFLINE (BRIDGE_OFFLINE): Controller is offline to UNINITIALIZED (HANDLER_MISSING_ERROR)
14:13:44.999 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - Handler disposed. Unregistering listener.
14:13:45.090 [DEBUG] [org.openhab.binding.zwave           ] - ServiceEvent REGISTERED - {org.eclipse.smarthome.config.core.status.ConfigStatusProvider}={service.id=319, service.bundleid=201, service.scope=singleton} - org.openhab.binding.zwave
14:13:45.125 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - Initializing ZWave thing handler.
14:13:45.136 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:1b7d2486:node6' changed from UNINITIALIZED (HANDLER_MISSING_ERROR) to INITIALIZING
14:13:45.137 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Controller status changed to ONLINE.
14:13:45.139 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Controller is ONLINE. Starting device initialisation.
14:13:45.142 [INFO ] [me.event.ThingStatusInfoChangedEvent] - 'zwave:device:1b7d2486:node6' changed from INITIALIZING to OFFLINE (BRIDGE_OFFLINE): Controller is offline
14:13:45.143 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 6: Command received zwave:device:1b7d2486:node6:switch_binary --> REFRESH
14:13:45.144 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - Controller handler not found. Cannot handle command without ZWave controller.
14:13:45.149 [INFO ] [smarthome.event.ThingUpdatedEvent   ] - Thing 'zwave:device:1b7d2486:node6' has been updated.
[... cut ...]
14:13:52.938 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: Recv Message: class=AddNodeToNetwork[0x4A], type=Request[0x00], priority=High, dest=255, callback=0, payload=01 06 00 00
14:13:52.938 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Checking transaction complete: class=AddNodeToNetwork, callback id=0, expected=AddNodeToNetwork, cancelled=false        transaction complete!
14:13:52.939 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveTransactionCompletedEvent
`

cdjackson, 2017-05-13:
It's pretty hard to read this log, but it looks like it's something completely different. Can you format the log so it's readable please?

4integration, 2017-05-13:
Testing to attach a file
[openhab-zwave-an157.txt](https://github.com/openhab/org.openhab.binding.zwave/files/998703/openhab-zwave-an157.txt)

4integration, 2017-05-20:
Tested to exclude these devices/things and included them again (Aeotec USB Z-stick Gen5) and then it started to work again :-)
Closing this report

closed 2017-05-20 by 4integration