openhab/org.openhab.binding.zwave #407: Issues with FGWP102-5

helmar74, 2017-03-16:
Hi,

I've trouble with a Fibaro FGWP102-5.
I can integrate the Smart Plug in my Z-Wave Network and it's also possible to integrate in OH2. I can switch the plug and measure Watts and kWh. 
This are my problems:
1. kWh are meassured wrong. I plugged in LED's which have 8 Watt power. After 20 minutes 201313 KWh are meassured. I think this is  a wrong value
2. In Habmin no configuration parameters are shown. So its not possible for example to change LED Ring or turn it off.

Screenshot of habmin:
[http://i.imgur.com/LT4M5WL.png](url)

cdjackson, 2017-03-16:
As requested, please provide a debug log.

helmar74, 2017-03-16:
how can i create a log?

cdjackson, 2017-03-16:
Take a look at the loging section in the docs.

helmar74, 2017-03-16:
OK, I'll provide a log, but I had to read the  docs first... Will take a while...

cdjackson, 2017-03-16:
Reading the docs is always useful ;)

helmar74, 2017-03-16:
of course ;-) but I don't know where i find the docs. so googling around will take time

cdjackson, 2017-03-16:
http://docs.openhab.org/administration/logging.html <http://docs.openhab.org/administration/logging.html>

helmar74, 2017-03-24:
Created a DEbug Log.
[openhab_debug.zip](https://github.com/openhab/org.openhab.binding.zwave/files/867065/openhab_debug.zip)

cdjackson, 2017-03-24:
Closing as being discussed in multiple places.

closed 2017-03-24 by cdjackson