openhab/org.openhab.binding.zwave #415: Missing zwave_deviceid in database

alphaZ, 2017-03-20:
I own a MCO Home MH7 thermostat which was not automatically recognised by OH2 but flagged as unknown device.
It turns out that the device exists in the cd-jackson database but has a different zwave_deviceid than the one I have. Since only one figure is wrong, it might be a typo, or the device is available with different zwave_deviceids.

This is the database entry:
 http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/439

As one can see there, the zwave_deviceid currently available is:
References (Type:Id)	0702:3102

My device, however has the following data:
References (Type:Id)	0702:5102

so the "3" is replaced by a "5"

What do I need to do, to
1. get this fixed / added in the database
2. make sure the updated entry is available in my OH2 installation (Raspberry Pi running openHABian)

Thanks a lot!
Rainer

cdjackson, 2017-03-20:
Please see the link below for adding devices to the database -:

http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide <http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide>

alphaZ, 2017-03-20:
It says: "Please note that you must contact the administrator to get edit rights to the database once you have registered."
I registered with the username "alphaZ"

cdjackson, 2017-03-20:
Access was updated...

cdjackson, 2017-03-24:
This was included in this PR -:
https://github.com/openhab/org.openhab.binding.zwave/pull/423/files

closed 2017-03-24 by cdjackson