openhab/org.openhab.binding.zwave #418: Fibaro FGS-213 Switch is not in the database

RafalLukawiecki, 2017-03-21:
Although the double-relay Fibaro FGS-223 is in the database, the single-relay version of the same device is not. Could you, please, add it?

* Device name: Fibaro Single/Double Switch 2 FGS-213
* Device version: Single
* Manual: http://manuals.fibaro.com/content/manuals/en/FGS-2x3/FGS-2x3-EN-T-v1.1.pdf

I cannot see an xml file for it, but I would be happy to generate one if there is a procedure I could follow. I can see XML files for all other devices in my OH1 /var/lib/openhab/zwave folder.

I have also opened [this thread](https://community.openhab.org/t/database-missing-z-wave-fibaro-fgs-213-switch/25298) on OH, but no one has replied, I am afraid.
Thank you, very much.

cdjackson, 2017-03-21:
The device is included in the database already. Maybe you have a different version, or maybe you are using an old version of the binding,

Please check the online database at www.cd-jackson.com and if your device type/id isn't there then please feel free to add them,

closed 2017-03-21 by cdjackson

RafalLukawiecki, 2017-03-21:
Thank you, Chris. I swear that when I did a search this item did not show up, but I can see it in the db now. I am still getting errors for it in OpenHAB 1.8.3, though...

Sorry for having taken your valuable time with this, and many thanks for making the DB in the first place.

cdjackson, 2017-03-21:
Ah - if you’re looking for it in OH1, then it might not exist - it would need to be exported into the OH1 repo (this issue list is for OH2)… Do you know how to create a PR? If so, it would be great if you could do so. If not I’ll take a look at it.

RafalLukawiecki, 2017-03-22:
I have ported the OH2 entry to OH1, see PR https://github.com/openhab/openhab1-addons/pull/5148. Unsure if it was supposed to go into the master branch or not.

cdjackson, 2017-03-22:
Thanks - very (very!) quick look looks fine. I'll take a better look tomorrow.

RafalLukawiecki, 2017-03-22:
Thank you, Chris. I do not know by means of what Java build magic this update becomes a .jar that is available to replace the one in my OH. When you have processed the PR, kindly let me know where the built version appears. Thanks!

cdjackson, 2017-03-22:
Will do… Thanks.