openhab/org.openhab.binding.zwave #421: Disable device polling for FLIR devices

espentj, 2017-03-23:
My POPP smoke detector and Siren states that it should never be polled as it is a FLIR device. However, when I look at the possible values for device polling on the Zwave the longest period is one day. 
http://manuals-backend.z-wave.info/make.php?lang=en&sku=pope004001&type=popp

> Quick trouble shooting
Here are a few hints for network installation if things dont work as expected.

> - Make sure a device is in factory reset state before including. In doubt exclude before include.
> - If inclusion still fails, check if both devices use the same frequency.
> - Remove all dead devices from associations. Otherwise you will see severe delays.
> - Never use sleeping battery devices without a central controller.
> - **Dont poll FLIRS devices.**
> - Make sure to have enough mains powered device to benefit from the meshing

cdjackson, 2017-03-23:
Strange - there should be absolutely no problem polling FLiRS devices at all. I would just set this to 1 day and it will be fine. I suspect they say this as it might reduce battery power if polled very often, but I see no other reason not to poll a FLiRS device - it’s certainly quite common with door locks which are also FLiRS.

espentj, 2017-03-23:
Actually battery drain is a main problem with the device atm. I am getting a few months (4) worth of battery while it should last at least 12 according to the manufacturer.

cdjackson, 2017-03-23:
This won’t be an issue with polling - 1 poll every 24 hours will not impact battery life - its a few milliseconds of transmission every day.

closed 2017-03-24 by cdjackson