openhab/org.openhab.binding.zwave #427: Reopen issue 304

antares2001, 2017-03-25:
Hello,

could you please re-open issue https://github.com/openhab/org.openhab.binding.zwave/issues/304 since it still does not seem solved.

Thank you!

cdjackson, 2017-03-25:
Ensure that you are running the latest of everything - not just the binding but also the framework.

antares2001, 2017-03-28:
I've install the latest openHAB from jFrog (currently apt-get updated version running for about a month) on an x86_64 Debian 8 when this occurred. I'll try a fully fresh install to see whether it persists.

cdjackson, 2017-03-28:
Thanks. If the problem persists please open an issue and provide full debug logs.

closed 2017-03-28 by cdjackson

robkamp, 2017-08-31:
I still have this with a fresh install.

rille, 2017-09-26:
I am having this issue as well when trying to add a Fibaro Door 2 device to my z-wave network. I get this exception every time I wake up the device in attempting to initialise the manufacturer and other attributes.

Thanks in advance for any progress or solutions around this.