openhab/org.openhab.binding.zwave #433: Exception during ZWave thread: Input 2. ArrayIndexOutOfBoundsException: 3

antares2001, 2017-03-28:
Hello @cdjackson,

here's a full report of the issue (which is not any more an NPE but an ArrayIndexOutOfBoundsException).

* OH2 Version: 
```
$ apt-cache policy openhab2
openhab2:
  Installed: 2.1.0~20170325134201-1
```

And the Stacktrace:
```
2017-03-28 15:31:42.903 [ERROR] [ve.internal.protocol.ZWaveController] - Exception during ZWave thread: Input 2. {}
java.lang.ArrayIndexOutOfBoundsException: 3
	at org.openhab.binding.zwave.internal.protocol.commandclass.ZWaveSecurityCommandClass.isSecurityNonceReportMessage(ZWaveSecurityCommandClass.java:1067)[183:org.openhab.binding.zwave:2.1.0.201702121957]
	at org.openhab.binding.zwave.internal.protocol.SerialMessage$SerialMessageComparator.compare(SerialMessage.java:692)[183:org.openhab.binding.zwave:2.1.0.201702121957]
	at org.openhab.binding.zwave.internal.protocol.SerialMessage$SerialMessageComparator.compare(SerialMessage.java:1)[183:org.openhab.binding.zwave:2.1.0.201702121957]
	at java.util.concurrent.PriorityBlockingQueue.siftUpUsingComparator(PriorityBlockingQueue.java:374)[:1.8.0_121]
	at java.util.concurrent.PriorityBlockingQueue.offer(PriorityBlockingQueue.java:491)[:1.8.0_121]
	at java.util.concurrent.PriorityBlockingQueue.add(PriorityBlockingQueue.java:463)[:1.8.0_121]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.enqueue(ZWaveController.java:546)[183:org.openhab.binding.zwave:2.1.0.201702121957]
	at org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer.sendMessage(ZWaveNodeInitStageAdvancer.java:260)[183:org.openhab.binding.zwave:2.1.0.201702121957]
	at org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer.addToQueue(ZWaveNodeInitStageAdvancer.java:1170)[183:org.openhab.binding.zwave:2.1.0.201702121957]
	at org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer.advanceNodeStage(ZWaveNodeInitStageAdvancer.java:1082)[183:org.openhab.binding.zwave:2.1.0.201702121957]
	at org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer.handleNodeQueue(ZWaveNodeInitStageAdvancer.java:230)[183:org.openhab.binding.zwave:2.1.0.201702121957]
	at org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer.ZWaveIncomingEvent(ZWaveNodeInitStageAdvancer.java:1306)[183:org.openhab.binding.zwave:2.1.0.201702121957]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.notifyEventListeners(ZWaveController.java:567)[183:org.openhab.binding.zwave:2.1.0.201702121957]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingRequestMessage(ZWaveController.java:250)[183:org.openhab.binding.zwave:2.1.0.201702121957]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingMessage(ZWaveController.java:214)[183:org.openhab.binding.zwave:2.1.0.201702121957]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.access$7(ZWaveController.java:208)[183:org.openhab.binding.zwave:2.1.0.201702121957]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController$ZWaveInputThread.run(ZWaveController.java:1324)[183:org.openhab.binding.zwave:2.1.0.201702121957]
```

cdjackson, 2017-03-28:
Probably caused by security key not being set or not correctly set looking at the error. As previously requested please revise a full debug log.

antares2001, 2017-05-30:
Issue gone after reinstall.

closed 2017-05-30 by antares2001