openhab/org.openhab.binding.zwave #435: Fibaro smoke sensor (FGSD-002) / Not getting all values from the device

Fejitatete, 2017-03-28:
Many channel state are not updated by ZWave events : heat ON and OFF, smoke ON. Temp and tamper are OK.

* OH2 Version:
`167 | Active   |  90 | 2.1.0.201703251201    | openHAB Core     org.openhab.core`
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console):
`212 | Active   |  80 | 2.1.0.201702121957    | ZWave Binding      org.openhab.binding.zwave`

* Device name:
FGSD002 Smoke Detector
* Device version:
3.1 & 3.2
* [Link to [database]](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/129)
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.
```
<node>
  <deviceClass>
    <basicDeviceClass>ROUTING_SLAVE</basicDeviceClass>
    <genericDeviceClass>SENSOR_NOTIFICATION</genericDeviceClass>
    <specificDeviceClass>NOTIFICATION_SENSOR</specificDeviceClass>
  </deviceClass>
  <homeId>0xe8680794</homeId>
  <nodeId>9</nodeId>
  <version>4</version>
  <manufacturer>0x10f</manufacturer>
  <deviceId>0x1002</deviceId>
  <deviceType>0xc02</deviceType>
  <listening>false</listening>
  <frequentlyListening>false</frequentlyListening>
  <routing>true</routing>
  <security>false</security>
  <beaming>true</beaming>
  <maxBaudRate>40000</maxBaudRate>
  <nodeInformationFrame>
    <commandClass>ZWAVE_PLUS_INFO</commandClass>
    <commandClass>VERSION</commandClass>
    <commandClass>MANUFACTURER_SPECIFIC</commandClass>
    <commandClass>DEVICE_RESET_LOCALLY</commandClass>
    <commandClass>ASSOCIATION_GROUP_INFO</commandClass>
    <commandClass>ASSOCIATION</commandClass>
    <commandClass>POWERLEVEL</commandClass>
    <commandClass>WAKE_UP</commandClass>
    <commandClass>BATTERY</commandClass>
    <commandClass>ALARM</commandClass>
    <commandClass>CRC_16_ENCAP</commandClass>
    <commandClass>CONFIGURATION</commandClass>
    <commandClass>SENSOR_MULTILEVEL</commandClass>
    <commandClass>MULTI_INSTANCE_ASSOCIATION</commandClass>
    <commandClass>APPLICATION_STATUS</commandClass>
    <commandClass>SENSOR_ALARM</commandClass>
    <commandClass>FIRMWARE_UPDATE_MD</commandClass>
  </nodeInformationFrame>
  <supportedCommandClasses>
    <entry>
      <commandClass>CRC_16_ENCAP</commandClass>
      <crc16EncapsulationCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
      </crc16EncapsulationCommandClass>
    </entry>
    <entry>
      <commandClass>DEVICE_RESET_LOCALLY</commandClass>
      <deviceResetLocallyCommandClass>
        <version>1</version>
        <instances>1</instances>
        <versionSupported>1</versionSupported>
      </deviceResetLocallyCommandClass>
    </entry>
    <entry>
      <commandClass>SENSOR_ALARM</commandClass>
[... 547 lines cut ...]
00:24:33.535 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 10 00 04 00 09 0A 71 05 00 00 00 FF 01 00 01 03 60
00:24:33.551 [DEBUG] [ve.internal.protocol.ZWaveController] - Process Message = 01 10 00 04 00 09 0A 71 05 00 00 00 FF 01 00 01 03 60
00:24:33.569 [DEBUG] [ve.internal.protocol.ZWaveController] - Message: class=ApplicationCommandHandler[0x04], type=Request[0x00], priority=High, dest=255, callback=0, payload=00 09 0A 71 05 00 00 00 FF 01 00 01 03
00:24:33.583 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 9: Application Command Request (ALIVE:DONE)
00:24:33.590 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 9: Starting initialisation from DONE
00:24:33.611 [DEBUG] [ve.internal.protocol.ZWaveController] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@1c2b8ae already registered
00:24:33.617 [DEBUG] [ssage.ApplicationCommandMessageClass] - NODE 9: Incoming command class ALARM
00:24:33.621 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 9: Received ALARM command V5
00:24:33.625 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 9: Process NOTIFICATION_REPORT V5
00:24:33.630 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 9: NOTIFICATION report - 0 = 0, event=0, status=255
00:24:33.633 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 9: Alarm Type = SMOKE (0)
00:24:33.638 [DEBUG] [ve.internal.protocol.ZWaveController] - Notifying event listeners: ZWaveAlarmValueEvent
00:24:33.641 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Got an event from Z-Wave network: ZWaveAlarmValueEvent
00:24:33.646 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Got a value event from Z-Wave network, endpoint = 0, command class = ALARM, value = 255
00:24:33.661 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
00:24:33.670 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter NOTIFICATION event is 0, type OnOffType
00:24:33.681 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 9: Updating channel state zwave:device:gateway:node9:alarm_smoke to OFF [OnOffType]
00:24:33.697 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
00:24:33.701 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
00:24:33.705 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION
00:24:33.709 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - Alarm converter processing NOTIFICATION

```

Thank you.

cdjackson, 2017-03-29:
Events are only updated when they occur - have you triggered these events?

Fejitatete, 2017-03-29:
> Pour : "openhab/org.openhab.binding.zwave" <org.openhab.binding.zwave@noreply.github.com>

As you can see in the first part of the log, I triggered the smoke event by pressing the button during 10 sec. The heat event is not shown in the log but I tested it by pressing the button in the green menu of the fgs.

----- Message de réponse -----

cdjackson, 2017-03-29:
> Pour : "openhab/org.openhab.binding.zwave" <org.openhab.binding.zwave@noreply.github.com <mailto:org.openhab.binding.zwave@noreply.github.com>>

Are you sure the heat alarm is sent through pushing the button? I don’t think so? From my previous tests, then only way I could get it to work was to actually put the sensor in front of a fire.  If the device isn’t sending an alarm, then the binding can’t really do anything about this.

For smoke I will take a look.

cdjackson, 2017-03-29:
One other point - you are using a very old binding so I would strongly suggest to update to a more recent snapshot given there have been changes to notifications in this timeframe.

I've found an error in the database where the smoke alarm was lookign at the wrong event. I'm slightly concerned that this change might stop it working on other devices but let's see.

This should close this issue - please advise.

Fejitatete, 2017-03-29:
I don't understand, I updated my openhab to the last snapshot this WE. Why is so old ?
When you enter the green menu of the fgs. I guessed it send an heat alarm to off. I will check that.

Thk

----- Message de réponse -----

Date : mer., mars 29, 2017 11:55

cdjackson, 2017-03-29:
I don’t know how often the pacakge is updated, but your version is about 6 weeks old I think so I guess it’s not updated very often.

Regarding the heat alarm, if there’s nothing in the log, then the binding can’t do anything. Maybe there is a way to get it to send the heat alarm, but when I tested this a few months ago, the only way I found to do it was to put it in front of the fire.

cdjackson, 2017-04-11:
Closing this - if it doesn't work once you get the binding updated then please re-open.

closed 2017-04-11 by cdjackson