openhab/org.openhab.binding.zwave #453: Code fails static analysis checks

kaikreuzer, 2017-04-02:
We have recently started to do [static code analysis for the openHAB code base](https://community.openhab.org/t/static-code-analysis-tool/24489) (many PRs with fixes on https://github.com/openhab/openhab2-addons already), but as ZWave is in its own repo, this hasn't yet been addressed.

I noticed [some errors being reported on recent PRs](https://openhab.ci.cloudbees.com/view/All/job/PR-org.openhab.binding.zwave/764/console), mainly regarding logging statements. Would be nice to have these adapted, so that the binding passes all checks.

cdjackson, 2017-04-02:
I fixed all the static logger errors yesterday - it will take longer to go through and look at everything else, and I don’t really want to do this on the current branch as it will not be continued. It’s a shame that this presents as a failed check rather than just a list of warnings or something like other such systems...

closed 2017-04-02 by cdjackson

kaikreuzer, 2017-04-03:
You can simply set failOnError=false, then it will only generate the report, but won't fail the build.

cdjackson, 2017-04-03:
Ah - ok, thanks. Where is this set?

I've not really looked at how this is all configured - on some of my other projects I'm using a separate CI system (codacy.com) to do such checks - it looks like it's fundamentally the same backend as being used here, but managed through a web interface.

kaikreuzer, 2017-04-03:
Check https://github.com/openhab/static-code-analysis

The correct parameter to set is `report.fail.on.error`.