openhab/org.openhab.binding.zwave #457: Enhancement Solution to set HexCode for a RFID Tag on Minikeypad Zwave (Zipato)

borstelmahlsdorf, 2017-04-03:
Openhab 2.0.0 Snapshot
I use a Zipato Mini Keypad RFID / ZWAVE
The Keypad has the Node ID 4 
=============== node4.xml =======================
`<node>
  <deviceClass>
    <basicDeviceClass>ROUTING_SLAVE</basicDeviceClass>
    <genericDeviceClass>ENTRY_CONTROL</genericDeviceClass>
    <specificDeviceClass>NOT_USED</specificDeviceClass>
  </deviceClass>
  <homeId>0xf4d909c4</homeId>
  <nodeId>4</nodeId>
  <version>4</version>
  <manufacturer>0x97</manufacturer>
  <deviceId>0x4501</deviceId>
  <deviceType>0x6131</deviceType>
  <listening>false</listening>
               :
</node>               
`
===============END NODE 4 ======================
patched in 2.0.0  Snapshot the the   ZwaveUserCodeCommandClass
################# SNIP Method #################################
``
public SerialMessage setUserCode(int id, String code) {
          boolean codeIsZeros = false;
          boolean isHexCode=false;
          logger.info("############# code is "+code+"#############################");
          if (!StringUtils.isEmpty(code)) {
              code=code.trim();
          } else {
              code="";
          }
         if (code.startsWith("0x")) {
            isHexCode = true;
        } else {
            // Zeros means delete the code
            try {
                codeIsZeros = Integer.parseInt(code) == 0;
            } catch (NumberFormatException e) {
                logger.debug("NODE {}: Error parsing user code. Code will be removed");
            }
            if (codeIsZeros) {
                code = ""; // send no code since we will set UserIdStatusType.AVAILBLE
            }
        } // END else ((code.startsWith("x")))
        if (codeIsZeros || userCodeIsValid(code)|| isHexCode) {
            logger.debug("NODE {}: {} user code for {}", this.getNode().getNodeId(),
                    codeIsZeros ? "Removing" : "Setting", id);
            SerialMessage message = new SerialMessage(this.getNode().getNodeId(),  SerialMessageClass.SendData,
                    SerialMessageType.Request, SerialMessageClass.SendData, SerialMessagePriority.Get);
            ByteArrayOutputStream outputData = new ByteArrayOutputStream();
            outputData.write((byte) this.getNode().getNodeId());
            outputData.write(4 + code.length());
            outputData.write((byte) getCommandClass().getKey());
            outputData.write(USER_CODE_SET);
            outputData.write(id); // identifier, must be 1 or higher
            if (codeIsZeros) {
                outputData.write(UserIdStatusType.AVAILABLE.key); // status
            } else {
                outputData.write(UserIdStatusType.OCCUPIED.key); // status
                try {
                    if (isHexCode) {
                        code=code.replace("0x","");
                        int len=code.length();
                        if (! (len%2 == 0)) {
                            logger.info("code length must be a vielfaches of 2");
                            return null;
                        }
                        int countBytes=len;
[... 92 lines cut ...]
2017-04-01 18:25:50.809 [DEBUG] [ve.internal.protocol.ZWaveController:1140] - Event Listener org.openhab.binding.zwave.internal.protocol.initialization.ZWaveNodeInitStageAdvancer@1847bf2 already registered
2017-04-01 18:25:50.810 [DEBUG] [ssage.ApplicationCommandMessageClass:152] - NODE 4: Incoming command class USER_CODE
2017-04-01 18:25:50.811 [DEBUG] [mmandclass.ZWaveUserCodeCommandClass:99] - NODE 4: Received UserCode Request
2017-04-01 18:25:50.812 [DEBUG] [mmandclass.ZWaveUserCodeCommandClass:111] - NODE 4: USER_CODE_REPORT 2 is OCCUPIED
2017-04-01 18:25:50.813 [DEBUG] [mmandclass.ZWaveUserCodeCommandClass:128] - NODE 4: USER_CODE_REPORT 2 is OCCUPIED [��J^P=P^A^D^@^@]
2017-04-01 18:25:50.814 [DEBUG] [ve.internal.protocol.ZWaveController:564] - Notifying event listeners: ZWaveUserCodeValueEvent
2017-04-01 18:25:50.815 [DEBUG] [ding.zwave.handler.ZWaveThingHandler:931] - NODE 4: Got an event from Z-Wave network: ZWaveUserCodeValueEvent
2017-04-01 18:25:50.817 [DEBUG] [ding.zwave.handler.ZWaveThingHandler:940] - NODE 4: Got a value event from Z-Wave network, endpoint = 0, command class = US
####################################     END LOG 
Now i can use the RFID Tag like a usercode with pressing keys on the keypad 
@chris: 
        - is it possible to show the key code in HEX and ASCII
          ( to show the incoming unknown TAG CODE and to show the known USER Code occupied by a 
           usercode) 
           to use incoming unknown Tag Code (Hex) to set the Usercode as HEX ( done with my patched Method)
        - the Zipato Documentation says 
        "After sending the User Code Set , including a Unique User Indentifier (UID) the in use State (0x01) and the TAG Code 
        or a keypad sequence using ascii codes. the mini keypad will accept the code and notify other devices using the alarm command class.  ...."
        It seems to be that only the keypad sequence is in asccii code    
        thanks in advance         
  

Thanks for the rapid reply and i hope, it is now better readable       

Alexander

borstelmahlsdorf, 2017-04-03:
[github.zip](https://github.com/openhab/org.openhab.binding.zwave/files/891201/github.zip)
Here the Complete Sourcecode of  ZWaveUserCodeCommandClass.java

cdjackson, 2017-06-24:
Hex codes are supported in dev branch now.

PR #547 (2017-04-23): Update user code command class to support hex codes
Closes #457
closed 2017-06-24 by cdjackson