openhab/org.openhab.binding.zwave #515: There is no XML-file created for the device: Remotec Z-URC 550EU

RayBe-M, 2017-04-06:
See this discussion:
[https://community.openhab.org/t/no-xml-file-created-for-new-device/24644/30](url)

As asked, the pasted explanation from the above discussion:
> The problem with node 2 is it is not reporting the WAKEUP command class in its NIF (which is should, so this is non-compliant to zwave). It then sends a message for this command class, so the binding says "ah - it supports wakeup" - so it adds this class. Then during initialisation we request the supported version, and the device returns a message stating that it doesn't support the wakeup class, so the binding removes it again and it gets stuck in this loop.

cdjackson, 2018-03-15:
>  but with another device,

Please don't add the information for different devices - it will be a totally different issue if a device is not initialising as it will be device specific.

cdjackson, 2019-08-11:
Closing as this is very old (2+ years). If the problem persists please reopen.

closed 2019-08-11 by cdjackson