openhab/org.openhab.binding.zwave #521: As a user, I would like OH2 to control my GE zwave plus smart dimmer model 28167.

gk4, 2017-04-13:
GE (Jasco) has upgraded their GE smart dimmer to z-wave plus. This new model number is 28167.

https://byjasco.com/products/ge-z-wave-plus-plug-smart-dimmer-single-outlet-500s

zwave_class_basic ROUTING_SLAVE 
zwave_class_generic MULTILEVEL_SWITCH
 zwave_frequent false 
zwave_neighbours 1,2
 zwave_version 5.21 
zwave_listening false
 zwave_plus_devicetype LIGHT_DIMMER_SWITCH
 zwave_deviceid 12344
 zwave_nodeid 3 zwave_routing false
 zwave_beaming false 
zwave_class_specific POWER_SWITCH_MULTILEVEL
 zwave_manufacturer 99
 zwave_devicetype 20548

Rasbian packages:

ii  openhab2                    2.0.0-1      all          openhab2
ii  openhab2-addons      2.0.0-1      all          openhab2-addons

Controller: Raspberry Pi3b with a RaZberry2 z-wave daughter card.

[oh2-logs.zip](https://github.com/openhab/org.openhab.binding.zwave/files/918182/oh2-logs.zip)

cdjackson, 2017-04-13:
Please add this to the database - see the link below -:

http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide <http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide>

gk4, 2017-04-13:
@cdjackson I don't have an "Add" button when I'm looking at the database.   I created user gkraft4@cd-jackson.com

BTW, here is the XML for the GE (Jasco Products) 28167 (ZW3104) Plug-in Smart Dimmer that is z-wave plus.

[jasco-28167.zip](https://github.com/openhab/org.openhab.binding.zwave/files/921071/jasco-28167.zip)

cdjackson, 2017-04-14:
I’ve updated your access so you should be able to add/edit now.

Cheers
Chris

gk4, 2017-04-15:
@cdjackson 

I have uploaded the XML, but I'm sure about the following:

Device supports configuration, but no parameters are defined.
Device supports associations, but no groups are defined.

cdjackson, 2017-04-15:
This indicates that the device properly implements configuration parameters and association groups. These are normally described in the manual, and if they exist, then they need to be added to the database so you can change the configuration. If they don’t exist, then just ignore this warning, but given that the device supports this functionality, it probably does exist.

closed 2017-05-27 by cdjackson