openhab/org.openhab.binding.zwave #538: Z-Wave Exception after update

antares2001, 2017-04-16:
Hello,

I upgraded via apt-get to the current nighty (mid April 2017) and since then I'm facing the following exception:

```
2017-04-16 23:13:19.651 [ERROR] [ve.internal.protocol.ZWaveController] - Exception during ZWave thread: Input 2. {}
java.lang.NullPointerException
	at org.openhab.binding.zwave.internal.protocol.ZWaveNode.setNodeState(ZWaveNode.java:235)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.serialmessage.ApplicationCommandMessageClass.handleRequest(ApplicationCommandMessageClass.java:46)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingRequestMessage(ZWaveController.java:248)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingMessage(ZWaveController.java:214)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.access$7(ZWaveController.java:208)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController$ZWaveInputThread.run(ZWaveController.java:1324)[220:org.openhab.binding.zwave:2.1.0.201704151452]
2017-04-16 23:13:19.705 [ERROR] [ve.internal.protocol.ZWaveController] - Exception during ZWave thread: Input 2. {}
java.lang.NullPointerException
	at org.openhab.binding.zwave.internal.protocol.ZWaveNode.setNodeState(ZWaveNode.java:235)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.serialmessage.ApplicationCommandMessageClass.handleRequest(ApplicationCommandMessageClass.java:46)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingRequestMessage(ZWaveController.java:248)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingMessage(ZWaveController.java:214)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.access$7(ZWaveController.java:208)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController$ZWaveInputThread.run(ZWaveController.java:1324)[220:org.openhab.binding.zwave:2.1.0.201704151452]
2017-04-16 23:13:20.527 [ERROR] [ve.internal.protocol.ZWaveController] - Exception during ZWave thread: Input 2. {}
java.lang.NullPointerException
	at org.openhab.binding.zwave.internal.protocol.ZWaveNode.setNodeState(ZWaveNode.java:235)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.serialmessage.ApplicationCommandMessageClass.handleRequest(ApplicationCommandMessageClass.java:46)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingRequestMessage(ZWaveController.java:248)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingMessage(ZWaveController.java:214)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.access$7(ZWaveController.java:208)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController$ZWaveInputThread.run(ZWaveController.java:1324)[220:org.openhab.binding.zwave:2.1.0.201704151452]
2017-04-16 23:13:21.528 [ERROR] [ve.internal.protocol.ZWaveController] - Exception during ZWave thread: Input 2. {}
java.lang.NullPointerException
	at org.openhab.binding.zwave.internal.protocol.ZWaveNode.setNodeState(ZWaveNode.java:235)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.serialmessage.ApplicationCommandMessageClass.handleRequest(ApplicationCommandMessageClass.java:46)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingRequestMessage(ZWaveController.java:248)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.handleIncomingMessage(ZWaveController.java:214)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController.access$7(ZWaveController.java:208)[220:org.openhab.binding.zwave:2.1.0.201704151452]
	at org.openhab.binding.zwave.internal.protocol.ZWaveController$ZWaveInputThread.run(ZWaveController.java:1324)[220:org.openhab.binding.zwave:2.1.0.201704151452]
```

Since I'm new to all this, please let me know which more information (and how to find it) I can provide to help you figure out the issue.

cdjackson, 2017-04-16:
What system are you running? What version? What was the version you upgraded from?

antares2001, 2017-04-16:
Continued here, as there was a post related with the same problem: https://community.openhab.org/t/npes-in-zwave-binding-with-latest-sanpshot-build-2-1-0-201704151452/26834/6

closed 2017-04-16 by antares2001