openhab/org.openhab.binding.zwave #548: New Sensative Strip Model Number

thx2000, 2017-04-19:
@cdjackson I'm guessing the old xml will work for this version, but I just received a sensative strip with model #: 11-01-012 Is it possible to just duplicate the existing xml so that it supports this model as well?

cdjackson, 2017-04-19:
I guess it depends on how different it is.I’ll drop Strips an email - they are pretty quick to respond.

cdjackson, 2017-04-19:
Sensitive Strips confirm that the -012 version is just the US version of the -011. Please can you provide the XML file that OH generates for this device - or at least provide the device type/id for your version. We can then update the database with this information.

thx2000, 2017-04-19:
I played with it some more to try to get you those numbers, and it did finally register as is.  I only had it about 3 meters away from the z stick before, but I found I had to be right on top of it to get it to register.  Sorry to waste your time, thanks for your help!!!

closed 2017-04-19 by thx2000